robotology/mesh-support #1: Report of on-site visit of 10/3-14/3 to Heildeberg

lornat75, 2015-03-17:
@randaz8 @traversaro 

Can you write a brief report about the problems you found, how you solved them and the problems still present?

traversaro, 2015-03-17:
For the software part related to codyco/koroibot software, I already posted a detailed report in https://github.com/robotology/codyco-superbuild/issues/61 .

traversaro, 2015-03-18:
The main points anyway are: 
  * The torso torque control was not ready, so we tried to run the `torqueBalancing` module with just the 12 DOFs of the legs, but apparently there where numerical problems related to using just the legs. Regarding this we are going to test on `iCubGenova01` (as soon as it is fixed : ) ) the 12 dofs balancing to try to understand if we can easily fix it.

traversaro, 2015-03-18:
cc @hu-yue

hu-yue, 2015-03-18:
For the other problems that we encountered I would wait @randaz81.

But I would like to report errors due to the skin.
Actually when you were here you used the robotInterface with disabled skin.
I included them now and get tons of errors and warnings and robotInterface gets stucked in the errors on embObjSkin.
You can download the log from yarplogger here: 
https://www.dropbox.com/s/xvbfexw9t2xy407/ypmrprunlog_18_03_2015_01_34_27.log?dl=0

For some reasons, if I start robotInterface from a yarpmanager launched on a machine in the cluster different than icubsrv (the laptop), the log is different and stops earlier, do you have any clue about why? This is the log of robotInterface in this case:
https://www.dropbox.com/s/xj83p17jyljtd2q/ypmrprunlog_18_03_2015_01_17_16.log?dl=0

lornat75, 2015-03-18:
@marcoaccame, @randaz81 ?

marcoaccame, 2015-03-18:
best thing to do is that we arrange a teamviewer session in which i can help fixing the problems related to skin.

from what i see from the log, there are the following issues.

1. the detection in embObjSkin of errors in skin can boards attached to EMS 10.
-> In files hardware/skin/xxxx-skinSpec.xml there should be:
-> diagnostic =  false  (and not true as you have)
-> that will remove the prints "embObjSkin error code:......."

2. the FW version of boards EMS 10 and 11 is still 1.63 and should be updated to 1.68. 
 -> that may help in some situations.

3. diagnostics messages from EMS 11 tell that the skin can boards are transmitting but there is a buffer overflow inside the EMS. That happens if: (a) the EMS cannot empty the buffer maybe because they are not in run mode, (b) the skins are unfortunately still switched on thus canframes keeps on arriving, and (c) FW version of EMS is before 1.63 or older.  
-> i believe that the problem is that there are IP packet losses in the channel and the switch off
-> command of skins in CAN1 got lost. but arrived the switch off of CAN2's skin, and the
-> exit from run mode.

Having said that, better arranging a teamviewer session so that i can verify things, and if needed put dedicated debug FW on EMS 10/11.

hu-yue, 2015-03-18:
Thanks.
I changed the diagnostic to false and got rid of the error in which robotInterface was stucked.
Now it runs and the skin works, however there are still errors and warnings.
I did not change the firmware.
The new log is here: 
https://www.dropbox.com/s/shgdpkhpqnhjghh/ypmrprunlog_18_03_2015_06_16_53.log?dl=0

marcoaccame, 2015-03-18:
try to run it by enabling only the skins. i hope that by doing that there are no lost ip packets and that we can have a clearer test condition.

lornat75, 2015-03-19:
Two suggestions:
- Can the EMS send a 'switch off' to all channels when exiting run mode so it is robust to missed packets? 
- We should increase the socket buffer size to avoid packet loss + schedule an upgrade to the pc104 with realtime kernel, it has been tested on our iCub for a long time and I think it can be considered stable

hu-yue, 2015-03-19:
I updated the firmware on boards 10 and 11 to ver 1.68 and now basically all the warning and errors are gone.
This is the log of robotInterface launched with skin only:  https://www.dropbox.com/s/61xkaf6cwz9lvet/yamrprunlog_19_03_2015_09_57_24.log?dl=0
This is the log of robotInterface launched with everything including skin: https://www.dropbox.com/s/vikr18kulwp2w7e/yamrprunlog_19_03_2015_10_35_29.log?dl=0

marcoaccame, 2015-03-19:
OK Hu, I am happy. But I would like to investigate some more. The reason is that fw version 1.68 just does not fill the buffer with skin canframes when skin should not be used. But it may be that the skin can boards on CAN1 are still transmitting, and it should not happen.

By the way, you sent me a wrong log: the first one is with fw ver 1.63. the second is with 1.68 but skin only. what is missing is everything bu fw version 1.68.

After I have seen this last log, if it ok for you i would like to test a special fw to see exactly what happens.

marcoaccame, 2015-03-19:
To Lorenzo:
when I wrote:
-> i believe that the problem is that there are IP packet losses in the channel and the switch off
-> command of skins in CAN1 got lost. but arrived the switch off of CAN2's skin, and the
-> exit from run mode
I was wrong: the skin is not switched off by robotinterface but at exit of run mode.

The behaviour of skin activation / disactivation is the following.
1. embObjSkin sends a tx-enable command for a given set and the EMS enables all the skin can boards on this set and puts skintxstatus = on. this command is always processed by the EMS either it is in run or cfg mode.
2. then embObjSkin sends the EMS in run mode. It may be that the EMS was already in run mode if another device, e.g., embObjMotionControl, did it before.
2. at CTRL-C, the EMS exits run mode. If skintxstatus is on, then the skin is silenced.

hu-yue, 2015-03-19:
Sorry, I must have done some mistake in uploading the logs.
Then this is the one with only skin as you said: https://www.dropbox.com/s/vikr18kulwp2w7e/yamrprunlog_19_03_2015_10_35_29.log?dl=0
And this is the log with everything: https://www.dropbox.com/s/kzxq1y1l8oye55p/ypmrprunlog_19_03_2015_12_05_16.log?dl=0

randaz81, 2015-03-23:
Heildelberg visit activity report:

HW:
1- The new fw build performs checks to verify the quality of the encoder measurements. After the update it was possible to detect a large number of errors due to electromagnetic interferences (EMI) These errors were initially affecting only the ankle joints. After an attempt to insulate the AEA chip from its metal support, these errors disappeared from the ankle joints and appeared on other joints (knee, hip joint). These errors do not show when the joints are in idle mode, so the source of the noise is certainly the motors PWM. According to the plots, the encoder spikes are 20-80 degrees high and last a couple of milliseconds, enough to trigger a visible jerky movement of the joints. In order to hide the large number of error prints on the robotInterface console (which were causing a general slowdown of the robot performances), a custom fw was deployed on the EMS boards.
2- EMI were also responsible for a large number of ETH packet loss. I recommended a general check of the robot wiring, verifying the shielding of the AEA and ETH cables, their routing, the distance between the EMS and the 2FOC boards, the grounding of the robot frame etc

FW:
1- we studied and implemented a new torque control law. The fw is experimental and is currently running on boards 6 7 8 9 (legs joints). We are now performing final checks to validate the system and make the new algorithm the standard one on all ETH robots. The formula is reported in: http://mathb.in/28925

2- new firmware implements metric units (instead of machine units). The  configuration section now implements the following parameters:
``` 
  <group name="TORQUE_CONTROL">
        <param name="controlLaw">    motor_pid_with_friction_v1 </param> 
        <param name="controlUnits">  metric_units               </param> 
        <param name="kp">         100          100              </param>       
        <param name="kd">           0            0              </param>       
        <param name="ki">           0            0              </param>       
        <param name="maxPwm">    8000         8000              </param>       
        <param name="maxInt">     500          500              </param>       
        <param name="shift">        0            0              </param>       
        <param name="ko">           0            0              </param>       
        <param name="stictionUp">   0            0              </param>       
        <param name="stictionDwn">  0            0              </param>   
        <param name="kff">          1            1              </param>         
        <param name="kbemf">        0            0              </param>  
        <param name="filterType">   0            0              </param>        
        <param name="ktau">       214          178              </param>          
    </group>
```
On the robots the are not updated with the new firmware, embobjmotioncontrol prevents the switch to torque control mode if parameters 'controlUnits' and 'controlLaw' are not found.

SW:
1 - We identified parameters kff, kbemf, ktau using the friction identification procedure (matlab) and we validated the FW torque control law for single non-coupled joints (robots legs). For the torso (which implements a 3-joints coupling), we are currently validating the signs of the motor velocities vs joints velocities. (See also issue #5 and https://github.com/robotology/robot-testing/issues/14 )

lornat75, 2015-03-23:
@maggia80, @giorgiometta @ale-git @iron76

lornat75, 2015-03-23:
@randaz81, @ale-git, @traversaro: the material in this report should go in a wiki page of the manual.

traversaro, 2015-03-23:
:+1: , as soon as these changes are "deployed" we can review the Force Control page.

hu-yue, 2015-04-10:
Hi,
Quite some time has passed, I would like to ask if you can tell me how should we address the hardware problems of the robot as per point 1 of the report.
As we discussed about the possibility to send the robot back to IIT, I need to know exactly if and when we have to do this asap, possibly within next week the latest.
Thanks!

super-ste, 2015-04-17:
Hi YueHe,
we shall get  the robot back here in Genova in order to troubleshoot and solve the reported iCub issue.
Unfortunately I cannot foresee how long it will take to fix it (there can be multiple causes) 

When do you think you can ship it back? [iCub Supporters are going to be out in a mission until April 27th, so no hurries on that].

I'd say after one week from receiving we shall be able to tell how long to ship it back.

When would you need it?
Cheers

hu-yue, 2015-04-17:
Hi, 
My idea was to bring it back myself around 30th April/1st May, I will stay for some time in Genova (at least one week for the time being), but if it will take too long (like more than 3 weeks/one month or so) then I will come back to Heidelberg and ask you to ship it instead of bringing it back.
We definitely need it latest by the end of May, but I guess that will depend on the problems you will find out.

What do you need apart from the robot itself? The laptop? Power cable, power suppliers etc?
Thanks!

super-ste, 2015-04-17:
Hi @hu-yue 
cool if you can drive it here yourself. We will ship it back once it's fixed. No problem.

Please bring the laptop. We shall be able to power it with our cables/power-supply.

hu-yue, 2015-04-17:
Ok @super-ste I will let you know a precise date when I will know.

hu-yue, 2015-04-23:
Hi @super-ste, I confirm that I will bring the robot to IIT on the 30th April (next Thursday), is morning around 8.30-9.00 ok?
I will take with me only the robot and the laptop without the power suppliers and any cables.

super-ste, 2015-04-23:
Hi Yu-He
I confirm it’s OKAY.

We’ll work on it and le you know our findings as soon as possible.

lornat75, 2015-10-20:
Any update on this issue?

hu-yue, 2015-10-20:
I think we can close it for the time being, the problems of this issue have been solved and the new issues of the robot do not really regard this issue anymore.

closed 2015-10-20 by hu-yue