robotology/mesh-support #3: Problems on the tactile sensors of the hands

lornat75, 2015-03-17:
@alecive, @massimoregoli, @pattacini reported that the tactile sensors on the hand do not always work after startup. 

In particular we noticed that if the palm is pressed at startup the fingertip reach saturate to max value and never come back. This suggests a hardware issue.

lornat75, 2015-03-30:
@maggia80

maggia80, 2015-04-17:
I am working on a board that will be mounted on top of the mais. The board will give us the possibility to test each individual fingertip without unsolder anything. This is because we are having a lot of problems with the fingertips in this period and the debug (even with the skin diagnostic enabled) is complicated.

maggia80, 2015-04-27:
next week we will received the boards, and we will test it on the new hands (2.0) of the black robot.

bhigy, 2015-06-29:
Have there been any update on this? Today, I noted issues on both hands. 

On the left one, the palm is not working. 

On the right one, the fingertip of the middle finger is not working while the thumb, the index and the ring fingers stopped working after a few minutes. Also, there were artifacts on the little finger (it was flickering even though I didn't touch it).

super-ste, 2015-06-29:
This issue will be addressed once we have some proof the new board as per issue #33 is resolutive.

lornat75, 2015-10-20:
Issue #33 was closed. Can we close this issue too?

bhigy, 2015-10-20:
Right hand fingertips were still not working a few days ago.

massimoregoli, 2015-10-20:
I didn't experience this issue anymore, it's probably due to recent firmware updates. I guess we can close it.

closed 2015-10-20 by massimoregoli