robotology/mesh-support #4: iCub network speed goes down and up

gsaponaro, 2015-03-19:
Last week, during intense usage of our robot for the POETICON++ review, we experienced occasional network slowdowns that turned the robot and the yarp network very difficult to use. For example, while we were running IOL and other vision modules, camera framerate suddenly went down to 2-3 fps, stayed there for a couple of minutes, then automatically resumed to full speed again.

We would like to locate and address this problem asap, can you recommend us any tool for that? We have already used **iperf** to test the network throughput, but that gives good speed results. @vtikha mentioned that some network monitoring tool is used at IIT.

Our iCub network is Gbit on a dedicated Cisco Catalyst 2960 switch. Software sources and binaries are shared within the cluster via NFS.

/cc @lorejam

lornat75, 2015-03-19:
Hi Gioanni,
one useful thing to do is to log to the CISCO monitoring interface (using the browser), check the available bandwidth and (although it does not seem the case here) that all ports are working @1Gb.

If the iperf tests are sane in ALL directions and using mcast, udp and tcp I would then check CPU usage on the involved machines. This can affect the framerate.

robertocalandra, 2015-03-20:
Hi all,
On a related topic, we had problems streaming from the eyes (at 640x480) and at the same time using direct position control.
The network seems to get saturated and the stream from the eyes decrease to about 13fps.
Is there any way-around?

lornat75, 2015-03-30:
@gsaponaro any news on this?

lornat75, 2015-03-30:
@robertocalandra, yes please open a new issue.

gsaponaro, 2015-03-30:
Just verified again that **iperf** tests are sane. Testing between the pc104 and another machine in the cluster we get the following (both directions):

tcp 900 Mbits/s
udp 800+ Mbits/s
udp with big datagrams 950 Mbits/s
mcast 950 Mbits/s

Cisco swich performance: we asked our network administrator to configure Cacti graphs, hopefully it will be ready soon.

I think in the end it should be a CPU issue. If I remember correctly it was also @vtikha 's suspicion when he was here. At the moment I am travelling, soon @lorejam and I will do more intense tests. (Feel free to close issue, I will reopen if needed.)

Just one more thing: do you have any other Debian/Ubuntu tool recommendations to monitor the state of network and CPU load on a cluster? @vtikha mentioned that Alessandro Bruchi installed something like that at IIT, but I don't know his GitHub nick.

lornat75, 2015-03-31:
@gsaponaro: thanks. We can leave this open, please update as soon as you have more info. 

For the switch performance: all you need is the ip for the administrative interface and the administrator account. The tool is pretty easy to do, it covers most of the networking debug you need to do.

For the CPU you can use TOP.

gsaponaro, 2015-11-16:
Closing for now as we haven't observed network problems for some time. If they resurface, we'll perform a network analysis similar to the one done in https://github.com/robotology/icub-main/issues/237#issuecomment-156399405 and report back.

closed 2015-11-16 by gsaponaro