robotology/mesh-support #5: Robot velocities (log request)

randaz81, 2015-03-19:
Hi @hu-yue,
we are validating the new torque control law implemented last week on your robot.
As double check, we need a quick log from your robot. Here are the steps:
1) Put all the motors of the robot in idle
2) type the following command: yarp read ... /icub/left_leg/stateExt:o > log_left_leg.txt
3) move (push) the joints randomly. Speed is not important. 30 seconds are ok. Just be sure that all the joints (including the difficult ones, such as the ankles) have been pushed back and forth a couple of times.
4) press ctrl+c to terminate and save the file.
Repeat for the other leg and the torso. The whole procedure should not take more than 10 minutes. Thank you!

hu-yue, 2015-03-20:
Hi, here are the logs: 
https://www.dropbox.com/sh/cm5oqcgaliy65av/AABby84149TlvWseLDcVhye7a?dl=0
Let me know if there are any problems.

randaz81, 2015-03-23:
thank you, logs are perfect.

closed 2015-03-23 by randaz81