robotology/mesh-support #8: OS to run real-time Linux on PC104

marcoaccame, 2015-03-30:
From SW meeting of 26/03/15: 
We need to upgrade the PC104 OS to run real-time Linux (robots Darmstadt, Shaffield, Heidelberg).

If I remember well, issues are:

1. verify if the version of the USB live key is the right one supporting real-time. Alberto, and Matteo: can you please help me in that?

2. verify that the required system network settings are correct. I remember there was a redmine issue with the settings written down. I will look for it, then we need to verify if they are also in the relevant config file.

3. be sure that icub-main is compiled with the correct CMake flag. I will check that with Ali Paikan.

4. verify if camera streaming works fine, as Alberto recalls that at the beginning streaming together with real time gave anomalies.

@barbalberto 
@mbrunettini 
@drdanz 
@apaikan

mbrunettini, 2015-03-30:
@marcoaccame
@barbalberto

1)      Real Time kernel is supperted from Icub-live version 5  http://wiki.icub.org/wiki/The_Linux_on_the_pc104

2)      The network setting have been imported into the latest icub image versions

Matteo

mbrunettini, 2015-03-30:
In details, network settings in the icub-live v5 are (see http://wiki.icub.org/wiki/ICub_PC104_Linux_Image_-_Detailed_info#Other_improvements ) :
  The following files are modified in order to obtain better real time netowrking performaces
  /etc/security/limits.d/icub.conf
    icub	soft	rtprio	99
    icub	hard	rtprio	99
  /etc/sysctl.d/20-net-rbuffer.conf
    # This sets the max OS receive buffer size for all types of connections
    net.core.rmem_max=8388608

marcoaccame, 2015-03-30:
Thank you Matteo, the system settings are OK with what we have used in following task of redmine (https://redmine.icub.iit.local/issues/87).

barbalberto, 2015-03-30:
I did a quick check for older version of usb key images.
The version 3.2 is the one with the dual boot, normal kernel and rt one. It is quite old as a key (kernel 2.6.33) and probably is the one that was giving issues with the firewire streaming, so I'd suggest to not use it.
The image 4.0 has only normal kernel, probably because previous rt version was not satisfying.

If I understand correctly the icub-live v5 has only the rt kernel, so simply updating the image to version v5 will make sure the pc104 gets the rt kernel with all related network configuration already set-up, right?

mbrunettini, 2015-03-30:
Icub-live V5 has only the RT  kernel.

marcoaccame, 2015-03-30:
from Ali:
icub-main must be compiled with ICUB_REALTIME_EXPERIMENTAL set to ON (default is OFF).

apaikan, 2015-03-30:
Some comments on what `ICUB_REALTIME_EXPERIMENTAL` does: 
* The option affects the PC104 which has been delivered with the Real-time Linux kernel.
* The option (if enabled) will adjust the thread scheduling policies and priorities of the embObjLib and the robotInterface's interfaces/communication threads on the PC104 as shown here: 

|         Threads           | Priority  |       Policy       |
| -------------------------------|:-----------:|:---------------------:| 
| embObjLIB (receiver) |     49    |  SCHED_FIFO |
| embObjLIB (sender)  |     48    |  SCHED_FIFO |
| robotInerface (all)      |     33    |  SCHED_FIFO |

* The priorities have been chosen so that they are higher than other process's priorities and less than the operating system thread-interrupt priorities (i.e. > 50). 
* This ensures that the robotInterface has the highest priority to communicate withe the controller boards and with the remote YARP processes.  

**Notice**: Theoretically the `ICUB_REALTIME_EXPERIMENTAL` can be used with the non-Real-time Linux kernels (it is also a good practice). However, while launching robotInterface on a non-real-time PC104, the user may get a warning message saying that `Cannot set the scheduler to FIFO. (check superuser permission)`. This is because adjusting real-time properties of processes and threads on Linux requires a specific permission which can be adjusted by editing `/etc/security/limits.conf` on the PC104 as shown here: 
```
icub          soft    rtprio          99
icub          hard    rtprio          99
```
**This is already adjusted on the PC104 which has been delivered with the real-time kernel!**

marcoaccame, 2015-03-30:
Ok, I would suggest that we test the icub-live version 5 key once more on iCubGenova04 with icub-main being compiled with ICUB_REALTIME_EXPERIMENTAL ON.

Then the same on iCubGenova01 and also on iCubSheffield01 which is here in Genova now.

marcoaccame, 2015-04-01:
errata: iCubGenova01 
corrige: iCubGenova02 (the purple one).

About iCubGenova04: the icub-live v5 was not in use yet. 
Thus, Matteo and I prepared the system to use it and since the afternoon of 31 march the robot is running the new RT kernel. We haven't seen any problem in camera streaming.

The icub-live v5 was tested in version icub-live_5.0-15.03.30-rt-amd64.iso.gz

Instruction about how to migrate from the old to the latest icub-live v5 will soon be provided in the icub wiki. we shall test them in performing migration for iCubGenova02 and the iCubSheffield01 robots.

traversaro, 2015-04-16:
On the `iCubParis02` the robotInterface saturates the pc104, and this result in a lot of FT sensors dropped. Just to give an idea, the jitter can reach up to 200 milliseconds to a first quick check, on a robot running only the `robotInterface` and no cameras or any external module. 

While this behavior should be properly debugged, it could make anyway sense to deploy the new realtime PC104 image even on `iCubParis02`.

lornat75, 2015-04-23:
@traversaro moving your question to a separate issue #17

lornat75, 2015-04-23:
This issue triggered investigation on pc104 usage with ethernet. At the moment we introduced a set of parameters to reduce number of packets sent by the EMS boards and the frequency with which they are received and transmitted by robotinterface.

lornat75, 2015-10-20:
@marcoaccame: can you open individual issues for the upgrade to RT kernel of individual robot so we can close this issue?

marcoaccame, 2015-10-22:
Created new issue #87 for the upgrade to RT kernel of robots and closed this issue.

closed 2015-10-22 by marcoaccame