robotology/mesh-support #9: Network Slowing Down

robertocalandra, 2015-03-30:
Hi all, 
we experienced problems streaming from both eyes (at 640x480) and at the same time using direct position control for the arms+torso.
The network seems to get saturated, the stream from the eyes decrease to about 13fps, and the direct position control erratic (obviously).
Do you know what could be the cause of this issue? Is it an hardware bottleneck? Is there any way-around?

lornat75, 2015-03-30:
Are you streaming the images with a) RGB or b) bayer encoding?

Which switch are you using? The netgear shipped with the iCub?

At which frame are you sendng directPosition requestes? Can you post the relevant code?

lornat75, 2015-10-20:
Closing for lack of feedback, we can re-open it if the problem becomes relevant again.

closed 2015-10-20 by lornat75