robotology/mesh-support #10: FTsens sensor send an unexpected message

traversaro, 2015-04-02:
Sometimes (I did not understood a correct way to replicate this) the `right_leg` FT sensor of the `iCubGenova03` is giving an error:
```
468335		130549.704503		ERROR		AnalogWrapper: right_leg: Sensor returned error with code 1.
```
By further inspecting, we discovered that the sensor is receiving an unexpected message:
~~~
468338		130550.388788		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468339		130550.388833		ERROR		b3
468340		130550.388843		ERROR		84
468341		130550.388851		ERROR		54
468342		130550.388858		ERROR		81
468343		130550.388866		ERROR		cb
468344		130550.388873		ERROR		86
468345		130550.388880		ERROR		1
468346		130550.388888		ERROR		1
468347		130550.388896		ERROR		eb
468348		130550.388903		ERROR		3
468349		130550.388910		ERROR		0
468350		130550.388917		ERROR		0
468351		130550.388924		ERROR		6
468352		130550.388931		ERROR		0
468353		130550.388937		ERROR		b2
468355		130551.979145		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468356		130551.979206		ERROR		b0
468357		130551.979227		ERROR		7e
468358		130551.979243		ERROR		2b
468359		130551.979258		ERROR		82
468360		130551.979273		ERROR		27
468361		130551.979287		ERROR		7e
468362		130551.979302		ERROR		1
468363		130551.979317		ERROR		80
468364		130551.979333		ERROR		ea
468365		130551.979349		ERROR		3
468366		130551.979364		ERROR		0
468367		130551.979379		ERROR		0
468368		130551.979393		ERROR		6
468369		130551.979408		ERROR		0
468370		130551.979422		ERROR		b3
468371		130552.210789		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468372		130552.210862		ERROR		ae
468373		130552.210879		ERROR		7e
468374		130552.210894		ERROR		2b
468375		130552.210907		ERROR		82
468376		130552.210921		ERROR		2c
468377		130552.210934		ERROR		7e
468378		130552.210947		ERROR		1
468379		130552.210960		ERROR		80
468380		130552.210974		ERROR		ea
468381		130552.210987		ERROR		3
468382		130552.210999		ERROR		0
468383		130552.211012		ERROR		0
468384		130552.211024		ERROR		6
468385		130552.211036		ERROR		0
468386		130552.211049		ERROR		b0
468387		130552.502802		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468388		130552.502863		ERROR		aa
468389		130552.502884		ERROR		7e
468390		130552.502900		ERROR		2d
468391		130552.502914		ERROR		82
468392		130552.502929		ERROR		26
468393		130552.502943		ERROR		7e
468394		130552.502957		ERROR		1
468395		130552.502972		ERROR		80
468396		130552.502988		ERROR		ea
468397		130552.503002		ERROR		3
468398		130552.503017		ERROR		0
468399		130552.503032		ERROR		0
468400		130552.503046		ERROR		6
468401		130552.503060		ERROR		0
468402		130552.503074		ERROR		b3
468403		130552.694126		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468404		130552.694188		ERROR		b6
468405		130552.694202		ERROR		84
468406		130552.694213		ERROR		57
468407		130552.694223		ERROR		81
468408		130552.694234		ERROR		cd
468409		130552.694244		ERROR		86
468410		130552.694254		ERROR		1
468411		130552.694264		ERROR		1
468412		130552.694274		ERROR		eb
468413		130552.694284		ERROR		3
468414		130552.694294		ERROR		0
468415		130552.694304		ERROR		0
468416		130552.694314		ERROR		6
468417		130552.694323		ERROR		0
468418		130552.694333		ERROR		b0
468422		130553.056670		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468423		130553.056730		ERROR		a9
468424		130553.056751		ERROR		7e
468425		130553.056768		ERROR		2a
468426		130553.056783		ERROR		82
468427		130553.056798		ERROR		26
468428		130553.056813		ERROR		7e
468429		130553.056828		ERROR		1
468430		130553.056844		ERROR		80
468432		130553.056878		ERROR		ea
468434		130553.056910		ERROR		3
468436		130553.056942		ERROR		0
468437		130553.056957		ERROR		0
468438		130553.056972		ERROR		6
468439		130553.056987		ERROR		0
468440		130553.057001		ERROR		b4
468444		130553.348677		ERROR		CanBusAnalogSensor: unknown error with device 14 with msg of len 15, message is:
468445		130553.358769		ERROR		ae
468446		130553.358811		ERROR		7e
468447		130553.358834		ERROR		2a
468448		130553.358851		ERROR		82
468449		130553.358867		ERROR		2b
468450		130553.358882		ERROR		7e
468451		130553.358897		ERROR		1
468452		130553.358913		ERROR		80
468453		130553.358929		ERROR		ea
468454		130553.358944		ERROR		3
468455		130553.358959		ERROR		0
468456		130553.358974		ERROR		0
468457		130553.358989		ERROR		6
468458		130553.359004		ERROR		0
468459		130553.359018		ERROR		b2
~~~
The usual `len` of periodic messages from the sensor is 6, and is 7 for overflow messages. 

The error is usually not continuous (it is getting an unexpected message but also a lot of "proper messages"), so it does not disturb the use of the sensor, but this issue may be interesting. The sensor is running the latest stable `STRAIN` firmware 3.0.1 . 

cc @randaz81

super-ste, 2015-11-17:
@randaz81 any news on this?
@traversaro, is the issue still present?

traversaro, 2015-11-17:
I don't know. It would be worth to check if this happens on the new `STRAIN` firmware. @DanielePucci

traversaro, 2015-11-17:
Anyway, it is definitely to a problem, just a minor annoyance , we can close the issue anyway.

DanielePucci, 2015-11-17:
Yes. I agree with @traversaro: we can close this issue. We are about to test the new firmware @valegagge @davege89 @marcoaccame

traversaro, 2015-11-17:
@DanielePucci I suspect anyway that if this does not happend on ETH robot may be also due to the fact that the FTSens ---> EMS ---> ethAnalogSensors (pc104) does not print out unexpected messages.

DanielePucci, 2015-11-17:
Ack.

closed 2017-04-23 by amaroyo