robotology/mesh-support #11: [iCubParis02] addresses of feet 6-axis FT sensors

traversaro, 2015-04-07:
Hi everyone, 
  currently there are some CAN-based iCub's that have 6-axis FT sensors in the ankles: 
 * iCubGenova01 
 * iCubGenova03 
 * iCubParis01

On all this robots, the convention for the feet FT sensors is to have a CAN ID of `0x01`, on the two CAN networks 3 and 4. 

However, we recently added the feet FT sensors to `iCubParis02`, but on that robot (v2.0, CAN based) unfortunately the `0x01` CAN ID on networks 3 and 4 is already used by the `6SG` board. [1]

Considering that even if we don't use the 6SG board in the robotInterface we want to have it properly connected to the net, we had to choose another CAN ID for the two FT sensors of the foot. 
However, in the long term which ID should be used for this sensors?

Thanks for the help, 
      Silvio

cc @maggia80 @lornat75 @rlober @darwinlau 

[1] : http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware#iCub_v2

lornat75, 2015-10-20:
@maggia80 please decide so we can close this issue

maggia80, 2015-10-20:
we can use address 3 for LeftLeg Strain and 4 for RightLeg Strain 

http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware#CFW2CAN4:_right_leg_3

closed 2015-10-20 by maggia80