robotology/mesh-support #12: iCubParis02 leg BLL board not working (digital part)

iron76, 2015-04-08:
Dear iCub-support,
the joint0 and joint1 of the iCubParis02 left_leg suddenly stopped working. The behaviour was the following: initial calibration was failing and left_leg exited the calibration with all joints in IDLE mode (this behaviour is normal after failing the calibration). The reason for not concluding the calibration turned out to be some control issue at the very low level. When running the motors from the robotMotorGui, the joint0 and joint1 were performing jerky movements, sometimes getting stuck in a specific configuration while making strange electronic noises (like high currents running through the motor windings). The issue seems to be related with wrong brushless phases commutation (seemingly the Hall effect sensors, but no error in this sense was reported by the board).

After some debugging with @rlober we found out that the BLL was causing the issue (the digital part of the BLL-BLP group). We took the right leg BLL and the problem disappeared. Now we need a replacement as soon as possible.

lornat75, 2015-04-08:
@giorgiometta

maggia80, 2015-04-27:
after replacing the board 3B6 the problem seems to be solved.

closed 2015-04-27 by maggia80