robotology/mesh-support #13: Different MOT_speed_estimation and MOT_accel_estimation values in similar robots

traversaro, 2015-04-09:
Hi icub-support, 
    we are working on the iCubParis02, and we noticed an interesting patterns in the motorControl configuration files.
TL;DR :  Interestingly **`MOT_speed_estimation` and `MOT_accel_estimation` have a value of 1 for all robots, but on `iCubGenova01` they have a value of 5**.

#### Boring explanation
The `MOT_speed_estimation` and `MOT_accel_estimation` parameters are usually set to 1 in the configuration file of most robots: 
* `iCubGenova03` : https://github.com/robotology/icub-main/blob/master/app/robots/iCubGenova03/hardware/motorControl/icub_left_leg.xml
* `iCubGenova04` : https://github.com/robotology/icub-main/blob/e4f946a5d6bc21ddeb8017f465f8b498728eb65f/app/robots/iCubGenova04/hardware/mechanicals/left_lower_leg-ems7-mec.xml
* `iCubParis01` : https://github.com/robotology/icub-main/blob/e4f946a5d6bc21ddeb8017f465f8b498728eb65f/app/robots/iCubParis01/hardware/motorControl/icub_left_leg.xml
* `iCubParis02` : https://github.com/robotology/icub-main/blob/e4f946a5d6bc21ddeb8017f465f8b498728eb65f/app/robots/iCubParis02/hardware/motorControl/icub_left_leg.xml
* `iCubHeidelberg01` : https://github.com/robotology/icub-main/blob/master/app/robots/iCubHeidelberg01/hardware/mechanicals/left_lower_leg-ems7-mec.xml

For more occurrences of this, check https://github.com/robotology/icub-main/search?p=1&q=MOT_speed_estimation&utf8=%E2%9C%93 .  

However on the `iCubGenova01` (traditionally one of most fine tuned iCub) the  `MOT_speed_estimation` and `MOT_accel_estimation` parameters are missing, and so their default value (5, check https://github.com/robotology/icub-main/blob/f7804c37e27fbe7b87cf37613f60de2753b6f4b5/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp#L1630 ) is used. The `JNT_speed_estimation` and `JNT_accel_estimation` are also missing, but their default value (5) matches the one fond in the configuration files of all the other iCubs. 

Do you think there is any reason for this? @pattacini @randaz81

pattacini, 2015-04-09:
I'm not involved in this respect, but I believe we should standardize those values for `iCubGenova01`.

lornat75, 2015-10-20:
@randaz81? this issue is very old we need to address and close it!

closed 2018-01-16 by julijenv