robotology/mesh-support #14: Report of on-site visit of 7/4-10/4 to Darmstadt

randaz81, 2015-04-13:
HW issues:
- Broken data cable in the CAN connector of the torso skin. I repaired the cable. I also found the connector tampered by a previous attempt of fixing it. Two cables were indeed swapped, and this is most likely the reason why I found CAN line 9 burned on the cfw2 board. I moved the torso skin on another free bus (CAN line 8).
- Unable to connect to legs skin. Most likely, a shortcut on one of the patches is responsible for the continuous reboot of skins connected to boards EMS10 and EMS11. I did several tests to identify the faulty skin patch, but unsuccessfully(*). This issues should be directly addressed by icub-support@iit.it

FW issues:
- EMS firmware now updated to build 71. MAIS boards updated to build 3.1.2 (MAISbug fix). Skin boards updated to build 2.10.17 (on the legs it was not possible to update the skin FW)
- fixed shoulder decoupling matrix issue on torque control (EMS boards) in build 72. Now the decoupling is correct. 
- Checked the rotor signs. It was found that regardless the different type of motors employed, the signs are the same of iCubGenov02 (arms only) and icubHeidelberg01 (torso/legs only), with the exception of torso J0 (inverted respect to Heidelberg01)
- fixed issue related to the wrong resolution of the optical encoders in the 2FOC firmware (all major joints). Darmstad robot employs kollmorgen motors with optical encoders which have a resolution of 1000TPR instead of 9000TPR of the IIT robot (equipped with Moog motors). Now fixed installing on the robot an ad-hoc firmware on the 2FOC boards (build 1.0.61)
- fixed torque matrix coupling issue on the torso (EMS5)

SW issues:
- general update of yarp, icub-main, codyco repositories. All the repos are now up-to-date. 
- Since the skin of the legs is not working, a locally modified icub_all.xml file is currently stored in the path ~/.local/share/yarp/robots but it is not committed to the main repo.
- installed an the tested robotInterface with the usb key containing the realtime kernel for the pc104. On the pc104, the current build-pc104 is the one containing the binaries compiled fot the RT kernel. The previous NON-RT build-pc104 was saved for backup. The pc104 pwd is now ‘live’.
- improved stability of the position PID controller of the arms, and performed basic tuning of the torque controller gains for the arms, legs and torso using the new matlab tool (torque-control-params-estimation). All changes committed to the main repo. Fine tuning still required.
- fixed an issue related to camera bayer pattern. (compiled yarp turning on the optional carrier)
- torso position gains seem to be too weak (to be verified in IIT): 1.5deg of static error on J0; J1 and J2 seem to generate not enough torque to move the torso when it is bended forward.
- shoulder position gains on j1 seem too aggressive (position tracking close to instability during direct position control). To be verified in IIT.
- damping coefficient in impedance control seems to generate high frequency vibrations. To be verified in IIT.
- improved demoForceImitation demo in order to test directPosition control on Darmstadt robot. Resulting tracking accuracy was very good both in stiff and in compliant mode (with/without gravity compensation)
- improved gravitycompensation module in order to be run also with WholeBodyDynamicsTree (and not only with legacy WBD)
- gravityCompensator module still needs an input port to allow the user to set a torque offset to the controlled joints. 

ADDITIONAL NOTES:
- Please remember to tighten (a lot!) the loose tendons of shoulders, wrists and torso.
- I recommend to keep the system aligned with our master branch (one update every 1-2 weeks)

(*) on EMS10
```
can1 (front)      can1 (back)         can2
      X                                                 ---> OK
                                        X               ---> OK
                         X                              ---> CANNOT BE TESTED
      X                  X                              ---> NOT WORKING
      X                                 X               ---> NOT WORKING
      X                  X              X               ---> NOT WORKING
```

lornat75, 2015-04-13:
@maggia80 @giorgiometta

iron76, 2015-04-22:
@traversaro and @DanielePucci can you please have a look at this list and figure out what we really need to be fixed/improved to try the CoDyCo experiments and Y1/Y2 validation scenarios in Darmstadt?

lornat75, 2015-10-20:
Closing, remaining issues will be addressed in #78.

Please try to open individual issues and avoid issues with multiple items, it is very difficult to track progress otherwise.

julijenv, 2015-11-03:
Hi @robertocalandra ,
what kind of test did u perform to understand and deduce that the problem with the skin is due to patch short circuit and not just drop of voltage on can line due to to little or flaky red or black power line wires which cant bring enough current on the line?
Are they all restarting ? or just part of them?
can you remove all the connectors (8 pins) from every MTB (skin) boards and check if the MTB work all fine when the patch skins ( the 8 pin connector) are not connected to the MTB board.
Let us know

robertocalandra, 2015-11-11:
Hi @julijenv,
I did not performed any test myself and I wouldn't know how to answer your questions. @randaz81 discovered the issue. Perhaps he can answer.
If necessary I can perform additional tests with your guidance.

francesco-romano, 2015-11-21:
Closing for #120

closed 2015-11-21 by francesco-romano