robotology/mesh-support #16: Jaw Motor Movement problem.

super-ste, 2015-04-17:
Katrin from Heriott-Watt requests:
we are getting a error from the jaw motor now and it is not moving after it first moved very very slowly.

super-ste, 2015-04-17:
Replied that iCub-Support Team is travelling until April 28th and requests logs in hte meanwhile.

BrutusTT, 2015-04-24:
Log file from robotInterface can be found here: http://pastebin.com/EJiXED86

super-ste, 2015-04-24:
Update per Ingo:
I attached the log file that shows the failing motor for face joint 0.

In the recent days I also have seen head joint 0 sometimes failing but it was not consistent enough for me to take a log. In contrast to the face joint 0 for the head joint 0 I was able to use the robotMotorGui to switch it idle and run again. Which worked for the head motor but not for the face one.

cc @maggia80 @randaz81 can you also have a look so we can gather ideas for Wednesay when we will be able to intervene?

super-ste, 2015-04-27:
@BrutusTT 
please note this line in the log you sent.

``` [ERROR]TalkingFace_Calibrator : unable to clear the hardware fault detected on joint  0  before starting the calibration procedure! ```

This tells you that robotInterface cannot clear an error prevoiusly encountered. 
Please provide a log after a clean start so we can assess the root couse of the problem.

cc @julijenv

BrutusTT, 2015-04-27:
Thats the problem we are trying to solve. It was a clean start. Everything was switched off and on again before starting the robotInterface.

We even tested it with enforcing the cleaning by putting the clearHwFaultBeforeCalibration flag into the proper config file. But it wasn't able to clear the HWFAULT state.

randaz81, 2015-04-27:
According to the line:
`[DEBUG]TalkingFace_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  -87.0769`
I suspect an hardware issue in the AEA encoder of the jaw. In this case, the joint will go in HWFAULT state immediately after the recovery attempt done by the calibrator.
To confirm this theory: can you open the robotMotorGui (after robotInterface calibration is complete) trying to manually move the jaw (gently), and checking the value of the encoder of joint 0 (face tab): correct values are in the range 0 +30deg (approx)

KatSol, 2015-04-29:
Hi, 
I think it is not moving at all, but maybe we can just Skype?

julijenv, 2015-07-16:
Hi @KatSol , we are closing some issues still open ion github, can you confirm that one was solved?
Lert me know

KatSol, 2015-07-16:
Dear Julien,
no it is not solved as my student mentioned in his last email:

Hi,

after Katrin fixed the encoder problem we are still having problems with the jaw joint.

I attached two log files of different runs. The first indicates trouble with some motors during the shutdown procedure whereas the second seems to be normal.

However, in both runs the jaw motor was just making some ticking noises if I tried to set a position. As further I set the position from the initial one as much more ticking I got. We were wondering if this might be due to a problem with the internal gears or something like this.

Thanks,
Ingo

--
---
Dipl.-Inf. Ingo Keller
PhD Student

School of Mathematical and Computer Sciences
EM Building, Room 2.22 (RoboticsLab)
Heriot-Watt University
Edinburgh, Scotland
EH14 4AS, United Kingdom
http://www.macs.hw.ac.uk/RoboticsLab/

Best regards,
Katrin

Dr. Katrin S. Lohan

Deputy Director of the Robotics Laboratory
Assistant Professor/Lecturer in Computer Science and Robotics

School of Mathematical and Computer Sciences
Heriot-Watt University
Edinburgh
K.Lohan@hw.ac.uk<mailto:K.Lohan@hw.ac.uk>

We invite research leaders and ambitious early career researchers to 
join us in leading and driving research in key inter-disciplinary themes. 
Please see www.hw.ac.uk/researchleaders for further information and how
to apply.

Heriot-Watt University is a Scottish charity
registered under charity number SC000278.

BrutusTT, 2015-07-17:
The logs can be found here http://pastebin.com/8cUYNUyP and here http://pastebin.com/8a1qa5Uz

pattacini, 2015-09-02:
ping @julijenv

julijenv, 2015-09-03:
Hi Edinburgh user of the icub talking head,
AFAI can tell, thre is no way to understand what's going wrong on your jaw motor from the log files sent, please get in contact with us with skype ID : icub-support, and let see what can be done to see whether the motor is dead or not?\
Can the motor spin when direct tension is applied to it? (0-5V max)
Let us know

lornat75, 2015-10-20:
Any update after the skype chat?

julijenv, 2015-10-29:
There are some problems with the AEA encoders of th jaw and the eyelids, we sent an AEA spare part to them, didnt receive any news from them.
Please @BrutusTT and @KatSol , let us know how goes the Head?

KatSol, 2015-10-29:
The jaw motor is still not working properly.
Best regards,
Katrin
Dr. Katrin S. Lohan

Deputy Director of the Robotics Laboratory
Assistant Professor/Lecturer in Computer Science and Robotics

School of Mathematical and Computer Sciences
Heriot-Watt University
Edinburgh
K.Lohan@hw.ac.uk<mailto:K.Lohan@hw.ac.uk>

julijenv, 2015-10-29:
Hi @KatSol  and @BrutusTT ,
Please give me more details on what's going wrong. Send a logfile, try to indentify where the issue could come from.
I'm everyday behind the skypeId: icub.support just ring me, whenever u can and let's try to solve this out.
best regards,

closed 2017-12-12 by julijenv