robotology/mesh-support #17: pc104 saturates CPU usage

lornat75, 2015-04-23:
@traversaro reported this issue:

on the iCubParis02 the robotInterface saturates the pc104, and this result in a lot of FT sensors dropped. Just to give an idea, the jitter can reach up to 200 milliseconds to a first quick check, on a robot running only the robotInterface and no cameras or any external module.

While this behavior should be properly debugged, it could make anyway sense to deploy the new realtime PC104 image even on iCubParis02.

lornat75, 2015-04-23:
Quoting @barbalberto on another issue, this may explain 

By chance, the feature of ROS topic introduced a bug for the analogWrapper in which the period was initialized to 0, therefore they were running at max speed draining a lot of CPU usage. This was introduced on monday 13/04 and on 21/04.

traversaro, 2015-04-23:
It was fixed on 21/04 ? Then it can definitly explain what we saw. 
@rlober when you have time you can try to update the yarp on pc104?

rlober, 2015-04-24:
Hey man. First thing monday I will try. Have a good weekend.

pattacini, 2015-09-02:
ping @rlober

rlober, 2015-09-02:
@pattacini 

I haven't noticed the behavior since updating but I will confirm this next time I am on the robot. Should be before the middle of next week.

pattacini, 2015-09-02:
cool, thanks @rlober

lornat75, 2015-10-20:
Closing for now.

closed 2015-10-20 by lornat75