robotology/mesh-support #19: Fingertip not working

massimoregoli, 2015-04-24:
Tactile sensors on the index finger of the iCubGenova02 right hand are not working properly. Occasionally, when the fingertip is touched on specific points, almost all taxels assume suddenly very high values. Waiting for a while, or touching again the fingertip (in somehow) everything goes back to normal.

lornat75, 2015-04-27:
@maggia80

maggia80, 2015-04-27:
I guess the fingertip is going to break soon. I would suggest a replacement as soon as there is another fingertip (code 1465.6) available.

@super-ste

pattacini, 2015-06-28:
This upcoming week a new board will be installed on one hand of `iCubGenova04`, which is specifically devoted to process fingertips data.

This solution should allow fixing the various problems experienced on the fingertips.

maggia80, 2015-06-29:
Today the board will be installed and we will test it.

lornat75, 2015-06-29:
@pattacini: the issue is referring to iCubGenova02 (aka purple robot). For iCubGenova04 see: #33

pattacini, 2015-06-29:
yep, it was just to add here in the stream the info about such a solution.

maggia80, 2015-06-29:
since I am color-blind I have confused the violet label with the blue one (that is why I have add purple in the label). Therefore I have to rectify: the board will be mount on the blue icub. @julijenv is checkink the purple one. I am going to buy the components to assembly other FTC boards, in case we do not managed to solve the issue in this robot and we have to put those boards on it as well. The FTC boards will be a standard for new robots.

massimoregoli, 2015-07-13:
Apparently now all fingertips are working fine on both hands, there is just an issue related to the pinkie of the left hand, sometimes when pressed it activates taxels on the other fingertips.

maggia80, 2015-08-27:
FTC boards are installed

massimoregoli, 2015-08-27:
With the boards, almost all fingertips seem to work properly, but there are still problems with the one of the right thumb, apparently it sends just random values.

bhigy, 2015-09-10:
Has the issue with the pinkie of the left hand been solved? I tried it this morning. Everything was working fine until I pressed the pinkie. It did activate some taxels from other finger tips and from the palm and then nothing was working anymore.

maggia80, 2015-09-15:
I did a change in the MTB3 firmware, since there was a bug. You should update the firmware of the board and test if it is ok. @massimoregoli

massimoregoli, 2015-09-22:
The firmware is up to date now, and the fingertips seem to work properly so far. I'll let you know about any issues it may occur

bhigy, 2015-09-22:
There is no freeze anymore but we still have weird activations when the pinky is touched. Also, after several trials with the pinky, all the taxels of the hand showed activations and stayed like this.

lornat75, 2015-10-20:
Any update on this?

massimoregoli, 2015-10-20:
All fingertips seem to work fine now, even the right pinky.

closed 2015-10-20 by massimoregoli