robotology/mesh-support #20: Report of on-site visit to iCubSheffield01

pattacini, 2015-04-25:
This is a list of things deserving attention/improvement which we spotted
working with **iCubSheffield01** during the [WYSIWYD](https://github.com/robotology/wysiwyd)
integration week held in Sheffield from 20th to 24th of April 2015.

###Mechanics
1. During `ARE` fingers calibration, the left thumb did not behave as expected, while
the right thumb did calibrate nicely. The **abduction of the left thumb** (`joint 8`)
does not move via the `Gui`, whereas by manually rotating the corresponding screw
we managed to make it move.
2. The right arm collides with the torso cover during the first calibration. Apparently,
this might be due to a tiny deformation of the cover of the torso in the correspondence
of the collision.

###robotInterface
1. `robotInterface` **never run after the first motor start-up**. No matter how long
we waited, some boards were not being heard of. Further, we had the same problem after
a **fault button** event followed by a shutting down of `robotInterface`. In both
the above situations, a second motor power-cycle made `robotInterface` start correctly.

###Low-level Software
1. Frequent **HW faults** have been triggered throughout the week, in the neck and
in the wrist.
2. The kinesthetic teaching of `ARE` did fail. The teaching went on in admittance
mode and, looking into the code, this can happen when the the function [`setTorque`](....)
returns false.
3. See robotology/icub-main/#140

###icubsrv configuration
The **icubsrv** laptop has been shipped back with the following wrong settings:

1. The xml file used to launch the **calibrated cameras** was not the updated version;
as result, the left and right image streams were swapped. This was quite a nasty
bug to find, causing lots of problem with stereo vision. **`Fixed`**.
2. Somehow, there were two **bash files** devoted to contain iCub related env variables:
`.bash_iCubrc` and `.bashrc_iCub`, with mixed configuration. The standard file must
be the former, while `.bashrc` was pointing to the latter, messing up the overall system.
**`Fixed`**.

###Yarp
1. See robotology/wysiwyd/#12

That's the commiserations over, now the good news:

1. Thanks to the **new pronosupination**, we have been experiencing a very improved
matching between the kinematic predictions and the stereo-vision observation in the
`depth2kin` experiment.
2. We managed to successfully **integrate the object-recognition** layer of
[IOL](https://github.com/robotology/iol) into the `wysiwyd` system!

/cc @lukeboorman @davidetome @super-ste @marcoaccame @randaz81 @lornat75

lornat75, 2015-04-27:
@mbrunettini @davidetome: check the issue with the configuration of the laptop:

-> Somehow, there were two bash files devoted to contain iCub related env variables: .bash_iCubrc and .bashrc_iCub, with mixed configuration. The standard file must be the former, while .bashrc was pointing to the latter, messing up the overall system.

Just to make sure we do not end up with the same problem in the future.

pattacini, 2015-04-28:
@randaz81 torque mode for `iCubSheffield01` was purposely disabled, hence `ARE` kinesthetic teaching was clearly not available.

lukeboorman, 2015-05-01:
We are getting more frequent HW faults on neck joint 0 and I assume this then puts joint 1 into idle...

After startup (viewed with robotMotorGui) and during testing using the ball demo. I can send the robotInterface log file.

Could this be hardware?

mbrunettini, 2015-05-04:
Hi,
the installation instructions here:
http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions
reports only the correct .bashrc_iCub file, as Lorenzo told.
@davidetome : do you confirm you followed the above instructions? In case you’re using an HD image, can you chack it?

Thanks

Matteo

closed 2015-11-27 by super-ste