robotology/mesh-support #21: Errors when starting robotInterface in log

lornat75, 2015-04-30:
Posted originally by @Tobias-Fischer in https://github.com/robotology/icub-main/issues/143. It requires instructions for fixing configuration files @marcoaccame @randaz81.

Hi all,
we just recently updated yarp, icub-main and icub-firmware-shared on pc104 and our main machine. As we also started using yarplogger, we noticed several errors when launching robotInterface.

The error is as follows: 
```
233		6.002969		ERROR		TORQUE_PIDS: 'controlUnits' param missing. Assuming machine_units. Please fix your configuration file.
```

The same error is repeated for in total 12 times. Going through the recent commits, we noticed that this has something to do with the ETH robots (e.g. https://github.com/robotology/icub-main/commit/1a38740c51c76633b4098a47419de951629c1878). However, we have an CAN-BUS based robot.

The robot however starts nicely, and we don't have any issues running any other modules.

Would be great if this could be fixed :+1: 

Please let me know if you need the complete logfile.

Thanks a lot,
Tobias

randaz81, 2015-04-30:
Under investigation. @Tobias-Fischer Can you tell me the name of your robot?

Tobias-Fischer, 2015-04-30:
Hi @randaz81,
our robot is iCubLondon01.
As it seems to be issues with the config files, could we also try sorting out some of our warnings :+1:?

```
0 0.000002 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/./icub_all.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
 
1 0.001916 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/./icub_all.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
 
2 0.001974 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/./icub_all.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
```

Can I change `app/robots/iCubLondon01/icub_all.xml` from
```
<robot name="icub">
```
to
```
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">

<robot name="iCubLondon01" build="1" portprefix="icub">
```
?

Some more warnings:
```
3 0.002010 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/cartesian/left_arm_cartesian.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
 
4 0.002075 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
 
5 0.002112 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/cartesian/left_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
 
6 0.002202 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int].
```
The previous warning repeats for `cartesian/right_arm_cartesian.xml`, `wrappers/motorControl/left_arm_mc_wrapper.xml`, `wrappers/motorControl/right_arm_mc_wrapper.xml`, `wrappers/motorControl/head_mc_wrapper.xml`, `wrappers/motorControl/torso_mc_wrapper.xml`, `wrappers/VFT/left_arm_VFT_wrapper.xml`, `wrappers/VFT/right_arm_VFT_wrapper.xml`, `wrappers/VFT/torso_VFT_wrapper.xml`, `hardware/VFT/left_arm_virtual_strain.xml`, `hardware/VFT/right_arm_virtual_strain.xml`, `hardware/VFT/torso_virtual_strain.xml`, `wrappers/FT/left_arm_FT_wrapper.xml`, `wrappers/FT/right_arm_FT_wrapper.xml`, `wrappers/MAIS/left_hand_mais_wrapper.xml`, `wrappers/MAIS/left_hand_mais_wrapper.xml`, `wrappers/MAIS/right_hand_mais_wrapper.xml`, `hardware/FT/left_arm_strain.xml`, `hardware/FT/right_arm_strain.xml`, `hardware/MAIS/left_hand_mais.xml`, `hardware/MAIS/right_hand_mais.xml`, `wrappers/skin/left_arm_skin_wrapper.xml`, `wrappers/skin/right_arm_skin_wrapper.xml`, `wrappers/skin/torso_skin_wrapper.xml`, `hardware/inertial.xml`.

I assume I can just add:
```
<!DOCTYPE devices PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">
```
To all these files?
Also, change
```
<devices>
```

to
```
<devices robot="iCubLondon01" build="1">
```
?

And the last set of warnings:
```
27 0.020401 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/hardware/motorControl/icub_left_arm.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
 
28 0.020444 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/hardware/motorControl/icub_left_arm.xml at line 4 . Trying to import a file for the wrong robot. Found iCubLondon01 instead of icub 
 
29 0.020487 4 Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLondon01/hardware/motorControl/icub_left_arm.xml at line 4 . Import a file for a different robot build. Found 1 instead of 0 
```
The previous warning repeats for `hardware/motorControl/icub_left_hand.xml`, `hardware/motorControl/icub_right_arm.xml`, `hardware/motorControl/icub_right_hand.xml`, `hardware/motorControl/icub_head.xml`, `hardware/motorControl/icub_torso.xml`, `hardware/skin/left_arm.xml`, `hardware/skin/left_arm-cfw2_can7-skinSpec.xml`, `hardware/skin/right_arm.xml`, `hardware/skin/right_arm-cfw2_can8-skinSpec.xml`, `hardware/skin/torso.xml`, `hardware/skin/torso-cfw2_can9-skinSpec.xml`, `calibrators/head_calib.xml`, `calibrators/torso_calib.xml`, `calibrators/left_arm_calib.xml`, `calibrators/right_arm_calib.xml`, `calibrators/left_hand_calib.xml`, `calibrators/right_hand_calib.xml`.

Again, should I add
```
<!DOCTYPE devices PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">
```
in these files? Don't know how why it complains about `Found iCubLondon01 instead of icub` and `Import a file for a different robot build. Found 1 instead of 0` though. Will that be fixed if I change `icub_all.xml`?

Please let me know if my assumptions are correct, and I can change the files if you want.

However, I have no idea how to fix the error in the initial post.

Best,
Tobias

Tobias-Fischer, 2015-04-30:
Actually there are some more warnings related to my initial post:
```
230 6.002854 4 TORQUE_PIDS: 'kbemf' param missing
 
231 6.002891 4 TORQUE_PIDS: 'ktau' param missing
 
232 6.002929 4 TORQUE_PIDS: 'filterType' param missing
```

Many thanks in advance!

Tobias-Fischer, 2015-04-30:
And the last warning(s) ....

```
256 4.506002 4  ControlBoardWrapper2 device:
 
258 4.506122 0 ************************************************************************************
 
260 4.506193 0 * ControlBoardWrapper2 'name' parameter for port name does not follow convention,  *
 
262 4.506264 0 * it MUST start with a leading '/' since it is used as the full prefix port name   *
 
264 4.506334 0 *     name:    full port prefix name with leading '/',  e.g.  /robotName/part/     *
 
265 4.506369 0 * A temporary automatic fix will be done for you, but please fix your config file  *
 
266 4.506404 0 ************************************************************************************ 
```

Tobias-Fischer, 2015-04-30:
Hi,
I just created a new pull request to fix the warnings related to the DOCTYPE / robot name etc.

However, the following still needs fixes:
1) Error from initial post
2) Warnings related to TORQUE_PIDS
3) Warnings related to ControlBoardWrapper2

Thanks, Tobias

randaz81, 2015-04-30:
fixed by commits: https://github.com/robotology/icub-main/commit/4dbf067cfd7fc33b54ddc9f87b40981a015bfc29 and https://github.com/robotology/icub-main/commit/a39f54c4e00a80aafa49beb2b2db7350a1fb543e

closed 2015-04-30 by randaz81

Tobias-Fischer, 2015-05-01:
Works like a charm. Thank you very much!