robotology/mesh-support #25: Wrist yaw joint 6 calibration

rueckert, 2015-05-15:
Hi,

we had some slack in the wrist yaw joint and fastened the tendons using the tensioners. Now after starting the robot the wrist yaw joint is not centered any more (it is at the positive extreme). We have to calibrate joint 6. 

I guess we have to change the 'Zeros' entry in the right_lower_arm-ems4-mec.xml file (Thanks to @randaz81 how pointed me into that direction). 
I have two questions:

1. Is this the correct location of the config file: $ICUB_DIR/share/iCub/robots/iCubDarmstadt01/hardware/mechanicals

2. How do I find out the new 'Zeros' value. Currently the value is set to -176.95. Is the unit degrees? 
I can manually correct the wrist yaw joint using the robotMotorGui. When I set joint 6 to -20 degrees it moves torwards the center but is still off by 2-5 degrees. Does this mean that I have to set the 'Zeros' value to -196 or -200.

Thank you very much for your help. 
Elmar

lornat75, 2015-05-19:
The location of the file is correct 1). You just have to make sure you do not have a copy in ~/.local/share/yarp which will take precedence over it.

For 2). Usually the 'Zeros' parameter is appropriate for small values (a few degrees). If you experience an offset >20 degrees it means the calibration is wrong. Maybe this is due to the fastening of the tendon. In this case you have change the calibration parameter.

@julijenv can you help with this?

rueckert, 2015-07-10:
@julijenv 
Thank's for helping us. Here is the dropbox link to the log files and videos I recorded during the calibration procedure: https://www.dropbox.com/sh/59dnnj5yr0j7p7v/AAA1WcU65BrLOs26uqHRkWBJa?dl=0

From the motor gui erverything looks ok, just that the right wrist has the offest. Let me know if you need aditional information.

julijenv, 2015-07-10:
Hi iCubDarmstadt01 users,
Hopefully it is quite nothing to fear, but i think one of your motor is not working at all, the reason is , telling that by experience, the connector going to the control board is disconnected or one of the wire inside the connector is broken. The 2 motors coupled for the movement of the wrist are the tagged 2M1 and 2M2, you should find them connected to the board 2B2 underneath the cover of the upper arm. (proceed with care to remove this cover).
Afterwards try to find out the 2 motors quoted earlier, look also if the power supply cord is well connected(you may need to remove 4 screws olding the support of the stack of boards to check that). For me it is either motor connectors related or power line.
[image]
in the picture above you will be able to see 3 different kind of arrows, the blue ones are where are connected the motors 2M1 and 2M2 and the white one it is the power supply connector.
Check those connectors thoroughly try to pull out every single wire inside to check the eventual weak wires not making contact. (do you have the crimper and associated crimps to operate in case of detection of broken wires).
here is the file on which youll find the tools needed for basic and more advances repair (i already colored blue the lines for the crimper and crimps you may need).

[image]

Let 's proceed with cautious, Until further investigation from your side let's hope that one of the motor or the control board are not dead - worse case scenario-.
Let us know your findings.

rueckert, 2015-07-13:
Thank's @julijenv for your reply.

I carefully removed the upper arm cover and had a look at the connectors and cables. I could not find any loose cable or connector: 
  
[image]
[image]

After putting joint 6 in idle mode I recorded the extreme positions of the encoders:
right wrist: -5.3, -42,-81.5 in degrees (max to the right, center, max to the left)
left wrist:  45.8, 0 ,-24

We had some weeks ago problems with the isulation tape on the palm. I had to renew it. The conductor of the encoder ripped-off the glue tape and the palm itself is conductive. Maybe this shortcut or some leakage current could have caused problems in the motor boards? 

Elmar

julijenv, 2015-07-13:
Hi Elmar,
as you can notice the range of both wrist is the same except the shift coming the center and due to the non calibration of the wrist. You've got encoder values for either joint 5 and 6 that are coupled which means that the encoder AEA should work properly. The position of the wrist in the center part depends actually just from one motor. I guess this motor has an issue as i was explaining last time. Is the motor intern for the wrist moving, after calibration? I guess not.
Did you unplug the 2 motors and check the integrity of the wires - did you unplug every single crimp from the connector and check if they were properly still holding the wire. did you make a continuity test in between every single crimp and 5 cm away on the flatstrip wires of the motor. If you detect discontinuity it probably come from the crimp. remove it and recrimp it, and test the robot again.

rueckert, 2015-07-14:
Hi,

@julijenv, please check the movies I put in the dropbox folder.

I opend the cover of the lower arm and found that both motors (2M1 and 2M2) are moving, see the latest movies in https://www.dropbox.com/sh/59dnnj5yr0j7p7v/AAA1WcU65BrLOs26uqHRkWBJa?dl=0
I played with joint 5 in the yarpmotorgui to inspect the correct function of the motors. The joint nicely reached its limits. However, joint 6 is not working. Do you still believe that is a connector or cable issue?

The third movie shows that the wrist is shaking after the calibration finished. 

Elmar

julijenv, 2015-07-15:
Hi guys,
from the video i can not say that the motor is not spinning but i dont know if it is from its own will ( because actually joint 5 is the only joint coupled - but joint 6 use just the 2M2 motor to spin - so i guess that when you ask joint 5 to move nothing tell me that 2M1 just help the other motor to spin but when joint 6 need to move just 2M2 is concerned and at this moment by itself is not able to move.
So the only way to make sure the motor works is to unplug 2M2 and power it with an external power supply (12V max but with from 2 to 5V it should move).
[image]
the 2 last wires are the supply line. Make sure the motor spin.
please let's test it. It would be simpler.
Let me know

rueckert, 2015-07-15:
Hi,

@julijenv  thank's for the hind, I already though about testing the dc motors directly. 

Both motors, 2M1 and 2M2 are working and the wrist moves when supplying the motors with about 4 Volts. 

I still don't believe that its a motor or connector problem. One can see that both motors are actively driven when setting joint 5 to some desired joint position. The cables and the connectors are ok. I unpluged them and checked the cables. 

Could there be a sign bug in the firmware. I looks like that the wrist moves torwards the wrong direction when calibrating. 

Best, Elmar

julijenv, 2015-07-15:
Hi, so finally we know tha tthe motor is ok. So obviously now we could think about problem with config files, but at this i'd like to know if you havent touched anything in between the day it was working properly and the day it wasnt ??? because the robot got out from us working and the only way that is not working anymore now - if it isnt a mechanical problem I mean -  is that you went to some files and changed things that make the robot behave wrongly. have you rewrite some files and if yes, which one? send me the copy of those files as well as the respective other arm-hand files( I mean the problem is on the right arm so give me also the files from  the left arm in order to compare them).
In any case I went looking inside the logfilefail you sent me last time, and i see that the repository that found config files are in /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01 - just zip it and send it to me i'll have a look at it.
thx in advance,
JJ

rueckert, 2015-07-15:
@julijenv 

The used calib files are equal to the ones in the git repo. Checked with diff. Can you check if these two files are correct? 
icub-main/app/robots/iCubDarmstadt01/hardware/motorControl/right_lower_arm-ems4-mc.xml 
and left_lower_arm-ems2-mc.xml 

The icub_all.xml file used (in build-pc104/share) is equal to the one at icub-main/app/robots/iCubDarmstadt01

We installed updates of the firmeware, icub-main, yarp at:

[1] Mo 16. Feb 16:43:24 CET 2015
[2] Di 7. Apr 16:08:26 CEST 2015
[3-...] Thereafter Marco R. updated the configuration files several times and fixed some bugs when he was here in Darmstadt. Its an overstatement that the icub was fully operating when shipped to us, ask Marco ;) 

I can only guess, but probably the wrist was working before update [2]. 

Elmar

lornat75, 2015-10-20:
Any update on this? Can we close it and address in #78?

rueckert, 2015-10-21:
@julijenv  and @lornat75 

thank you for your support again. The wrist issue is still open. Here is a short summary of the situation:
- After tightening the tendons of 2M1 and 2M2 the wrist yaw joint had an offset of about 20 degrees. 

- the wrist calibration procedure fails during the boot process. In the motor gui it says hardware fault. CHECK the rightWristProblemStartup.log file in https://www.dropbox.com/sh/59dnnj5yr0j7p7v/AAA1WcU65BrLOs26uqHRkWBJa?dl=0 from TODAY

- the motors work correctly (tested by directly applying 3Volts to them) 

- the used config files are identical to the ones in the git repo (e.g., icub-main/app/robots/iCubDarmstadt01/hardware/motorControl/right_lower_arm-ems4-mc.xml)

- the conductive palm causes a short-cut when tightening the two tiny screws (the screws connect the palm with the tendon driven metal cylinder in the wrist). 

- the two metal pins in metal cylinder in the wrist do not stick out any more  of the cylinder. This is probably only a mechanical issue.

- after booting, the wrist can be set to idle model (if the two screws were loosen), manually moved to the center position, and put to position mode again. Thereafter, the wrist yaw joint (id 5 in the motorgui) can be actuated. Also the joint angle sensors of all wrist joints operate correctly. 

Elmar

super-ste, 2015-10-22:
Hi @rueckert 
the fault could be due to a contact between the palm (it's made of carbon fiber i.e. conductive) and the AEA board undelying (see pic below)

[image]

Please remove hte palm and apply a thin layer of insulating material on the palm in the area circled in red.

Normally there should be some kapton tape there. Can you confirm there is some still? 
Can you update a pic?

Let us know

rueckert, 2015-10-22:
@super-ste 
Here are images of the palm and the conductor. The conductor causes the short cut. When tightening the screews, it is pressed into the carbon palm (almost one millimeter deep). We already added 2 layers of plastic tape in July and today to fix that. No success. Such tricks should not be a permanent solution?  

https://www.dropbox.com/s/jkths0sy92mpiq4/IMG_3494.JPG?dl=0
https://www.dropbox.com/s/wrgyr9pimigiy81/IMG_3495.JPG?dl=0
https://www.dropbox.com/s/j9gpqi1oyvp02rd/IMG_3496.JPG?dl=0

julijenv, 2015-10-29:
@rueckert ,
Plese send me another log file for the calibration failure of the wrist un fortunately the one on the link dropbox mentionned previuously doesnt show any proble with the Hardware fault, I need the entire (from start to ending phase log file).
The pictures of the wrist you ve done doesnt show any sign of problem, plese come on skype and let's ahve a look.
skye ID: icub.support
best reagards,

julijenv, 2015-10-30:
Hi @rueckert ,  the capacitor unfortunately touch the palm so please add any kind of insulating tape in between the palm and the capacitor ( in your case the insulating tape got burnt above the capacitor).
LEt us know

rueckert, 2015-10-30:
@julijenv 

I renewed the isulating tape which did the job. We could boot the icub without HW fault. Thank you. 

Now, we can focus on the wrist calibration issue. Here is the link [1] to a log file created 2 minutes ago with yarplogger. The file lists two errors:
793		42,745833		ERROR		Right_Arm_Calibrator : checkGoneToZeroThreshold: joint  5  Timeout while going to zero! 
807		43,752119		ERROR		Right_Arm_Calibrator : set 3 : some axis got timeout while reaching zero position... disabling this set of axes
 
[1] https://www.dropbox.com/s/lf9n2767jwy5434/ynachm.rprunlog_30_10_2015_03_18_41.log?dl=0

It seems that the offest we have (approx. 20 degrees) is too much for the calibration proceedure. Is there a way to manuall correct the offset by making first the tendon loose and then rotating the wrist?

julijenv, 2015-11-04:
Hi @rueckert,
ok here how u gonna proceed,
step1: open the file general.xml and change FALSE onto TRUE for the 2 first lines (skipCalibration and useRawEncoderData).
step2 : TURN ON the motors (or do power recycle if they were already ON
step3 : push the red button (FAULT the robot - it shall not start )
step4 : launch robotInterface (from a pc104 window shell - you will be able to see what is going on) and click ojn the enter button for about 20 times (logicqally just the one requested from the window - but it is better to do more.
step5 : open yarpmotorgui and read the value of the 2 motrs of the wrist when it is hold in the good straight position. (as it should be positionned after the calibration)
sstep6: send us  reply with the values for joint 5 and 6 and we wiill indicate you how to proceed later on.;
C U

rueckert, 2015-11-04:
@julijenv 
Hi,

we followed your instructions and here are the raw encoder values.
joint 5: 1907
joint 6: -2850

julijenv, 2015-11-06:
Hi @rueckert ,
Please undo all the changes you made for trying to get the wrist bask on place by changing the Delta.
First we will make a change in your local repository in order to see if the changes we gonna make are effective, if yes you will have to apply them in the iCubDamrstadt01 repository inside the icub-main github repository.
So please change in the files:
.local/share/yarp/robots/iCubDarmstadt01/calibrators/right-arm.xml
the line:
<param name="calibration3"> 20079.5 64911.4 63999.4 55519.8 0.0 2002.5 1952.5 0.0 0.0 6000.0 8000.0 6000.0 -7000.0 6000.0 7000.0 14000.0 </param>
by
<param name="calibration3"> 20079.5 64911.4 63999.4 55519.8 0.0 3954.5 802.5 0.0 0.0 6000.0 8000.0 6000.0 -7000.0 6000.0 7000.0 14000.0 </param>

relaunch robotInterface and let me know

serena-ivaldi, 2015-11-12:
quick note for @rueckert @robertocalandra : after chatting with @julijenv and doing a remote session, the suggested line above does not work (caused almost breaking the wrist yaw!).
I took new raw encoder values and transmitted to julien - tomorrow we should have the new line.

closed 2015-11-16 by serena-ivaldi