robotology/mesh-support #26: fixing issues generated by recent demos

reafrancesco, 2015-05-26:
After recent demos, the iCubGenova03 (red) shows several issues. I try to list them here and ask @reafrancesco to supervise the restoration:
- index finger (proximal) of right arm is malfunctioning.
- thumb finger (abduction) of the left arm is malfunctiong.
- the ocular bulbs of both the eyes touch the cover. This might cause the issue #18. @drdanz 
- feet cover of the robot are still in fast prototyping. The material is delicate and gets broken easily. We should restore the feet cover as better as we can while we wait for the new ones.
- left camera electronic board has got a bad soldering clearly showing through a brighter image.

Please let me know.

drdanz, 2015-05-26:
Adding a couple of issues:

  * A couple of days after we precisely calibrated the eyes using the screws inside, the calibration was lost. Apparently nobody touched it, so perhaps it changed during the calibration, when the eyes move at end course.
* If I remember correctly, when @julijenv mounted the back cover of the head, he said that it was the wrong cover, or that there was some other issue.

super-ste, 2015-11-20:
Hi @reafrancesco are these issues still there? Shall we give @julijenv heads up?

reafrancesco, 2015-11-20:
Yeah, go ahead. Things are now working OK. Cheers

closed 2015-11-20 by super-ste