robotology/mesh-support #27: Problems iCubSheffield

lukeboorman, 2015-06-02:
I spoke to Julien last night and we have got further diagnosis on some of the problems...

Here is the list:

1. Thumb oscillation on left hand joint 9 (proximal?) 'probably' due to hall effect sensor issue... We discussed sending the arm back to IIT, is this ok? and what time scale is need also what do I need transportation / packing..... I can remove it....

We checked sensor readings and they seemed linear for joints 8 to 10 and had the full range, Also thumb moved ok when manually turning the three motors (middle motor near elbow, motor left below wrist, and motor on back of hand).

2. Left arm not starting: The left arm produces a hardware fault in joint 3 (Elbow) if the arms starts from outside the calibration location (worse from off position). See log files (failed starts 1 to 7 and 8 started successfully by holding arm in calibration position). Also images from robotMotorGui in google drive folder.

3. Segmentation faults and general errors on robotInerface during startup (we wait ages before running robotInterface after initial starting / cycling motor power switch). Lots of problems see included error logs (Starts will seg faults and cant see board 5...). I can do a firmware update if needed.

4. Very occasional neck faults, can include robotMotorGui images if needed, these occur more at startup but also when using ikingazectrl...

5. Can we run compliance?

Message me if you cannot access the link....

https://drive.google.com/folderview?id=0B8b3zfboj7AIfm5SSnA2M3k1bzdTWGdUVGY4NnFLcjVlTGVnSGxsUWdndVlmNWhXd1VxZFk&usp=sharing

pattacini, 2015-06-02:
Hi @lukeboorman for the point 5 I'm afraid the answer is still `no`
see https://github.com/robotology/icub-support/issues/20#issuecomment-96977201.

Proper tuning of the `torque-PID` needs to be carried out beforehand.

/cc @randaz81

lukeboorman, 2015-06-02:
@pattacini No worries

closed 2018-01-16 by julijenv