robotology/mesh-support #28: skin problems on left upper arm

matejhof, 2015-06-04:
Hi,
The skin on left upper arm had problems already before the blackie went to Shanghai. One triangle seemed to be often “stuck” - permanently activated.  I think the one on the picture below.
[image]
Phantom activations were appearing also on other regions of the upper arm. Can you please have a look?

Thanks!
Matej

matejhof, 2015-06-29:
The problem still persists. From what I understood from @julijenv , to fix it may require replacing the PCB.

maggia80, 2015-06-29:
Yes, the only way to solve it, is to replace the pcb.

super-ste, 2016-01-04:
If I am not mistaken the skin is being replaced right now. @Julijenv ,can you confirm and provide possible restoration time?

closed 2018-01-16 by julijenv