robotology/mesh-support #29: iCubLondon01 skin fingertips hardware failures

martz21, 2015-06-05:
Hi!

The fingertips tactile sensors of our iCub are not working properly: they are functioning only sometimes, apparently at random, and in any case, in the few times they are up, a big pressure must be applied in order to have some useful signals out of them. 
It was suggested that this issue sounds like probably broken pcb.
Do you have any suggestions on how to proceed?

Thanks,
Martina

martz21, 2015-06-08:
Please find attached the pictures required!
I hope this helps

LEFT HAND
[image]
[image]
[image]

RIGHT HAND
[image]
[image]
[image]

Tobias-Fischer, 2015-06-15:
Hi,
I just wanted to follow up whether you have an idea what the problem might be?

Thanks,
Tobias

martz21, 2015-06-18:
Hi!

We tried to unplug and plug the boards on the knuckles, however nothing changed: the fingertips are only working irregularly. 
How sensitive are the fingertips supposed to be? If we try to only slide on surfaces, it looks like they never get activated (the skinGui does not show any activity).

Any suggestion you could give will be greatly appreciated. Thanks in advance.
M.

maggia80, 2015-06-29:
We will check the fingertips once we will have the arm here in IIT

martz21, 2015-07-07:
Yesterday I tried the tactile sensors: in the first trial the fingertips were giving some signals when a certain pressure was applied, but after this first trial, they were not giving any signal, like they didn't get activated anymore. Did you experience this behaviour before?

maggia80, 2015-07-07:
We have found an issue in the flexible pcb of the fingertips. We discovered broken traces in some fingertips due to fabrication process. And the bad thing is that they breaks in such a way they still work in some conditions. I can not say if the problem you have experienced is due to this issue or a broken wire , but I can tell you that as soon as we will have your arm we can check on it the status of the fingertips.

martz21, 2015-07-14:
Is there a way we can check here if the problem is related to the pcb or to some connections?

Tobias-Fischer, 2015-07-14:
The reason @martz21 asks is, that https://github.com/robotology/icub-support/issues/31 is now resolved, and therefore we do not necessarily need to ship the arm back. So maybe it would be possible to replace the PCB's you are referring to ourselves? If not, which part of the iCub do you suggest to ship back? The forearm?

Best, Tobias

julijenv, 2015-07-14:
Hi iCubLondon01 users,
As it has been said previously by @maggia80 , some trace connectedd directly on the wire pads of the fingertips pcb are really easily breakable.
left fingertip pcbs
[image]
right fingertip pcbs
[image]
To try to restore the pcbs, you will need a very powerful magnifying lens and a sharp knife tip or cutter.
remove a bit of insulation copper (green) above the thin line of power (K pad) on the pcb
and put some soldering paste on it to gete a better contact.
left pad
[image]
right_pad
[image]
Let us know, if you make it.

martz21, 2015-07-15:
Hi! Thanks for the previous message.
Are the boards you mention the small ones (5 in total) on the knuckles?
Can they be taken off completely from the hand to make this work?
Thanks in advance for your help.

julijenv, 2015-07-15:
No, the pcb on the pictures are the ones on the fingertips itselves- just remove the nail and you should see it.
best of luck if you try it out!

martz21, 2015-07-17:
Hi! We had a look at the boards as you suggested but we are not sure whether the boards are the same (they seem to look different); also, we cannot tell if the problem is the same that you showed in your pictures. I upload below the pictures we took from one of our fingertip board.
Please let us know if the pictures give you any hint regarding the problem.
Thank you in advance!

[image]
[image]
[image]
[image]
[image]
[image]

maggia80, 2015-07-21:
Hi,

The circuit is the same. You do not see the break in the trace, it is weird but where is the red circle in the picture below, there could be the break. You should as Julien wrote you remove the green solder mask and solder over the trace. If just a couple of fingertips have this issue none of them are working. Because they share the same bus. 

[image]

Tobias-Fischer, 2015-09-24:
Hi all,
Through some mystery, our fingertips seem to work fine nowadays. I have the feeling that changing the `yarpserver` to run on our desktop machine rather than `ikart` seemed to have improved the detection of touches on the fingertip (for some weird reason, maybe it just randomly works better). Do you think we can close this issue, @martz21?

Best, Tobias

maggia80, 2015-09-24:
I dont think so. However I have recently updated the firmware of the boards reading the fingertip. You should install it thought the canLoadere application. The parameters are cfw002 CAN NET 7 and 8 and the firmware is in the icub-firmware-build git repo, skin.hex

Tobias-Fischer, 2015-09-28:
Hi @martz21,
we just tested the fingertips again and they are working fine now. Can you please close this issue?

Thanks, Tobi

closed 2015-12-15 by super-ste