robotology/mesh-support #30: ARE Shaking iCubBarcelona01

jypuigbo, 2015-06-11:
Hi!

We've been trying to run the different modules on the iCub and we are finding that turning on ARE makes the robot shake, only its arms. 

We've checked the torques and they are in the range -1.7/1.9 Nm, usually, before turning on ARE, while it goes up to around 4.0 Nm when turning it on. 

The only modules running when this happens are: 
- robotInterface (pc104)
- iKinCartesianSolver --part right_arm (ikart)
- iKinCartesianSolver --part left_arm (ikart)
- iKinGazeCtrl (ikart)
- wholeBodyDynamics --autoconnect --no_legs (ikart)
- gravityCompensator --no_legs (ikart)

- actionsRenderingEngine (icub-debian1) --> and here it begins to shake

The installation is new and nothing is running from old config files. The ones used are those in build/share. No other package is running when this happens. Find attached the videos of the robot and icubGui. And also the log of ARE till I close it (just when the shaking begins). 

Thanks, 

Jordi

https://www.youtube.com/watch?v=ajDoTO1n7PU&authuser=0
https://www.youtube.com/watch?v=j8qSxysY1GQ&authuser=0

```bash
icub@icub-debian1:~/Robots/specs_apps$ actionsRenderingEngine 
||| clearing context
||| adding context [actionsRenderingEngine]
||| configuring
||| no policy found
||| default config file specified as config.ini
||| checking [/home/icub/Robots/specs_apps/config.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/home/icub/Robots/yarp/build-cluster/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/icub/Robots/icub-main/build-cluster/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/icub/Robots/iCubContrib/share/ICUBcontrib/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/icub/Robots/wysiwyd/main/build-cluster/share/wysiwyd/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/icub/Robots/yarp/build-cluster/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/Robots/icub-main/build-cluster/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/Robots/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/Robots/wysiwyd/main/build-cluster/share/wysiwyd/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.config/yarp/contexts/actionsRenderingEngine] (context YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/contexts/actionsRenderingEngine] (context YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/contexts/actionsRenderingEngine
||| checking [/etc/yarp/contexts/actionsRenderingEngine] (context YARP_CONFIG_DIRS)
||| checking [/home/icub/Robots/yarp/build-cluster/share/yarp/contexts/actionsRenderingEngine] (context YARP_DATA_DIRS)
||| checking [/home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine] (context YARP_DATA_DIRS)
||| found /home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine
||| checking [/home/icub/Robots/iCubContrib/share/ICUBcontrib/contexts/actionsRenderingEngine] (context YARP_DATA_DIRS)
||| checking [/home/icub/Robots/wysiwyd/main/build-cluster/share/wysiwyd/contexts/actionsRenderingEngine] (context YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/contexts/actionsRenderingEngine/config.ini] (context)
||| checking [/home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine/config.ini] (context)
||| found /home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine/config.ini
yarp: Port /actionsRenderingEngine/OPC:io active at tcp://10.0.0.102:20246
yarp: Port /actionsRenderingEngine/disparity:io active at tcp://10.0.0.102:20247
yarp: Port /actionsRenderingEngine/wbd:rpc active at tcp://10.0.0.102:20248
yarp: Port /actionsRenderingEngine/left/wrench:o active at tcp://10.0.0.102:20249
yarp: Port /actionsRenderingEngine/right/wrench:o active at tcp://10.0.0.102:20250
yarp: Port /actionsRenderingEngine/head/rpc:o active at tcp://10.0.0.102:20251
yarp: Port /actionsRenderingEngine/head/command:o active at tcp://10.0.0.102:20252
yarp: Port /actionsRenderingEngine/head/state:i active at tcp://10.0.0.102:20253
yarp: Port /actionsRenderingEngine/head/stateExt:i active at tcp://10.0.0.102:20254
yarp: Sending output from /actionsRenderingEngine/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /actionsRenderingEngine/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/state:o to /actionsRenderingEngine/head/state:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /actionsRenderingEngine/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[... 199 lines cut ...]
||| found /home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine/kinematic_offsets.ini
[INFO][Stereo -> Cartesian]: Homography
yarp: Port /actionsRenderingEngine/left/img:o active at tcp://10.0.0.102:20304
yarp: Port /actionsRenderingEngine/right/img:o active at tcp://10.0.0.102:20305
yarp: Port /actionsRenderingEngine/left/img:i active at tcp://10.0.0.102:20306
yarp: Port /actionsRenderingEngine/right/img:i active at tcp://10.0.0.102:20307
yarp: Port /actionsRenderingEngine/left/blobs:i active at tcp://10.0.0.102:20308
yarp: Port /actionsRenderingEngine/right/blobs:i active at tcp://10.0.0.102:20309
yarp: Port /actionsRenderingEngine/left/raw:i active at tcp://10.0.0.102:20310
yarp: Port /actionsRenderingEngine/right/raw:i active at tcp://10.0.0.102:20311
yarp: Port /actionsRenderingEngine/MIL/window:o active at tcp://10.0.0.102:20312
yarp: Port /actionsRenderingEngine/MIL/cmd:o active at tcp://10.0.0.102:20313
yarp: Port /actionsRenderingEngine/MIL/rec:i active at tcp://10.0.0.102:20314
yarp: Port /actionsRenderingEngine/MSR/cmd:o active at tcp://10.0.0.102:20315
yarp: Port /actionsRenderingEngine/MSR/rec:i active at tcp://10.0.0.102:20316
yarp: Port /actionsRenderingEngine/tracker:i active at tcp://10.0.0.102:20317
yarp: Port /actionsRenderingEngine/tracker:o active at tcp://10.0.0.102:20318
yarp: Port /actionsRenderingEngine/seg:o active at tcp://10.0.0.102:20319
yarp: Port /actionsRenderingEngine/cmd:io active at tcp://10.0.0.102:20320
yarp: Port /actionsRenderingEngine/get:io active at tcp://10.0.0.102:20321
yarp: Port /actionsRenderingEngine/rpc active at tcp://10.0.0.102:20322
^C[try 1 of 3] Trying to shut down
Module closing

```

pattacini, 2015-06-11:
Did you try to launch `ARE` in stiff mode by choosing `impedance off` in the configuration file?

It should suffice calling `actionsRenderingEngine --motor::impedance off`.

I'd expect there will be no shaking.

jypuigbo, 2015-06-11:
And you are right, I was rechecking the config file of ARE, I turned it off and there's no shaking at this moment. 

But then, my question would be: when would I be interested in using impedance control? When wouldn't it be needed?
Is it needed for the calibration commands like ```calib table``` or ```calib fingers```?

pattacini, 2015-06-12:
With **`impendace on`** ARE puts the low-level motor control in **compliant mode**, that is movements are accomplished according to the following law (see the [wiki](http://wiki.icub.org/wiki/Force_Control#Available_control_modes)):

[image],

which means that we use a different set of **PID gains** suitable for **torque control**. Therefore if we disable the `impendance` we switch back to **stiff mode** using the standard set of position PID gains.

Nonetheless, in stiff mode we still need to rely on the force/torque measurements in order to detect external contacts (see `calib table` [+]). This detection is triggered whenever the forces at the end-effector (the infamous red arrows) overcome a given threshold. That being said, since those arrows are getting crazy, if the F/T measurements are causing this trouble I would expect then that in stiff mode the robot won't move at all (you can verify this). By contrast, if the arrows are shooting in the air as an effect, it is very likely that somehow the low-level control is not configured properly.

A reason for such a mis-configuration might be that **Torque PID Gains** and dynamic parameters (**stiffness** and **damping**) are not correctly received. Could you then take note of these parameters reading them from the `yarpmotorgui`? For the first four joints of each arm you have to click on the `PID` button and go into two tabs: (1) `Torque PID` and (2) `Stiffness Parameters`.

The `Torque PID` need to be checked against the values stored in the files `icub-main/app/robots/iCubBarcelona01/hardware/motorControl/<icub_[left|right]_arm.xml>` in the "`TRQ_PIDS`" group. The `Stiffness Parameters` are to be checked against the values stored in the `ARE` configuration file, namely the options `impedence_arm_stiffness` and `impedence_arm_damping`.

[+] The `calib fingers` command is not concerned with force control.

pattacini, 2015-06-12:
As side note, I've seen that the configuration file found by `ARE` is `/home/icub/Robots/icub-main/build-cluster/share/iCub/contexts/actionsRenderingEngine/config.ini`. This means that it's likely you edited the `config.ini` in the `build` when I asked you to modify the `impendance` option. Remember that this is a wrong practice that may originate problems that are difficult to be caught. Instead, use the `yarp-config context --import actionsRenderingEngine config.ini` command to create a copy of the file in the `~/.local` path.

If you already did the copy, forget about this message :smirk:

jypuigbo, 2015-06-12:
As you have foreseen, in impedance mode it doesn't move although if it's not producing the shaking either. I've checked the two config files and they are almost equal to the ```yarpmotorgui```. 
The only changes are: 
* ```ktau``` is supposed to be ```1``` in all the joints and it's ```0``` in ```yarpmotorgui```
* stiffness should be ```(0.4 0.4 0.4 0.2 0.05)``` and ```yarpmotorgui``` says ```(0.3999 0.3999 0.3999 0.2 0.431)```, what doesn't seem too wrong. 

I attach a link with the full list, in case the whole list is needed. Marked in bold are the values that are different in more or less measure. 
https://docs.google.com/spreadsheets/d/1HwgHwb9HqgJrudmdNhZBLCEvWClqvRCSescK_Y1gADg/edit?usp=sharing

I take good note of the side note ;)

pattacini, 2015-06-12:
Thanks @jypuigbo for the test. They seem all fine. I think `ktau` is meaningful only for ETH robots, thus not for CAN robots: @randaz81 can you confirm that?

What I'm pretty sure about is that `ARE` is not responsible for such a daunting misbehavior, but I'm afraid this problem lies outside my competences...

pattacini, 2015-06-12:
Well, actually we can still argue that it's more likely a sensor problem rather than a control problem, since it keeps happening also in stiff mode.

Now I'm wondering whether you can see those red arrows going crazy even before starting up ARE.

jypuigbo, 2015-06-12:
No, not that crazy. The arrows are relatively small even before calibrating WBD, but it's true that they move in random directions all the time. Small but random. You can see them both in the videos. When the arrows stop being big, the ARE is already off. And before they go crazy, ARE is still not running. 

IcubGui shows the hand trembling a bit, although I can't notice those movements in the robot (neither by sight neither touching). But I assumed this should just be some noise. You can also appreciate this a bit in the video.

pattacini, 2015-06-15:
@robotology/icub-support-admin 
I think we would need a specific test for verifying the functionalities of the F/T sensors.

super-ste, 2015-06-18:
Okay guys,
a meaningful check to perform in order to massese the functionality of your F/T sensors is to scope the output of the F/T sensor and send us the relative screenshots.

[Check here how to setup yarpscope] (http://wiki.icub.org/yarpdoc/yarpscope.html)

Launch the robot interface and leave the robot in the calib position. Scope the 3 force channel and the 3 torque channel for both scopes and let's see where is the origin of the shakes.

Let us know

jypuigbo, 2015-06-22:
Is this more or less what you need? This is showing the daa of the first 4 indices [red, green, blue, orange] for the 3 ports that you can read on the titles, for both arms. I'm not very sure if is this what you were looking for. 

[image]

pattacini, 2015-06-22:
@jypuigbo I think we need you go deeper and log directly the F/T sensors output, therefore please redo the plot of the `icub/left_arm/analog:o` and `icub/right_arm/analog:o`, that is those ports that are upstream wrt `WBD`.

jypuigbo, 2015-06-22:
I hope this looks better. 
I guess it's normal, but for the previous plots, which were still running, the lines had already gone out of the boundaries. 
[image]

jypuigbo, 2015-07-05:
I just write to know if there are any news on this issue. 

In any case, trying to realize what was going wrong I found the plots I posted where both from the same arm... Here's the new one. 

Please, let me know of anything you think I can try and check. 

[image]

super-ste, 2015-07-07:
@jypuigbo 
Can you please zoom in all the curves so we can evaluate the amplitude of the noise.

@pattacini 
Can we perform the same test on one of our robot (Blackie) and use it as a "gold standard" to see how it compares with the ones in Barcelona. 

I am still puzzled if we have an hardware problem or not.

pattacini, 2015-07-07:
It would be helpful if we can use here the same `xml` files of `yarpscope` @jypuigbo used in Barcelona to acquire the data.
@jypuigbo can you send them to us?

jypuigbo, 2015-07-14:
I attach the soomed plot and at the same time I'm sending the xml file to Ugo.

[image]

pattacini, 2015-07-15:
There are the results on our `Blacky`.

Outputs of `analog ports`:
[image]

Outputs of `WBD ports`:
[image]

I don't think these results are decisive anyhow.

jypuigbo, 2015-08-24:
After an afternoon of troubleshooting with @pattacini we found that: 
* ARE is not the cause of the problem, as we can reproduce the error from ```yarpMotorGui```
* ARE with ```impedance    off``` works fine 
* Turning one single joint in ```torque``` or ```compliant``` mode is sufficient to reproduce the problem
* The problem appears more likely with a combination of joints
* The problem depends on the number of joints, the specific joints and the mode, but it is not deterministic

Any further help will be welcome :)

Thanks, 

Jordi

pattacini, 2015-08-24:
@jypuigbo tomorrow could you repeat the tests with `damping` multiplied by a coefficient `10`?

jypuigbo, 2015-08-26:
I will do that today.

jypuigbo, 2015-08-26:
@pattacini Conclusions about these new tests: (considering that increasing the ```damping``` multiplied by 10 resulted in a some threshold around 0.091)
* Increasing the damping slightly reduces the trembling, but it doesn't cancel it (remember that although I changed from ```0.03``` to ```0.3```, the value in ```yarpmotorgui``` was ```0.091```)
* tests with the left arm show that the right arm is the only one affected. When ```ARE``` runs and puts all of them in ```torque``` mode, the trembling is just coupled with all the other joints. No trembling was observed in the left arm when using both previous and current damping coefficients. 
* Independent tests with both values, for the right arm limit the trembling to joint 0. I wouldn't consider significant the amount of trembling originated by the other 3 joints. 

[Note] I think that the right side of the body was the one that received most damage during the accident. Do you think that joint 0 could have been damaged back then?

So I tried to see if the motors in both arms behaved differently when moving, both visually and through ```yarpscope```, but in both cases the results with both arms seemed the same (no trembling neither weird perturbations when controlling the robot in position mode). 

I hope this helps.

pattacini, 2015-08-27:
@jypuigbo thanks a lot for the extensive tests! Therefore we can conclude that `right_arm joint 0` is the one causing trouble.

When you say _accident_, do you mean that scary event when the box containing the robot fell down making humans shudder?

Final comment: I'd expect that if you launch `ARE` with only the `left_arm` enabled and in compliant mode, then you won't see any trembling. Is that right?

The configuration option to enable only one arm is: `--motor::part_used left_arm`.

super-ste, 2015-08-27:
Hi Jordi,
good that we finally nailed it down to one joint. Now let's try to understand what's going on.

a. Calibrate the robot and leave it in the home position

b. Take the right arm from the upper arm and try to rock it back and forth (robot is controlled in position therefore it should be stiff.

Is it there any backlash? Backlash is a condition when you can read the movement around joint 0 (e.g. with the yarpmotorgui) but the motor is not able to control the join within that range.

The pic below shows the direction around which I suspect there maybe backlash.
[image]

If possible, please take a movie while you shake it.

Me too curious about the "accident"

super-ste, 2015-09-18:
Hi @jypuigbo 
do you have any news on this issue?

Are you guys planning on bringing this robot to IROS? If so we need to identify the root cause of the problem so we can carry the relevant spares.

pattacini, 2015-09-30:
Always consider reducing the PID gains when the robot is on the iKart since the voltage supply goes up `53 V`. This reduction aims to prevent oscillations from showing up.

As thumb's rule, multiply all the PID gains (`Kp`,`Ki`,`Kd`) by a factor of approximately `39/53 V` (i.e. we tend to supply the robot with `39 V` when not mounted on the iKart).

pattacini, 2015-09-30:
...and so I've heard the rumor that all this mess was due to a badly tuned configuration parameter of `WBD` nobody was aware of: the filter cut-off frequency.

A clear example of how the system complexity can easily f*** us up!
I feel I need somebody to hate all night long.

randaz81, 2015-10-01:
I suspect that other robots may be also affected by the same issue.
The following parameters (contained in `TRQ_PIDS` section) have to be checked on other `CAN` robots.

These are the correct default values for a `CAN` robot:
`<param name="kff">          0         </param>`
`<param name="kbemf">        0     </param>`
`<param name="filterType">   3     </param>`
`<param name="ktau">         0       </param>`

`ETH` robots are not affected by this issue.

traversaro, 2015-10-01:
@pattacini Just to clarify: the cut-off filter on torque estimates was only added during the summer in `wholeBodyDynamicsTree` in the `codyco-modules` repo, while is not present at all in `wholeBodyDynamics` in the `icub-main`.

pattacini, 2015-10-01:
@traversaro I meant actually the one in the `FW` and not in `WBD`.
The _bad_ outcome remains the same.

Tobias-Fischer, 2015-10-01:
Hi @ugo and @randaz81,
I've quickly looked into those parameters for our robot (iCubLondon01).
Three issues (least important first):
1) In our config files, all parameters are missing.
2) It seems like all robots which have these parameters in their configuration files do not have the default values set you are referring to
3) Even if those values were set, they would not have any effect as far as I can see. If you look in the source, you find the following statement (line 1447):
```
xtmp = trqPidsGroup.findGroup("kbemf"); 
{for (j=0;j<nj;j++) this->_bemfGain[j] = 0; }
```
Besides being bad style, `xtmp` is used nowhere. Same goes for the other parameters.

I guess this is due to a partly copy and paste from line 1497:
```
xtmp = trqControlGroup.findGroup("kbemf"); 
if (!xtmp.isNull())
{for (j=0;j<nj;j++) this->_bemfGain[j] = xtmp.get(j+1).asDouble();}
else
{for (j=0;j<nj;j++) this->_bemfGain[j] = 0; yWarning ("TORQUE_PIDS: 'kbemf' param missing");}
```

It would be great if this could be fixed as soon as possible, at it again seems to affect quite a few robots ..
Also, could you please clarify which default values are correct? The ones in the config files or the ones you have written before?

Best, Tobi

traversaro, 2015-10-01:
Ack, I was just a lot confused. :+1:

pattacini, 2015-10-01:
@traversaro I'm confused too :unamused:
From the rumor I thought it was in `WBD`, but I was wrong; it's in `FW`.

DanielePucci, 2015-10-01:
I have a question. True, disabling the filter seems likely to cause the problem, and this filter must be put back. But, should it be in the ~~~FW~~~? In my opinion, the firmware should only do basic computations for, e.g., torque control, but leave the signals as it receives them. The data processing or filtering should not be in the firmware. I fear that adding filters here and there will be the source of other problems. In this case, the filter should be in ~~~WBD~~~, which is in charge of estimating the torques, and also in having some filtering policies for its the inputs outputs.

randaz81, 2015-10-01:
@Tobias-Fischer 
I apologize for generating confusion in my last answer (i'd like to remove the post!). My diagnosis was wrong, and I want to reassure iCubLondon01 users that they configuration parameters seems correct to me.
I'm going check if `filterType = 3` give better performances than `filterType = 0` but for now I can guarantee that default values imposed by the motioncontrol (regardless what is written in the xml file) are ok. 
Some notes:
- It looks like that for 'iCubBarcelona01' the problem was caused by `kff=1', the only parameter which was effectively used. This parameter enables a feed forward path in the controller which should be not used if the PID gains are not accurately tuned (i.e. each motor requires an identification).
- `TRQ_PIDS` and `TORQUE_PIDS` are deprecated section tags. New robots should use `TORQUE_CONTROL` section, which indeed also apply the kbemf, ktau etc. Please do not rename the sections by your own, we already have an open task for updating the configuration files.
- ktau is not used at all by the BLL boards firmware (current build 80-82).
- I'm going to check now if which robots have kff=1.

randaz81, 2015-10-01:
@DanielePucci please forget about the filter. It is off by default. The problem was not due to the filter.

DanielePucci, 2015-10-01:
Well, sorry then.

kt10aan, 2015-10-01:
i watch the thread with interest as iCubOsaka has the same issue (violent shaking) when using impedance. It's was not urgent for us as we were not using impedance mode the past months, but we will in the next weeks. I do not know if the shaking is for the same reasons as the iCubLondon though.

randaz81, 2015-10-01:
@kt10aan I'm currently checking all the robots one by one. Your icubOsaka01 seems ok to me. I suggest to open a new separated thread to investigate your issue.

kt10aan, 2015-10-01:
Thanks! I wil do!

pattacini, 2015-10-02:
To sum up, does anybody know what's the current status for this issue in particular?

jypuigbo, 2015-10-07:
Both changes, the one due to the difference in voltages when mounted in the iKart proposed by @pattacini  and the ```TRQ_PIDS``` changes proposed by @randaz81 solved the shaking problems in iCubBarcelona01. 

Thanks a lot.

pattacini, 2015-10-07:
Great! Closing then.

closed 2015-10-07 by pattacini

Tobias-Fischer, 2015-10-07:
Sorry, but solving it for one robot does not solve it for all robots ;-). 

It would be great if this issue stays open until the configuration files for all robots are fixed. Firstly, as I've written above, changes in `TRQ_PIDS` does not change anything right now due to the bug. So I guess the changes of `Kp`, `Ki`, and `Kd` proposed by @pattacini did the trick. Could you please report your new values, @jypuigbo?

Thans a lot,
Tobi

pattacini, 2015-10-07:
For the sake of filing issues I would rather keep this closed and open a new ticket on the general maintenance of those parameters, if not done yet.

Let's discriminate among headaches! :smirk:

pattacini, 2015-10-07:
@Tobias-Fischer would you mind open a new ticket in `icub-main` (citing this as shortcut)?