robotology/mesh-support #31: iCubLondon01 right arm elbow motor blocked

arturoribes, 2015-06-12:
Hi,

Yesterday after a long run of experiments (around 2 hours of continuous arm usage), the arm suddenly stopped responding. I closed my applications, started yarpmotorgui and realised that joint 3 (elbow) was in idle mode. I tried to enable it by setting it to position mode and issuing commands using the GUI, but it went back to idle. 
Then I closed robotInterface and after going to parking position, the elbow was still in the same position, about 90 degrees (not in any of the motor limits). The arm cannot be moved, it seems that the motor is somehow blocked, and of course we do not want to use a lot of force in case it breaks.

super-ste, 2015-06-12:
Hi Arturo,
looks like your elbow tendon might have snapped and is stuck in the way of the movement. Can you check that? Or shoot a pic of the elbow seen from close by for us to look at?

Cheers

arturoribes, 2015-06-12:
From what we can see, both tendons are in their correct position. No one seems to have snapped. Here are some pics, let me know if you need more.

[image]
[image]
[image]
[image]

super-ste, 2015-06-25:
cc: @maxime-petit 

Hi London Crew,
after the troubleshooting you had with @julijenv we draw the following action plan:

* Unwire the right arm (seek Skype support if you need guidance)
* Remove and carefully pack the arm
* Ship it back to IIT.
* We will revise and solve the motor issue (probably a motor replacement will be needed)
* We will have a look at the fingertip issue (please update Martina on this).

and finally

* we will come up with the arm to remount & test it.

PM iCubSupport for the logistics of the shipment and our intervention.

Cheers
S.

Tobias-Fischer, 2015-07-10:
Hi guys,
we've replaced the bearing of the motor and it (manually) can be moved again. We've put everything back together, and tried to start robotInterface.
Unfortunately, we get these two errors repeated thousands of times:
`44388		15.993706		ERROR		recv buffer overrun ( 500 >  500)`
`44394		15.993962		ERROR		sharedcan [2] board 1 ABS ENCODER ERROR AXIS 1`

We've checked all the connectors, and they seem to be fine. Also, the blinking of the lights is in line with the other arm.
Could you please advice which wires/cables might be responsible for the errors we are getting?

We've found http://robotcub-hackers.2198711.n2.nabble.com/FailureReport-11-2012-Broken-tendon-no-2-td7247396.html, and checked the two attached images.
For https://www.dropbox.com/s/8hjjmul9w10yf5l/S4-2B1.jpg we are sure that everything is fine. However, for https://www.dropbox.com/s/q7co2k0kdzjn226/elbow_S4.jpg , the cable seems to be a tiny bit damaged (see attachment).

It would be great if you could tell us which continuity tests etc. might help to find the origin of the issues.

Best,
Tobias and @maxime-petit

Tobias-Fischer, 2015-07-13:
Hey,
we checked continuity between AEA and the corresponding connector on the back of the robot. Continuity is fine. As suggested via Skype, we disconnected the connector on the back. Then, we get the following errors:
```
529 6.574757 5 sharedcan [2] board 2 ABS ENCODER ERROR AXIS 1
 
 
533 6.574902 2 sharedcan [2] msg from board 1: PWM DIS:1 
 
535 6.574973 5 sharedcan [2] board 1 ABS ENCODER ERROR AXIS 1
```

So additionally to board 1, we get the same error for board 2. Furthermore, we get the `PWM DIS` error.

Best, Tobias

Tobias-Fischer, 2015-07-14:
Hi, @maxime-petit  and I were able to resolve the issue. Board 1 is apparently not the one corresponding to the elbow, but the one of the shoulder. Visual inspection showed that all wires were attached on the sensor side, but one wire was out of board 2B0 (the one below 2B1 on https://www.dropbox.com/s/8hjjmul9w10yf5l/S4-2B1.jpg). After soldering the wire back on the crimp, everything works fine.
@arturoribes can you please close this issue? We still have another problem with the other hand, but it is unrelated to this one so I'll open another issue.

Thanks all for your help!
Best, Tobias

closed 2015-07-14 by arturoribes