robotology/mesh-support #33: problem with fingertips

lornat75, 2015-06-29:
@bhigy @massimoregoli can you add more details?

pattacini, 2015-06-29:
Last week I sent a failure report to @robotology/icub-support-admin (through email) about the following issues:

- Left hand fingertips do not work.
- Right hand fingertips undergo frequent freezing.

@super-ste is working on it right now.

/cc @giuliavezzani

bhigy, 2015-06-29:
On the left hand, the fingertips of the thumb, index, middle and little finger are not working.
On the right hand, the palm and the fingertips of the ring and little finger are not working.

maggia80, 2015-07-06:
We have fixed the lefthand. Let me know if you find any other issue in the left fingertips. I will update about the status of the right hand soon.

pattacini, 2015-07-06:
so far so good...

pattacini, 2015-09-02:
@maggia80 @julijenv 
at certain point we'd need to schedule the same upgrade on the right hand.

maggia80, 2015-09-15:
see #19

closed 2015-09-15 by maggia80