robotology/mesh-support #34: robotInterface not running

jypuigbo, 2015-07-05:
Hi, 
Yesterday I had an accident with the robot and the left elbow tendon broke. Luckily I have Alex here to fix it tomorrow, but for some reason that I don't know yet, robotInterface is giving some errors and starts without calibrating anything. Do you think is because of the broken tendon or it might be something else? We  have a deadline next Tuesday (07/07) so this happened in the worst moment. If there exists some way of running the robotInterface only with torso-head-right_arm, it would serve as a fast, temporal solution for the deadline. 

Thanks in advance, 

Jordi

This is the first error found (and several are similar): 
```
[DEBUG]sharedcan [6] msg from board 7: MAIS error 
```
and 
```
[ERROR]100ms Timeout occured in setControlModeRaw ( j:1), current mode: idl, requested: pos
```
I can post the full log of the robotInterface if needed.

traversaro, 2015-07-05:
To run the `robotInterface` without running any left_arm-specific device, you have to write a custom `robotInterface` configuration file. 
All the following commands are relative to the computer where you run the `robotInterface`, i.e. the `PC104`.
To modify the current configuration file, import the `robotInterface` configuration file that you are using (looking at the configuration files for iCubBarcelona01 I guess it is `icub_all.xml`) and copy it in the local directories:
~~~
yarp-config -- import iCubBarcellona01 icub_all.xml
~~~
This will copy the `icub_all.xml` configuration file in `~/.local/share/yarp/robots/iCubBarcelona01/icub_all.xml`. To avoid confusion, you can rename the file to make clear that is used to run the robot without all devices relative to the left arm.
~~~
cd ~/.local/share/yarp/robots/
mv icub_all.xml icub_no_left_arm.xml
~~~
Now you can remove (or comment out) from the `icub_no_left_arm.xml` file  the lines relative to the devices left arm devices, i.e : 
~~~
<devices file="cartesian/left_arm_cartesian.xml" />
<devices file="wrappers/motorControl/left_arm_mc_wrapper.xml" /> 
<devices file="hardware/motorControl/icub_left_arm.xml" /> 
<devices file="hardware/motorControl/icub_left_hand.xml" /> 
<devices file="wrappers/VFT/left_arm_VFT_wrapper.xml" /> 
<devices file="hardware/VFT/left_arm_virtual_strain.xml" /> 
<devices file="wrappers/FT/left_arm_FT_wrapper.xml" /> 
<devices file="wrappers/MAIS/left_hand_mais_wrapper.xml" /> 
<devices file="hardware/FT/left_arm_strain.xml" /> 
<devices file="hardware/MAIS/left_hand_mais.xml" /> 
<devices file="wrappers/skin/left_arm_skin_wrapper.xml" />
<devices file="hardware/skin/left_arm.xml" />
<devices file="calibrators/left_arm_calib.xml" /> 
<devices file="calibrators/left_hand_calib.xml" /> 
~~~
Once you modified the file and saved it as `icub_no_left_arm.xml`, you can launch the robotInterface using this configuration file with the following command.
~~~
robotInterface --config icub_no_left_arm.xml
~~~

Now the robot should run fine. Depending on the modules that you use, you could need some additional flags, for example you have to run `wholeBodyDynamics` with the `--no_left_arm` option.

jypuigbo, 2015-07-05:
Thanks @traversaro for the super-fast reply!

I tried that. ``` robotInterface``` doesn't load anything related to the left arm, cool until this point. But for some reason, the calibration sequence doesn't work at all, for no joint of the robot: 
```
[DEBUG]Head_Calibrator: checkGoneToZeroThreshold: joint: 5 curr: 0.000 des: 0.000 -> delta: 0.000 threshold: 2.000 output: 0.000 mode: idl
[ERROR]Head_Calibrator : checkGoneToZeroThreshold: joint  5  is idle, skipping! 
[ERROR]Head_Calibrator : set 3 : some axis got timeout while reaching zero position... disabling this set of axes
[DEBUG]head_calibrator finished calibration of device head_mc_wrapper 
```

And after finishing, it says it's working fine, I open ```yarpmotorgui``` and all the joints are idle. If I try to run any of them, the following error arises for each joint: 
```[ERROR]100ms Timeout occured in setControlModeRaw ( j:1), current mode: idl, requested: pos```

I'm not sure on what this could mean.

jypuigbo, 2015-07-05:
The problem magically disappeared after restarting. Left arm disconnected successfully and everything else ok. 
Thanks!

closed 2015-07-05 by jypuigbo