robotology/mesh-support #35: Multiple failures in sharedcan[x]

kt10aan, 2015-07-07:
Hi,

today we put our iCub in impedance mode and there was some shaking. Immediately we turned off the corresponding module.

After that however we can not start up robotInterface as we were getting multiple errors. Usually associated with [ERROR] shared can[x] ... Mainly in sharedcan[3] and [5].

The [3], the left leg, was an issue for some time, as the connector was not always making contact and we would disable often the legs.

Here is the relevant robotInterface log.

https://gist.github.com/kt10aan/650a2e5b61dbb9bcb1b6

I assume (and I hope) that one or more connectors somewhere got disconnected. Any ideas on how we should proceed with the troubleshooting?

julijenv, 2015-07-07:
Hi iCubOsaka01 users,
sharedcan[3] is as you find out the left leg and the sharedcan[5] is the left forearm - from the following lines of the logfile you sent us:
266 3.884559 4 sharedcan [5] joint: 0 can_address:  5 Unable to detect firmware version. Old firmware running?
267 3.884596 4 sharedcan [5] joint: 1 can_address:  5 Unable to detect firmware version. Old firmware running?
268 3.884633 4 sharedcan [5] joint: 2 can_address:  6 Unable to detect firmware version. Old firmware running? 
269 3.884668 4 sharedcan [5] joint: 3 can_address:  6 Unable to detect firmware version. Old firmware running?
THE BOARD ID 5 is not detected - so 2 possibilities : power supply pblms or can connector broken.
To check that.
First of all, close iCubInterface if ever opened, make sure the motors are turned ON.
launch canLoader:
select NET 5 and cfw2can and click CONNECT:
check if ever you can see 3 boards: 2 MC4, with ID5 and 7, and 1 MAIS with ID14 - if ever different for MC4 boards, expecially ID turned to 15 here is the problem, change it with its corresponding number and you can relaunch.
if you can see just 2 boards means that, as i was saying previously, the board 1B3 got an issue probably due to flaky can wires or power supply lines issue ( you can check that last possibility by just taking a look at the stack of boards underneath the cover of the upper arm, by lifting the elbow put it at 90 degree at watch inside the upper arm to detect if the second board doesnt have any light on it. if the 3 boards look like the same i mean unlighted in the same way, you will need to open the cover of the upper arm to find out the broken wire. TO proceed with that please make sure the robot is turned off totally. remove the screws holding the skin above the bicep cover, remove the screws holding the quadricep cover, the bottom (quadricep) should fall down, remove it carefully unconnect the 2 connectors of the skin from the quadricep cover.
you should have the 3 boards in front of your eyes now, unmount the 4 screws holding the alluminium stand on which the stack of board is connected in order to access to the upper part of the boards.
Check the wires on the 4 ways connectors of the second board.
Give me feedbacks or pictures of your findings if you cant figured out whats going wrong.
good luck

kt10aan, 2015-07-09:
The MC4 that was supposed to have ID5, had jumped to ID15... 
Changed it back to ID5 and now the robotInterface starts normally. I will run some tests to make sure that no others errors pop up, but it seems that surgery was avoided :)

Many many thanks!!

P.S. I used to get these "ID jumping" errors in a pcan/ecan iCub, but it seems that this type of malfunction is also happening with cfw2can. Is there any precautionary measure we can take to reduce/eliminate this kind of errors?

julijenv, 2015-07-09:
Hi iCubOsaka01 users,
AFAIK there is no way out to know when this happen, the boards reach this configuration in a random circonstances. There hs never been tested deeper to find out how it happens. It is by the way something that could be investigated but i guess that the priority is very low level (minus 15 or something :P). Just let us know if it happens often to you or just once a year - which would be pretty odd.
hope it has been very useful for you to get a constructive answer :D.

pattacini, 2015-07-09:
Eheh :clap: @julijenv

pattacini, 2015-07-09:
For me it's one of the worst bug we could ever have since it means we loose the control of the hardware...
But what @julijenv said seems to be true...

super-ste, 2015-10-23:
@marcoaccame 
apaprently this issue has been getting worst and worst. Has anyone on your side had a look at it

see also #91

marcoaccame, 2015-10-23:
I have seen an address change in mc4 only once. It is worth let @randaz81 know about that.

julijenv, 2016-02-08:
Is everything alright in Osaka? (I mean the iCub, hopefully you are doing ok ;P), can we close this issue?

kt10aan, 2016-02-08:
@julijenv, yes, the iCub is (mostly) ok. I will open another issue for the latest problem(it seems we have a lot of flakey connectors!). But this issue is dealt with a long time ago, thanks :) Closing.

closed 2016-02-08 by kt10aan