robotology/mesh-support #36: iCubSheffield01 - Sporadic failures of robotInterface when the robot is just switched on

marcoaccame, 2015-07-10:
At Sheffield people have experienced sporadic failures of robotInterface when the robot is just switched on from cold. Attached is the log file.

From analysis I have done, it seems that there is a failure to detect fw version of the 4DC board with addresses 2 and 3 which serves joints 2, 3, 4, and 5. On the same bus the BLL board at address 1 is instead seen.

At successive launch of robotInterface, the problem does not show.

Please, can you confirm / add more details of the phenomenon?

relevant lines of the log file
```
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
printing destinations and inverted map
0 1
2 2
4 3
[INFO]using ICUB_CANMASKS_STRICT_FILTER option
[DEBUG]class 0 set
[DEBUG]class 1 set
[DEBUG]class 2 set
[DEBUG]class 3 set
[DEBUG]Can read/write buffers created, buffer size: 500
[INFO]CanBusResources::initialized correctly
Registering new thread 0 out of 500
[ERROR]cfw2can [0] thread:0 msg:91 joint:2 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:3 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:4 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:5 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[INFO]
[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]cfw2can [0] joint: 0 can_address:  1 board type: 4 (BLL) version: 1.62 build: 80 CAN_protocol:1.2
[INFO]cfw2can [0] joint: 1 can_address:  1 board type: 4 (BLL) version: 1.62 build: 80 CAN_protocol:1.2
[WARNING]cfw2can [0] joint: 2 can_address:  2 Unable to detect firmware version. Old firmware running?
[WARNING]cfw2can [0] joint: 3 can_address:  2 Unable to detect firmware version. Old firmware running?
[WARNING]cfw2can [0] joint: 4 can_address:  3 Unable to detect firmware version. Old firmware running?
[WARNING]cfw2can [0] joint: 5 can_address:  3 Unable to detect firmware version. Old firmware running?
[INFO]**********************************
[INFO]
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]  It has been detected that your control boards are not using the same
[ERROR]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[ERROR]  Please update your system (iCubInterface and/or your control board firmware.
[ERROR]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device head_mc 
[WARNING]Cannot open device head_mc 
```

complete log file

```
icub@pc104:~$ robotInterface 
||| configuring
[... 914 lines cut ...]
[ERROR]left_arm_VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_arm_VFTserver 
[ERROR]right_arm_VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_arm_VFTserver 
[ERROR]torso_VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device torso_VFTserver 
[ERROR]left_arm_as_wrapperMais is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_arm_as_wrapperMais 
[ERROR]right_arm_as_wrapperMais is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_arm_as_wrapperMais 
[ERROR]left_arm_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_arm_skin_wrapper 
[ERROR]right_arm_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_arm_skin_wrapper 
[ERROR]torso_skin_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device torso_skin_wrapper 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[WARNING]There was some problem running actions for shutdown phase . Please check the log and your configuration 
[INFO]shutdown phase finished. 
[ERROR]Error in shutdown phase... see previous messages for more info 
Module failed to open
icub@pc104:~$ 

```

pattacini, 2015-07-10:
see remaining points in https://github.com/robotology/icub-support/issues/20.

pattacini, 2015-09-02:
@lukeboorman news on that?

/cc @super-ste @julijenv @robotology/icub-support-admin

pattacini, 2015-09-02:
/cc @maggia80

closed 2018-01-16 by julijenv