robotology/mesh-support #37: iCubSheffield01 - The left arm must be helped during calibration

marcoaccame, 2015-07-10:
At Sheffield people are experiencing failures of calibration of the left arm if the arm is not helped to reach its final position.

From comparison of the log files (two cases: OK and failure) it can be seen that in case of failure it happens the following.

there are several encoder errors which do not show if the arm is helped.
```
[ERROR] from BOARD 1, src LOCAL, adr 0, time 58s 683m 483u: (code 0x02000009, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data + INFO = NO MORE
```
it is detected a HW fault on left arm joint 3 which prevents calibration
```
[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint 3 before starting the calibration procedure!
....
[ERROR]unable to complete calibration: joint 3 in 'hw_fault status' inside doneRaw() function
```

log files follows

marcoaccame, 2015-07-10:
first log file: the arm is helped
```
icub@pc104:~$ robotInterface 
||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubSheffield01
||| checking [/etc/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
yarp: Port /robotInterface active at tcp://10.0.0.2:10006
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3349 active at tcp://10.0.0.2:10066
||| finding file hardware/electronics/pc104.xml
||| checking [/home/icub/hardware/electronics/pc104.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml
||| finding file cartesian/left_arm-cartesian.xml
||| checking [/home/icub/cartesian/left_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml
||| finding file cartesian/right_arm-cartesian.xml
||| checking [/home/icub/cartesian/right_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml
||| finding file hardware/motorControl/head-cfw2_can0-mc.xml
||| checking [/home/icub/hardware/motorControl/head-cfw2_can0-mc.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml
||| finding file general.xml
||| checking [/home/icub/general.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/general.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml
||| finding file hardware/electronics/head-cfw2_can0-eln.xml
||| checking [/home/icub/hardware/electronics/head-cfw2_can0-eln.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml
||| finding file hardware/mechanicals/head-cfw2_can0-mec.xml
||| checking [/home/icub/hardware/mechanicals/head-cfw2_can0-mec.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml
[... cut ...]
MANAGEMENT-appl-status: sign = 0xaa000000, board EB4 -> name = aEMS00v2mec
                        version = 2.3, built on date 19 6 2015, at hour 10:55
                        state = applstate_config
[DEBUG]ethResources::goToConfig() detected BOARD 4 with name aEMS00v2mec and version = ( 2 3 ) 
[DEBUG]ethResources::goToConfig() successfully sent BOARD 4 in cfg mode 
[INFO]Closing Server Inertial...
[DEBUG]Server Intertial closed
[INFO]shutdown phase finished. 
Module finished
icub@pc104:~$ 

```

marcoaccame, 2015-07-10:
second log file: the arm is not helped

```
icub@pc104:~$ robotInterface 
||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubSheffield01
||| checking [/etc/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
yarp: Port /robotInterface active at tcp://10.0.0.2:10006
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3235 active at tcp://10.0.0.2:10054
||| finding file hardware/electronics/pc104.xml
||| checking [/home/icub/hardware/electronics/pc104.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml
||| finding file cartesian/left_arm-cartesian.xml
||| checking [/home/icub/cartesian/left_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml
||| finding file cartesian/right_arm-cartesian.xml
||| checking [/home/icub/cartesian/right_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml
||| finding file hardware/motorControl/head-cfw2_can0-mc.xml
||| checking [/home/icub/hardware/motorControl/head-cfw2_can0-mc.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml
||| finding file general.xml
||| checking [/home/icub/general.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/general.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml
||| finding file hardware/electronics/head-cfw2_can0-eln.xml
||| checking [/home/icub/hardware/electronics/head-cfw2_can0-eln.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml
||| finding file hardware/mechanicals/head-cfw2_can0-mec.xml
||| checking [/home/icub/hardware/mechanicals/head-cfw2_can0-mec.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
[... cut ...]
MANAGEMENT-appl-status: sign = 0xaa000000, board EB4 -> name = aEMS00v2mec
                        version = 2.3, built on date 19 6 2015, at hour 10:55
                        state = applstate_config
[DEBUG]ethResources::goToConfig() detected BOARD 4 with name aEMS00v2mec and version = ( 2 3 ) 
[DEBUG]ethResources::goToConfig() successfully sent BOARD 4 in cfg mode 
[INFO]Closing Server Inertial...
[DEBUG]Server Intertial closed
[INFO]shutdown phase finished. 
Module finished
icub@pc104:~$ 

```

pattacini, 2015-07-10:
see remaining points in https://github.com/robotology/icub-support/issues/20.

pattacini, 2015-09-02:
@lukeboorman news on that?

/cc @super-ste @julijenv @robotology/icub-support-admin

pattacini, 2015-09-02:
/cc @maggia80

urielmtz, 2015-09-21:
We still have problems if I don’t help the left arm. See log file and picture.

||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubSheffield01
||| checking [/etc/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
yarp: Port /robotInterface active at tcp://10.0.0.2:10004
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3124 active at tcp://10.0.0.2:10005
||| finding file hardware/electronics/pc104.xml
||| checking [/home/icub/hardware/electronics/pc104.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml
||| finding file cartesian/left_arm-cartesian.xml
||| checking [/home/icub/cartesian/left_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml
||| finding file cartesian/right_arm-cartesian.xml
||| checking [/home/icub/cartesian/right_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml
||| finding file hardware/motorControl/head-cfw2_can0-mc.xml
||| checking [/home/icub/hardware/motorControl/head-cfw2_can0-mc.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml
||| finding file general.xml
||| checking [/home/icub/general.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/general.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml
||| finding file hardware/electronics/head-cfw2_can0-eln.xml
||| checking [/home/icub/hardware/electronics/head-cfw2_can0-eln.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml
||| finding file hardware/mechanicals/head-cfw2_can0-mec.xml
||| checking [/home/icub/hardware/mechanicals/head-cfw2_can0-mec.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/mechanicals/head-cfw2_can0-mec.xml] (robot)
[... cut ...]
MANAGEMENT-appl-status: sign = 0xaa000000, board EB4 -> name = aEMS00v2mec
                        version = 2.3, built on date 19 6 2015, at hour 10:55
                        state = applstate_config
[DEBUG]ethResources::goToConfig() detected BOARD 4 with name aEMS00v2mec and version = ( 2 3 ) 
[DEBUG]ethResources::goToConfig() successfully sent BOARD 4 in cfg mode 
[INFO]Closing Server Inertial...
[DEBUG]Server Intertial closed
[INFO]shutdown phase finished. 
Module finished

[image]

Cheers
Uriel

marcoaccame, 2015-09-24:
from email of  @urielmtz to me dated 24/09/2015:

```
Yes, the left arm needs help to start the calibration and even with help 
sometimes it doesn't start properly.
```

Thus, ... the problem is likely to be independent from the aid we give to the arm.

marcoaccame, 2015-09-24:
It seems that: **very likely there is some error on the reading of the AEA** which sometimes shows and some other times it does not.

Looking at the log file:

At time 58s 632m 85u we know that we are inside the control loop because the EMS sends up diagnostics about found CAN boards (the EMS does that as first operation inside the first iteration of the control loop).

Then, see the following log fragment:

```
[DEBUG] ethResources::goToRun() called for BOARD 1 
MANAGEMENT-appl-status: sign = 0xaa000000, board EB1 -> name = aEMS00v2mec
                        version = 2.3, built on date 19 6 2015, at hour 10:55
                        state = applstate_config
MANAGEMENT-appl-status: sign = 0xaa000000, board EB1 -> name = aEMS00v2mec
                        version = 2.3, built on date 19 6 2015, at hour 10:55
                        control loop timings: RX->400us, DO->300us, TX->300us
                        state = applstate_running
[INFO] from BOARD 1, src CAN1, adr 1, time 58s 632m 85u: (code 0x00000033,
   par16 0x0000 par64 0x0000020106010008) -> 
   SYS: the can discovery service found the board on the specified CAN bus / adr.
   In par64 there is the eObrd_typeandversions_t. + INFO = NO MORE
```

Then, at time 58s 683m 483u (50 iterations after), the EMS sends up an error diagnostics message about the AEA value of fourth joint (joint 3) being invalid.

```
[ERROR] from BOARD 1, src LOCAL, adr 0, time 58s 683m 483u: 
  (code 0x02000009, par16 0x0003 par64 0x0000000000000000) -> 
  MC: AEA encoder invalid data + INFO = NO MORE
```
The EMS thus forces a HW fault and repeats the error diagnostics message every 5 seconds:

```

[ERROR] from BOARD 1, src LOCAL, adr 0, time 63s 631m 484u: 
   (code 0x02000009, par16 0x0003 par64 0x0000000000000000) -> 
   MC: AEA encoder invalid data + INFO = NO MORE
... omissis 

[ERROR] from BOARD 1, src LOCAL, adr 0, time 68s 631m 484u: 
   (code 0x02000009, par16 0x0003 par64 0x0000000000000000) -> 
   MC: AEA encoder invalid data + INFO = NO MORE

... omissis

[ERROR] from BOARD 1, src LOCAL, adr 0, time 73s 631m 487u: 
   (code 0x02000009, par16 0x0003 par64 0x0000000000000000) -> 
   MC: AEA encoder invalid data + INFO = NO MORE

```

Then, we can see that between time 68s and 73s there is a diagnostics message written by the calibrator in robotInterface about the same joint:

```
[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  3  before
    starting the calibration procedure! 
```

That is what stops everything.

closed 2018-01-16 by julijenv