robotology/mesh-support #38: iCubLisboa01 - July 2015 firmware upgrade

gsaponaro, 2015-07-10:
Hi, we'd like to upgrade the firmware on our robot, which we haven't done in ages.

In [this comment](https://github.com/robotology/QA/issues/29), @traversaro mentioned that there is a new procedure, could someone briefly explain it?

Additional information and questions:
* robot is iCubLisboa01, current firmware we are running is reported at the bottom of this message;
* the way we used to do the upgrade was to run the script ```icub-main/app/robots/iCubLisboa01/firmwareUpdate.txt```, which executes a sequence of ```cfw2can``` commands, is that still valid?
* in commit https://github.com/robotology/icub-main/commit/eca005037c631a3e656917b500a3f8e56b8b4a40#diff-da5eadbaf731ff44dd80314c08c3c837, @davidetome has changed the name of some firmware binaries associated to iCubLisboa01, in particular 4DC.**2** was changed to 4DC.**1**, is this correct for our robot?
* we have already installed and updated the repositories icub-firmware-build and icub-firmware-shared on the pc104;

Firmware we are currently running (before the upgrade), from robotInterface log:
```
[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [1] joint: 0 can_address:  1 board type: 4 (BLL) version: 1.50 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 1 can_address:  1 board type: 4 (BLL) version: 1.50 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 2 can_address:  2 board type: 4 (BLL) version: 1.57 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 3 can_address:  2 board type: 4 (BLL) version: 1.57 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 4 can_address:  3 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 5 can_address:  3 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 6 can_address:  4 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [1] joint: 7 can_address:  4 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [5] joint: 0 can_address:  5 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 1 can_address:  5 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 2 can_address:  6 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 3 can_address:  6 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 4 can_address:  7 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 5 can_address:  7 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 6 can_address:  8 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [5] joint: 7 can_address:  8 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [2] joint: 0 can_address:  1 board type: 4 (BLL) version: 1.50 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 1 can_address:  1 board type: 4 (BLL) version: 1.50 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 2 can_address:  2 board type: 4 (BLL) version: 1.57 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 3 can_address:  2 board type: 4 (BLL) version: 1.57 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 4 can_address:  3 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 5 can_address:  3 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 6 can_address:  4 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]sharedcan [2] joint: 7 can_address:  4 board type: 3 (4DC) version: 1.19 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [6] joint: 0 can_address:  5 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 1 can_address:  5 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 2 can_address:  6 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 3 can_address:  6 board type: 3 (4DC) version: 1.28 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 4 can_address:  7 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 5 can_address:  7 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 6 can_address:  8 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]sharedcan [6] joint: 7 can_address:  8 board type: 3 (4DC) version: 1.30 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [3] joint: 0 can_address:  5 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [3] joint: 1 can_address:  5 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [3] joint: 2 can_address:  6 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [3] joint: 3 can_address:  6 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [3] joint: 4 can_address:  7 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [3] joint: 5 can_address:  7 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [4] joint: 0 can_address:  8 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [4] joint: 1 can_address:  8 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [4] joint: 2 can_address:  9 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [4] joint: 3 can_address:  9 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [4] joint: 4 can_address: 10 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]sharedcan [4] joint: 5 can_address: 10 board type: 4 (BLL) version: 1.51 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [0] joint: 0 can_address:  1 board type: 4 (BLL) version: 1.62 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 1 can_address:  1 board type: 4 (BLL) version: 1.62 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 2 can_address:  2 board type: 3 (4DC) version: 2.15 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 3 can_address:  2 board type: 3 (4DC) version: 2.15 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 4 can_address:  3 board type: 3 (4DC) version: 2.15 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 5 can_address:  3 board type: 3 (4DC) version: 2.15 build: 80 CAN_protocol:1.2
[INFO]**********************************

[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.2
[INFO]Firmware report:
[INFO]sharedcan [0] joint: 0 can_address:  5 board type: 4 (BLL) version: 1.54 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 1 can_address:  5 board type: 4 (BLL) version: 1.54 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 2 can_address:  6 board type: 4 (BLL) version: 1.52 build: 80 CAN_protocol:1.2
[INFO]sharedcan [0] joint: 3 can_address:  6 board type: 4 (BLL) version: 1.52 build: 80 CAN_protocol:1.2
[INFO]**********************************
```

/cc @lorejam

traversaro, 2015-07-10:
@gsaponaro the procedure is always the same, I just referring that http://wiki.icub.org/wiki/Firmware should be updated to take in consideration that now the firmware images are supposed to be downloaded from https://github.com/robotology/icub-firmware-build .

gsaponaro, 2015-07-10:
I see, thanks!

Now I'd just like to clarify about the 4DC.**2** vs 4DC.**1** question and I'm good to go. I'm a bit cautious when touching firmwares.

randaz81, 2015-07-28:
@gsaponaro firmware 4DC.2 is not an upgraded version of fw 4DC.1. It is supposed to run on a completely different hardware. So when you update your system please be sure to use the firmware files with the same exact version number. 
Only the build number (80) will change.

gsaponaro, 2015-07-28:
Thanks @randaz81, we confirmed we were going to flash the correct files, and the firmware was successful. :)

Just an observation: during the upgrade, some huge bogus numbers were printed, however in the end all boards reported upgrade went OK. Here is the log where I marked the "strange" lines in asterisks:

```
canLoader --canDeviceType cfw2can --canDeviceNum 2 --boardId 2 --firmware 2BLL.1.57.out.S
Initializing prompt version of canLoader...
Selecting canDeviceType cfw2can canDeviceNum 2, boardId 2, firmware 2BLL.1.57.out.S
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
[INFO]CONNECTED: 4 Boards found on bus 1
[DEBUG]    id   type     version
[DEBUG]   13       6     3.0.1
[DEBUG]    3       3     1.25.80
*** [DEBUG]-92599993   -81477707     -953434187.-583006203.-89913338 ***
[DEBUG]    2       4     1.87.80
downloading 2BLL.1.57.out.S, 1% done
downloading 2BLL.1.57.out.S, 25% done
downloading 2BLL.1.57.out.S, 50% done
downloading 2BLL.1.57.out.S, 75% done
downloading 2BLL.1.57.out.S, finished!
Program terminated successfully!

canLoader --canDeviceType cfw2can --canDeviceNum 2 --boardId 3 --firmware 4DC.1.19.out.S
Initializing prompt version of canLoader...
Selecting canDeviceType cfw2can canDeviceNum 2, boardId 3, firmware 4DC.1.19.out.S
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
[INFO]CONNECTED: 4 Boards found on bus 1
[DEBUG]    id   type     version
[DEBUG]   13       6     3.0.1
[DEBUG]    2       4     1.87.80
[DEBUG]    3       3     1.25.80
*** [DEBUG]-1800730140   -1789607997     -953434173.-583006203.-1798043642 ***
downloading 4DC.1.19.out.S, 1% done
downloading 4DC.1.19.out.S, 25% done
downloading 4DC.1.19.out.S, 50% done
downloading 4DC.1.19.out.S, 75% done
downloading 4DC.1.19.out.S, finished!
Program terminated successfully!

cfw2can 0 1 2BLLDC.1.62.out.S --> ALL OK
cfw2can 0 2 4DC.2.15.out.S --> ALL OK
cfw2can 0 5 2BLL.1.54.out.S --> ALL OK
cfw2can 0 6 2BLL.1.52.out.S --> ALL OK
cfw2can 1 1 2BLL.1.50.out.S --> ALL OK
cfw2can 1 2 2BLL.1.57.out.S --> ALL OK
cfw2can 1 3 4DC.1.19.out.S --> ALL OK
cfw2can 5 5 4DC.1.28.out.S --> ALL OK
cfw2can 5 7 4DC.1.30.out.S --> ALL OK
cfw2can 2 1 2BLL.1.50.out.S --> ALL OK
cfw2can 2 2 2BLL.1.57.out.S --> ALL OK
cfw2can 2 3 4DC.1.19.out.S --> ALL OK
cfw2can 6 5 4DC.1.28.out.S --> ALL OK
cfw2can 6 7 4DC.1.30.out.S --> ALL OK
cfw2can 3 5 2BLL.1.51.out.S --> ALL OK
cfw2can 3 6 2BLL.1.51.out.S --> ALL OK
cfw2can 3 7 2BLL.1.51.out.S --> ALL OK
cfw2can 4 8 2BLL.1.51.out.S --> ALL OK
cfw2can 4 9 2BLL.1.51.out.S --> ALL OK
cfw2can 4 10 2BLL.1.51.out.S --> ALL OK
```

closed 2015-07-29 by gsaponaro