robotology/mesh-support #41: Issues to be solved before delivering iCub + cluster to Lyon

pattacini, 2015-07-25:
@robotology/wysiwyd-developers consider to expand on the following:

- [x] @pattacini @AnneLaureM @Tobias-Fischer set up software on the cluster.
- [ ] @mbrunettini on the blades nodes we cannot employ `sudo`; here is the error message: **`sudo: effective uid is not 0, is sudo installed setuid root?`**. The current workaround is to ssh those machines as root. 
- [ ] @randaz81 verify the status of the iKart.
- [ ] @robotology/icub-support-admin @super-ste @julijenv we have seen a somewhat relevant tilt misalignment between the eyes. 
- [ ] @robotology/icub-support-admin @super-ste @julijenv when we control the eyes very fast in position we get weird movements so that we disabled the saccades for the time being.
- [ ] @robotology/icub-support-admin @maggia80 we should apply the latest fingertips patch.

Tobias-Fischer, 2015-07-27:
Hi @randaz81 @robotology/icub-support-admin @julijenv,
when starting the iCubLyon01 today in the morning, we got errors related to the MAIS of the left hand. Only the green light is on, whereas the red light does not switch on at all. On the right hand, the red light first shows orange, and later on becomes red.
We already checked whether the boards on the knuckles are mounted properly, and they are. We asked @vtikha, and he suggested to ask you guys.
Maybe it would be a good opportunity to apple the fingertips patch @maggia80?

Best, Tobi and @AnneLaureM

Tobias-Fischer, 2015-07-27:
Hi @pattacini, @mbrunettini,
I think the issue regarding `sudo` is resolved. I can ssh into `icub-b1` etc. as `icub` user, and then do e.g. `sudo test` without any issues.

super-ste, 2015-07-27:
Hi all,
regarding the MAIS issue, a new firmware was committed to help troubleshooting the MAIS problems.

So, if you can see the MAIS on the CANLoader i'd advise to load up the latest firmware version.

If the MAIS is not responding on the CAN then probably we have a flaky cable somewhere. Please report and we'll take it from there.

For the fingertip patch, we can certainly have a look at it when the robot comes back.

mbrunettini, 2015-07-27:
How did you fix it?

Matteo

AnneLaureM, 2015-07-27:
Hi @mbrunettini,

We have not done anything precisely. I guess it was a startup procedure error ... (between the server and nodes). Maybe It will more sure you take a look when the cluster will be at IIT. But normally the sudo works properly.

Thanks !
Anne-Laure & @Tobias-Fischer

mbrunettini, 2015-07-27:
Ok, I'll double-check it, once in IIT.
This system is not installed from scratch on every new system, but cloned from a working setup and we never faced the same issue in the past. But it's strange that the system has been rebooted many times when we saw the issue.
Anyway, thanks Anne-Laure & Tobias!

mbrunettini, 2015-07-27:
Just to record the issue, it seems that the SETUID bit is missing on some binaries (i.e. /usr/bin/sudo) or it has a wrong ownership
https://en.wikipedia.org/wiki/Setuid
I'll check it.

AnneLaureM, 2015-07-27:
Hi @super-ste,

Now the left hand works, it seems the problem came from a old connector that had moved. Do you think we should still update the firmware OR this can wait the iCub comes back to IIT ?

Thanks
Anne-Laure

super-ste, 2015-07-27:
Hi,
I guess you still could do the upgrade. It takes no more than 5 mins and could be useful to troubleshoot further issues.
@randaz81 can easily drive you thru it.

AnneLaureM, 2015-07-27:
roger that !

pattacini, 2015-07-28:
@robotology/icub-support-admin @super-ste @julijenv added note on the broken ethernet cable: see https://github.com/robotology/icub-support/issues/41#issue-97216747.

Tobias-Fischer, 2015-07-28:
Hi @pattacini, @super-ste, @julijenv,
additionally to the issue with the external Ethernet cable, we also have the problem that we only get a 100Mbps connection, rather than 1Gbps.

@randaz81:
Is the problem with the iKart fixed with https://github.com/robotology/ikart/commit/28e2bb435334ef9fbbb397124e98897a5a69f58f?

pattacini, 2015-07-28:
it came out that the `100 MB/s` problem was apparently due to a specific ethernet plug onto the switch.

AnneLaureM, 2015-08-03:
Hi @super-ste 
I left (at the end of VVV15) with the iCubLyon01, a laptop for the cluster console (brand LDLC with french keyboard ;-) ). The old laptop being a little too old to run several graphical interfaces (yarpmanager, Yarp view, iCubGui). I brought this computer for VVV15, it was not sent with the robot. So it is not listed in the material received by IIT. Will it be possible to return it at INSERM with the new cluster, the robot & the iKart when the robot will be fixed ?
Cheers,
Anne-Laure

super-ste, 2015-08-03:
Hi @AnneLaureM 
Sure!!

@pattacini gave us precise instruction wrt that!
One piece of info for @julijenv is whether the laptop has already been updated with the current configuration files (i.e. if we can use it to test the robot or if we better use the old one).

Cheers
S.

AnneLaureM, 2015-08-03:
@pattacini Thanks !!
@super-ste @julijenv, indeed It will be better to use the new one to test the robot if the French keyboard does not bother you (I'm also really sorry for that ... :-( ). I think we never will restart the old one ;-)
Just one more thing, I also noticed that the "Home" position on yarpMotorGui did not work. Is there a way to set it up?
Thanks very much to all of you !!
Anne-Laure

super-ste, 2015-08-03:
HI @AnneLaureM, 

regarding your question:

> Just one more thing, I also noticed that the "Home" position on yarpMotorGui did not work. Is there a way to set it up?

please check if there is an answer to that in:
https://github.com/robotology/QA/issues/42

if not, please open an new issue there.

mbrunettini, 2015-08-05:
@AnneLaureM,
I just checked the two racks about the "sudo"issue: everithing is ok and the systemis working smoothly.
As an additional precaution, since the CISCO network switch takes a lot of time to boot up, please wait until the light on the connected ports becomes GREEN (about 5 minutes) before powering the icub-server up, then after other 5 minutes the icub-server is started correctly and you can turn on the blades and the CUDA machine.
I also check the new laptop, I just changed the iKart network configuration from dynamic (DHCP) to static, with the ip 10.0.0.10, the same used in the old laptop.
Matteo

AnneLaureM, 2015-08-07:
@mbrunettini

Sorry for my late answer, I'm on vacation and the internet connection is not really great. I'll be careful to start the switch and the server. Thank you for checking.
Cheers,

Anne-Laure

julijenv, 2015-08-12:
@pattacini , @AnneLaureM ;
Please feel free to add something if missing @mbrunettini 
What has been made on iCubLuon01 here iCubFacility:
pc104 changed
iKart uptodate
check problems with eyes when saccades ON - everything looks alright - a video has been made!!!!
eyes alignment done
recalibration left arm and twicking right arm and head - still need to find out how to add the xls file from my pc to github, but every single files needed to launch robotInterface with perfect calibration are inside the repository $ICUB_ROOT/app/robots/iCubLyon01/ so please launch always from that location instead of .local/etc.... or else calibration wrong

pattacini, 2015-08-12:
@julijenv well done thanks!
... however, concerning the last point: that's very much against the policy :smirk:
Never start up the robot from `$ICUB_ROOT/app/robots/iCubLyon01/`, but always comply with the rules (see https://github.com/robotology/QA/issues/42)!

pattacini, 2015-08-12:
It should be just a matter of purging the `.local/share/yarp/robot` directory and cmaking `icub-main`.

@AnneLaureM closing the issue for the meantime; double check the bullet points when you'll receive the robot.

closed 2015-08-12 by pattacini