robotology/mesh-support #43: iCubLisboa01 - robotInterface errors and warnings, July 2015

gsaponaro, 2015-07-29:
Hi all, having recently updated all software and firmware on our robot (pc104 and cluster), we noticed some errors and warnings reported by robotInterface. I summarize them below, and I uploaded the full log here: https://gist.github.com/gsaponaro/48f6cac5983c3cde9daa

Any hints on how to address these issues? Thanks!

1)

Many skin errors:

```
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  14 sensor:  5 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  11 sensor:  0 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  1 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  2 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  3 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  4 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  6 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  7 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  9 sensor:  9 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  9 sensor:  10 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  9 sensor:  11 error:  01RR.0000.0000.0000 
[ERROR]canBusSkin error code:  canDeviceNum:  7 board:  8 sensor:  12 error:  01RR.0000.0000.0000 
```

2)

A few FAULT messages (in fact, occasionally some joints start in IDLE or in HARDWARE FAULT state, then we are able to control them from ```yarpmotorgui```):

```
[ERROR]sharedcan [1] board 3 FAULT OVERLOAD AXIS 1
[ERROR]sharedcan [1] board 4 FAULT OVERLOAD AXIS 0

[ERROR]sharedcan [6] board 5 FAULT OVERLOAD AXIS 1

[ERROR]sharedcan [6] board 6 FAULT OVERLOAD AXIS 0
```

3)

Warnings during XML parsing:

```
||| found /home/icub/.local/share/yarp/robots/iCubLisboa01/./icub_all.xml
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/./icub_all.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/./icub_all.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/./icub_all.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
```

```
||| found /home/icub/.local/share/yarp/robots/iCubLisboa01/cartesian/left_arm_cartesian.xml
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/cartesian/left_arm_cartesian.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/cartesian/left_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubLisboa01/cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
```
and so on for other XMLs.

4)

```
[WARNING] ControlBoardWrapper2 device:
************************************************************************************
* ControlBoardWrapper2 'name' parameter for port name does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name   *
*     name:    full port prefix name with leading '/',  e.g.  /robotName/part/     *
* A temporary automatic fix will be done for you, but please fix your config file  *
************************************************************************************ 
[WARNING] *** ControlBoardWrapper2 is using DEPRECATED parameter 'threadrate', use 'period' instead *** 
[WARNING]*** ControlBoardWrapper2: found both 'period' and DEPRECATED 'threadrate' parameters. Using 'period', please remove the deprecated 'threadrate' param *** 
```

/cc @lorejam

gsaponaro, 2015-09-11:
After having solved #23 and having rewired a couple of skin wires, here is a fresh robotInterface log: https://gist.github.com/gsaponaro/9eb6f6b7f854476988d3

The skin problem is solved, the other ones are still there, in particular 2). Any help or suggestion is warmly welcome (please look at the new log).

/cc @robotology/icub-support-admin

julijenv, 2015-09-24:
Hi @gsaponaro and @lorejam ,
I read your log file, but I am a bit astonished because their a bunch of error that is for sure, but sometimes it is like they arent real because they showed up without e\xplanation followed and looks like they have been deleted. (did you take a log and remove sentences from it???).
Do the robot start correctly and worlks fine?
It doenst matter the warning it is software parser related and doesnt affect a failure of the robot they are just WARNINGS. but ERROR can lead to failure of launching robotInterfface.
Is it your case?
please resend a logfile.

gsaponaro, 2015-09-25:
Hi @julijenv.

no, we have not edited the previous logs in this present issue. Just in case, here is a fresh one: https://gist.github.com/gsaponaro/a3401380cc1d8e22a4ca

As you can see we still get the following errors:
```
[ERROR]sharedcan [2] board 4 FAULT OVERLOAD AXIS 0  <- many times
...
[ERROR]sharedcan [0] board 2 FAULT OVERLOAD AXIS 0
[ERROR]sharedcan [0] board 2 FAULT OVERLOAD AXIS 1
```

The good news is that since we solved #23 a few days ago, the robot starts correctly and during operation it *usually* works fine, however we do get an occasional motor hardware fault now and then (that we can recover with **yarpmotorgui**).

We can close this issue (perhaps after we solve those remaining ERROR messages, if you agree).

julijenv, 2015-09-25:
Hi @gsaponaro ,
ok your are right you have quite a lot of message 
" [ERROR]sharedcan [2] board 4 FAULT OVERLOAD AXIS 0 "
could you open the right arm upper cover and check if all the connection are well plugged inside the board 2B3, and especially the FAULT wires and connector (2 by boards).
the other message pop up just on the close sequence phase and doesnt really mean anything is wrong so far.
LEt us know for the right arm,
best of search,
:P

gsaponaro, 2015-09-25:
Thanks, we checked those wires and connectors, they seemed ok, we reconnected everything just in case.

We then re-ran **robotInterface** a few times, and the message "[ERROR]sharedcan [2] board 4 FAULT OVERLOAD AXIS 0" is still there. In **yarpmotorgui**, all joints in the right arm are operational and I manage to move them.

lornat75, 2015-10-20:
@julijenv: can you check this issue? looks like there is still a FAULT problem

julijenv, 2015-10-29:
Hi @gsaponaro ,
is the robot arm joint are, for at least one, in idle mode after calibration?
There are some messages regarding FAULT ISSUE : ([ERROR]sharedcan [2] board 4 FAULT OVERLOAD AXIS 0) meaning a flaky connection, but you couldnt find it last time I asked u to check inside the upper cover, so as long as the wire still make connection most of the time I guess you will still get this firstthread of 10 times writing on the log file and afterwards we can say that the FAULt doenst pop up again.
If the robot start without idling any part of the robot (right arm especially) please close the issue.

gsaponaro, 2015-10-29:
Right. We have been using the robot for long hours these past few days: no issues on any joints of that arm, neither at startup (none of them starts in IDLE) nor during operation.

Closing this issue as you suggested, thanks.

closed 2015-10-29 by gsaponaro