robotology/mesh-support #44: iCubDarmstadt01: Position Direct Ctl is slow

paraschos, 2015-07-29:
The setPositions over the IDirectPositionCtl takes forever to return and I can  control the robot at max. frequency of 12Hz. I am "viewing" the interface from the PolyDriver which opens the "remote_controlboard", "left_arm".

I am getting warning messages whenever I am controlling the robot:

[WARNING] from BOARD 1, src LOCAL, adr 0, time 840s 190m 209u: (code 0x0000000d, par16 0x0235 par64 0x0000000000000000) -> SYS: the TX phase of the control loop has last more than wanted. In par16 there is the total execution time in usec + INFO = NO MORE
[WARNING] from BOARD 1, src LOCAL, adr 0, time 840s 195m 212u: (code 0x0000000d, par16 0x0238 par64 0x0000000000000000) -> SYS: the TX phase of the control loop has last more than wanted. In par16 there is the total execution time in usec + INFO = NO MORE

paraschos, 2015-07-29:
Since it could be a configuration issue I have to say that the robotInterface can't start normaly from the home but it says:

```
||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubDarmstadt01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubDarmstadt01
||| checking [/etc/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot//yarp/build-pc104/share/yarp/robots/iCubDarmstadt01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot//icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01
||| checking [/usr/local/src/robot//iCubContrib/share/ICUBcontrib/robots/iCubDarmstadt01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubDarmstadt01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot//icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/robotInterface.ini] (robot)
||| found /usr/local/src/robot//icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/robotInterface.ini
yarp: Port /robotInterface active at tcp://10.0.0.2:10007
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| found /home/icub/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3192 active at tcp://10.0.0.2:11891
[FATAL]Syntax error while loading /home/icub/./icub_all.xml at line 0 . Error document empty. 
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os4impl7LogImpl16forward_callbackENS0_3Log7LogTypeEPKcS6_jS6_+0x111a) [0x7f3dabe2674a]
robotInterface(_ZN4yarp2os9LogStreamD1Ev+0x259) [0x4141f9]
robotInterface(_ZN14RobotInterface9XMLReader7Private13readRobotFileERKSs+0x177) [0x432087]
robotInterface(_ZN14RobotInterface6Module9configureERN4yarp2os14ResourceFinderE+0x4e3) [0x41bf83]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0xa) [0x7f3dabea273a]
robotInterface(main+0x140) [0x413a80]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd) [0x7f3daacd4ead]
robotInterface() [0x413df9]
robotInterface(_ZN4yarp2os9LogStreamD1Ev+0x459) [0x4143f9]
robotInterface(_ZN14RobotInterface9XMLReader7Private13readRobotFileERKSs+0x177) [0x432087]
robotInterface(_ZN14RobotInterface6Module9configureERN4yarp2os14ResourceFinderE+0x4e3) [0x41bf83]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0xa) [0x7f3dabea273a]
robotInterface(main+0x140) [0x413a80]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd) [0x7f3daacd4ead]
robotInterface() [0x413df9]
```

The robotInterface starts happily from $ICUB_ROOT. Not sure if it is related.

paraschos, 2015-07-29:
An interesting point:

"One more question: what is the max rate that I can set cmd over the IPosDirect? Does the setPos blocks until the pos is reached from the robot or until the command is ack from pc104?"

@randaz81 said:
"setPos is not blocking. If you send new reference position, and the joint still has not reached the previous position, the target will be simply overwritten (and the PID will track the new reference). About the frequency, I never tried above 1KHz. "

So... if the call is not blocking why I can't execute my program faster than 12Hz? I should be able to get quite high frequencies...  

-a

barbalberto, 2015-07-29:
Indeed position direct is not blocking and it is meant for higher frequency loops. It shall take no time to return.
Can you plese comment out the SetPositions line and see if you loop speeds up or it takes about the same time?

paraschos, 2015-07-29:
Apparently the problem was the setPositionDirectMode(). I copied some code that was sending it over and over again at every time point. Now I can send commands as fast as I like. What is the max frequency you recommend? I tried for fun up to 5KHz and I could send them without congesting the system. My controller was also quite accurate with an overtime of 0.3ms at a movement of 3s. Up to 2KHz I have no overtime at all.

pattacini, 2015-07-29:
Since the FW runs at `1 KHz`, any sending rate > `1 KHz` won't have any effect on the control.

That being said, the sending rate `r` at the user level should be chosen to properly sample the waveform you aim to realize. Such a waveform will be limited in frequency up to `B` (i.e. the band), so that you can choose `r~10*B` (approximately), or something like that. From experience `r=100Hz` is pretty much fine.

closed 2015-07-30 by paraschos