robotology/mesh-support #45: Link error with iCub Hello World example

ibichind, 2015-07-30:
Hello,
I have installed YARP 2.3.64 and iCub 1.1.16 on Windows 7 32-bits with CMake 3.3.0. YARP compiles and runs fine. iCub also and the iCubSIM example works well. 
However when compiling iCub Hello World from this example http://wiki.icub.org/iCub/main/dox/html/icub_hello_icub.html, I get an error message:

Error	1	error LNK1104: cannot open file 'opencv_videostab.lib'	

Since I chose all options by default, I am not sure why I am getting this error. Any help would be welcome. 

Thank you in advance ! 

Lisa

ibichind, 2015-07-30:
Hello,
I have resolved the issue - I post here what I have done in case someone else gets the same error message. 
I modified the CMakeLists.txt file to include the OpenCV libraries:

cmake_minimum_required(VERSION 2.8.9)

if(COMMAND cmake_policy)
cmake_policy(SET CMP0003 NEW)
endif(COMMAND cmake_policy)

# find YARP and ICUB
FIND_PACKAGE(YARP)
FIND_PACKAGE(ICUB)
FIND_PACKAGE(OpenCV REQUIRED)
# add YARP and ICUB include directories
INCLUDE_DIRECTORIES(${YARP_INCLUDE_DIRS} ${ICUB_INCLUDE_DIRS})
# set up our program
add_executable(hello main.cpp)
# link with YARP and ICUB libraries
TARGET_LINK_LIBRARIES(hello ${YARP_LIBRARIES} ${ICUB_LIBRARIES} ${OpenCV_LIBRARIES})

Lisa

pattacini, 2015-07-30:
Hi @ibichind 
Happy to know that you solved the problem.
That was a very old `"hello world"` code; that's the reason of the bug. Thanks for spotting.

In the future, please consider posting SW related issues on https://github.com/robotology/QA or https://github.com/robotology/yarp rather than here, since `icub-support` is meant for maintenance issues.

closed 2015-07-30 by pattacini