robotology/mesh-support #46: Oscillations spotted on the head after maintenance

pattacini, 2015-07-31:
We have the hunch that after the maintenance work done to repair the Blacly's head there's been some weird problem with the head mechanics.

Refer to https://github.com/robotology/icub-main/issues/121 for the complete history.

Tobias-Fischer, 2015-09-29:
Hi all,
I'd like to report that we've seen these oscillations on several other robots as well.

1) At the BCBT summer school, we've seen the oscillations on `iCubBarcelona01` when using `IOL`. A video can be found here: https://slack-files.com/T053A59T4-F0B1WACGL-e64d5fbece (sorry, it's quite large)

2) However, the problem does not only occur using `IOL`, but also when the gaze is not even controlled. We've seen the same behavior on `iCubLondon01` several times when we were driving the iKart around to map the environment. I am not 100% sure which modules were running at the time, but typically I reduce it to a minimum. I am not even sure whether `iKinGazeCtrl` was running.

The oscillations occur randomly, and I don't know how to track them down. Please let me know if I can help resolving this issue. It would be great if someone could look into this ASAP, as it is quite worrying ..

I am also not sure when we've encountered this problem the first time. We haven't updated the firmware for a little while, however we got the head upgrade to V2 a little while back. Typically `yarp` and `icub-main`

pattacini, 2015-09-30:
@Tobias-Fischer 

Take https://github.com/robotology/icub-support/issues/30#issuecomment-144377458 into account regarding the oscillations when the robot is on the iKart.

/cc @maggia80 @randaz81 @super-ste

lornat75, 2015-10-20:
any update on this?

pattacini, 2015-10-20:
No progress at all.
Blacky's head still goes crazy every once in a while.

@hagenlehmann and @matejhof can provide a more recent report on this, I suspect.

I feel 99.9999999999999% confident to say that it's not an `iKinGazeCtrl` problem.

super-ste, 2015-10-22:
Action Plan after IFI meeting:

- [ ] Check mechanical stability of support
[image]

julijenv, 2015-11-06:
has been changed, let's see if it brings improvments

pattacini, 2016-05-10:
I'm the feeling we can close this :relieved:

closed 2016-05-10 by pattacini