robotology/mesh-support #47: iCubGenova04 - electrical problem (in a 2FOC board?)

ale-git, 2015-08-06:
Today in the late afternoon there has been smoke coming from the right shoulder electronics (2B0 - 2B1 zone). I've shutdown the robot after that.

julijenv, 2015-08-10:
the EMS has been changed but most of the MTB are gone.

julijenv, 2015-08-10:
A lot of MTB3 have been burned (or seemed ok bcoz they turned ON but at the end there was no way to talk to them from canLoader). It has been changed 5 boards on the overall of 7 on the right arm. Before reconnected the Right arm , I'll proceed to a test to know if it is the line of MTB boards which are responsible to burned the EMS or the contrary.

julijenv, 2015-08-10:
After recabling the entire forearm and the EMS and board in the back, I turn On the motors and try to check the overall behaviour of the repair. EMS ok, but MAIS and MTBs imposssible to detect. After a check it seems that the MAIS is gone, but the thing that I dont know is whether or not it was broken before the repair. A new hypothesis is that the MAIS just burned out and as it is connected via FC connector it is possible that the CAN line of the skin connected to EMS burned but it is also possible the other way around but in order to know it by certainty i should have check the integrity of the MAIS before connected everything - sorry my bad.

julijenv, 2015-08-12:
Everything that has been changed seems now working fine.
2 more things to check 
the calibration right hand joint 8 for thumb abduction.
the fingertips from index to pinky finger NOT working

lornat75, 2015-08-17:
Show is going to check these issues?

julijenv, 2015-08-17:
"Show" means "Whose" I guess, well ASAP we'll return from Woods Hole. Anyway this isssue was still opened because some little details need to be checked but I prefered having another issue opened (#49 in this case) because I have a least a trace of the number of repair which has been done.

closed 2015-08-25 by julijenv