robotology/mesh-support #48: iCubLondon01 - Head movements jerky (joint 0+1)

Tobias-Fischer, 2015-08-11:
Hi,
we are experiencing issues with the movements of joint 0 and 1 of our iCub. The issues appeared suddenly, a couple of days ago everything seemed to be fine.
When moving the joints in e.g. `yarpmotorgui`, the movements are sometimes very jerky, especially when the head is facing downwards and has to go back up.
We took a video, the whole body is vibrating when the head is moved: https://www.youtube.com/watch?v=TlmjLKbrvWc

Best, Tobias

Tobias-Fischer, 2015-08-11:
Hi,
in additionally, we realized that we cannot run ARE properly anymore. If we try `home all`, the head just goes crazy. It starts doing huge oscillations, and we have to stop it with the red button.
We tried setting `–imu::mode off` already.

It would be great if you could advice what to do :).

Thanks,
Tobias and @maxime-petit

pattacini, 2015-08-11:
The symptom seems similar to #129.
Did you try to update yarp and icub-main on the cluster and pc104 too?

pattacini, 2015-08-11:
Well, not that similar if you did use yarpmotorgui, but still...

Tobias-Fischer, 2015-08-11:
Hi @pattacini,
`yarp` and `icub-main` are very recent versions on all of our machines. I don't think this is the issue. Should we still try disabling the position direct mode as suggested in https://github.com/robotology/icub-main/issues/129, just to give it a try?

Thanks!
Tobi

vtikha, 2015-08-11:
Hi @Tobias-Fischer , yes please try disabling as @pattacini suggested with post #129. We also encountered similar issues as described with the guys in Lisbon and disabling it removed the jerk. 
Let us know if this is also the case and we will try to figure this out.

pattacini, 2015-08-12:
@Tobias-Fischer if you made use of `yarpmotorgui` it's very likely you did not employ `position-direct`; further, the issue robotology/icub-main#129 is meant to be solved by the latest commits in the repos, so that those fixes should be already available in your system, as you said.
With these premises I don't think the problem might be due to the `position-direct` mode. Nonetheless, an attempt will add up a bit of knowledge.

These days I came across a nice definition of specific nasty bugs that tend to hide themselves and change their behaviors as we try to narrow them down. Read more on [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug) and hope this is not one example of them :smirk:

pattacini, 2015-08-12:
Stupid question: did you make sure that the binaries you launch are from the latest compilation?

maxime-petit, 2015-08-12:
Hi,

Binaries are from the latest compilation yes, and as mentionned earlier
happens with a simple yarpmotorgui. Firmare is build 80

By desactivating the position-direct with the neck with ARE, I was able to
launch it and execute a "go home" and a pointing gesture :). I still have
the "jerky" move, but ARE is not going crazy anymore.

I will try some "look" ARE commands to check further. I have seen also that
objects localisation were flickering a bit with SFM/iol2opc, so actually it
may be due to an accumulation of small things with some synergy between
each other : flickering object localisation + jerky neck movement + ARE
looking at object = huge neck oscillations.

Because also the jerky beck movement (joint 0 and 1) seems to happens (or
are more noticeable) when the head look down, is it possible that it is due
to the motor that struggle to keep up (with gravity/balance more difficult
to deal with than when the neck is vertical?) or some physical
deterioration in cable/belt/strap?

Trying to move using velocity instead of position may provide some useful
informations?

pattacini, 2015-08-12:
Guys, if the jerk occurs just by playing around with the `yarpmotorgui`, well, stop doing any other test with `ARE`, `iKinGazeCtrl`, `position-direct` related stuff... simply because the latter modules/modes cannot be the cause of the problem.

pattacini, 2015-08-12:
Can you instead increase the speed values in the `yarpmotorgui` in order to make the jerk more evident, if possible?

maxime-petit, 2015-08-13:
Hi,

I have 2 videos to show the jerkiness of joint 0 and 1 with velocity 5 and 20. I have used yarpmotorgui with robotinterface launch :
- vel 5   : https://youtu.be/xthNcjO6I20
- vel 20 : https://youtu.be/JM2HO9fIjC8 

Logs of robotinterface are in attachment
[log_pc104_robotInterface_13-08-15.txt](https://github.com/robotology/icub-support/files/1402/log_pc104_robotInterface_13-08-15.txt)

pattacini, 2015-08-14:
Thanks @maxime-petit. The jerk is pretty much visible in both the tests.

@robotology/icub-support-admin @julijenv @super-ste

julijenv, 2015-08-15:
Hi icubLondon01 users,
OK we agree that the neck does not move as smooth as it could. From the first assumption that  made you raise this issue: "that everything looked alright few days before" usually it makes us probe to think that something at the level of software went wrong. BUT the 2 last videos you run without any specific software, such as ARE (so just robotInterface - if I catch it correctly) show definitively  that something should go wrong but at the level of hardware. The first thing that jumps in my mind is : did something happen that could affect the good behaviour of the head? If not did you make a first check to the neck cables and belt: tension/backlash, wear, slide. Some pictures of the neck from a narrow angle for the belt and cables could help us to certify what's going wrong. Please feel free to add them. Thx in advance

maxime-petit, 2015-08-19:
Hi, 

I will take a look today at the hardware level, sorry for the delay. I don't think anything happens, it was fine before VVV (when we have fixed the motor in the left elbow arm, involving cable manipulation in the back but not in the head/neck) and has not be used when we were out. 

In the log attached previously, there are a lot of messages like this : 
`sharedcan [X] have not heard from board Y (channel Z) since S seconds`

May it be the cause of troubles?

maxime-petit, 2015-08-25:
Hi,

Sorry for the delay. Here 2 pictures of the neck from 2 views. Let me know if you need another angle.

[image]
[image]

Tobias-Fischer, 2015-09-28:
Hi, we don't have this issue anymore, possibly it had something to do with https://github.com/robotology/icub-support/issues/61. We will re-open the ticket in case it happens again.

Thanks for your support guys!

closed 2015-09-28 by Tobias-Fischer