robotology/mesh-support #49: iCubGenova04 - joint 11 of left hand not working

bhigy, 2015-08-13:
Joint 11 of left hand (index proximal joint) is not moving when using yarpmotorgui.

lornat75, 2015-08-17:
Is this the same as: https://github.com/robotology/icub-support/issues/47#issuecomment-129376746 ?

lornat75, 2015-08-17:
Is this issue the same as: https://github.com/robotology/icub-support/issues/47#issuecomment-129376746

julijenv, 2015-08-17:
no it isn't the same issue: one was raised by @ale-git and referred to 2FOC boards which got burned but in the establishment of the repair I notice some details to be improved which could correlate a part of what is saying @bhigy .

bhigy, 2015-08-18:
This issue is about the left hand while #47 is about the right one.

vtikha, 2015-08-24:
@julijenv as @bhigy mentioned in this issue, the proximal joint of the left index finger is not working (although the little and ring do work) but I also noticed that the wrist of the left hand is extremely loose and needs tightening.

julijenv, 2015-08-26:
left index prox and distal fixed
sledge redone entirely to be able to repair distals index and middle fingers.
wrist cables retense

closed 2015-08-26 by julijenv

vtikha, 2015-08-26:
thanks @julijenv