robotology/mesh-support #50: Improve internal documentation/tests for maintenance group

pattacini, 2015-08-16:
Write up/update tests and documentation describing what needs to be done on a robot after its maintenance is complete and right before delivering the robot to the lab/customers.

We can benefit from the recently conceived [**`icub-tests`**](https://github.com/robotology/icub-tests) repository for that purpose (and we might want to extend it if necessary).

Tests include and are not limited to:
- red-ball demo
- gaze control
- skins
- fingertips
- force control 
- legs
- ...

I would need the help of @randaz81 @DanielePucci @maggia80 and others.

traversaro, 2019-10-03:
@pattacini I guess this issue probably needs to be moved somewhere else.

pattacini, 2019-10-03:
Thanks @traversaro for pointing it out.
Actually, we've been dealing with these issues internally, already.
Closing then.

closed 2019-10-03 by pattacini