robotology/mesh-support #52: iCubGenova02 - Calibration issues at startup

francesco-romano, 2015-08-19:
At the current state the purple robot has (big) issues starting up. In particular I found issues on the calibration phase on the following parts:

- torso (it must start perfectly aligned with the gravity)
- right_leg (the ankle does not move during the calibration phase. I had to manually move the joint to the desired configuration otherwise the `checkMotionDone` timeout occurs)

cc @randaz81

traversaro, 2015-09-09:
Situation evolved: now the `right_leg` always starts in idle because the calibration is failing when the robot is launched on the fixed base:
~~~
2844		32.575386		ERROR		Right_Leg_Calibrator : set 1 : some axis got timeout while reaching zero position... disabling this set of axes
 
2832		32.072039		ERROR		Right_Leg_Calibrator : checkGoneToZeroThreshold: joint  4  Timeout while going to zero! 
~~~
Furthermore, when the leg control mode is switched again in position, the low level position PID seem to me limited to ~5 Nm of torque (i.e. it is impossible to reach angles of 50/60 degrees with the first axis). Curiously the issue disappears if the interaction mode is set to compliant.

closed 2015-11-04 by francesco-romano

DanielePucci, 2015-11-04:
Users having a similar issue must be aware that this  workaround must NOT be replicated on other platforms other than iCubGenova02 at the current status. This platform has a specific hardware issue currently being investigated.

@randaz81 @marcoaccame