robotology/mesh-support #54: iCubGenova01 - issues after MBL summer school

alecive, 2015-09-02:
Here are the issue list for the blackie (I'll update it as if I can find something else):

 * `icub-b2` does not want to start/connect to the network. There is a yellow blinking light and we cannot access them
 * The forearm left skin does not work any more after me and @cciliber opened it for a fix and tried to put it back (with no luck)
 * The proximal joint of the middle finger has its tendon broken
 * There is a power supply issue on the left wrist which affects also the proximal joint of the index. Video available at this link: https://goo.gl/photos/scgYfrA9nQr3rRks5

Cc @julijenv @super-ste @cciliber @giuliaP @ommac

pattacini, 2015-09-02:
consider looking also at #46

GiuliaP, 2015-09-07:
Also, there is some data stored temporarily on the icub-tesla, to be moved asap. We had to use this machine to store the data collected by the students in their projects. In particular, in home/icub and /home/icub/tmp_giulia. I will move it as soon as I come to the IIT.

super-ste, 2015-09-23:
The robot has been repaired.

I remind everyone that iCub will be packed again on `Friday 25` to go to *IROS*

@GiuliaP grab the data from the Tesla

@julijenv add info here on the maintenance performed

@pattacini @alecive @vtikha 
can we thoughtfully test the robot (perhaps with a minimal setup) to verify everything is ok?

vtikha, 2015-09-23:
@super-ste thanks, will do all tests today and tomorrow morning, as we also need to verify the state of the robot in preparation for tomorrow's demos `Thursday 24th`.

vtikha, 2015-09-23:
@super-ste I have quickly tested the blacky and. 
1- At first run the right hand started in hardware fault then after a motor shutdown it seemed ok (still weird)
2- Finger abduction of the right hand is not responding (joint 7) 
more later..

super-ste, 2015-09-23:
We looked into it and:
1. Started okay. 
2. resoldered the cable on the motor
3. :0

Coming back to the skin on the left forearm: checks OK.

@mbrunettini: could you quickly have a look at the issue with the 'icub-b2' ? We're packing for IROS on Friday....

mbrunettini, 2015-09-24:
ìcub-b2' seems to be starting correctly now, I tried many reboots and everything went fine.
Probably a temporary network issue with the cable connections.
Please monitor it and in case of another failure, please contact me.

super-ste, 2015-09-24:
CLosing the issue as all related task has been sorted out. Please open new issue for future problems that may arise.

closed 2015-09-24 by super-ste