robotology/mesh-support #57: iCubSheffield01 - Torso hardware fault

urielmtz, 2015-09-21:
Hello,

We have an issue with iCub torso, which doesn't start during the calibration process. It seems that the motors are not powered. Also, in the yarpmotorgui the three joints of the torso are in red colour and with the label 'hardware fault'.

Just as a test, I sent some commands to the /icub/torso/rpc:i port, e.g. [set] [pos] 0 20, and I received 'ack' for the commands.

Here is the text for the torso printed with the robotInterface is launched:

...
[ERROR] from BOARD 5, src LOCAL, adr 0, time 106s 616m 479u: (code 0x02000009, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data + INFO = NO MORE
...
[ERROR]Torso_Calibrator : detected an hardware fault on joint  0  before starting the calibration procedure! 
Registering new thread 0 out of 500
[ERROR]Torso_Calibrator : detected an hardware fault on joint  1  before starting the calibration procedure! 
[ERROR]Torso_Calibrator : detected an hardware fault on joint  2  before starting the calibration procedure! 
[DEBUG]Torso_Calibrator : Joints calibration order: ( 0 1 2 )  
...
[ERROR] from BOARD 5, src LOCAL, adr 0, time 766s 616m 479u: (code 0x02000009, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data + INFO = NO MORE
...

I am wondering if do know what would be the cause of this problem with the torso?

Thanks.
Uriel

marcoaccame, 2015-09-24:
This issue is similar to #37 but here is **a failure of the AEA in joint 0 of the torso (board EB5)**.

The evidence is the same as in #37:
* the AEA on BOARD 5 has invalid data on joint 0.
* the error from the calibrator in roboInterface happens after the AEA error has been reported.
* the AEA error diagnostics messages are separated exactly by 5 seconds, as seen in the following complete log.

In here is the complete log file as sent to me by @urielmtz:

```

||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubSheffield01
||| checking [/etc/yarp/robots/iCubSheffield01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
yarp: Port /robotInterface active at tcp://10.0.0.2:10004
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3247 active at tcp://10.0.0.2:10052
||| finding file hardware/electronics/pc104.xml
||| checking [/home/icub/hardware/electronics/pc104.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/pc104.xml
||| finding file cartesian/left_arm-cartesian.xml
||| checking [/home/icub/cartesian/left_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/left_arm-cartesian.xml
||| finding file cartesian/right_arm-cartesian.xml
||| checking [/home/icub/cartesian/right_arm-cartesian.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/cartesian/right_arm-cartesian.xml
||| finding file hardware/motorControl/head-cfw2_can0-mc.xml
||| checking [/home/icub/hardware/motorControl/head-cfw2_can0-mc.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl/head-cfw2_can0-mc.xml
||| finding file general.xml
||| checking [/home/icub/general.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/general.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/general.xml
||| finding file hardware/electronics/head-cfw2_can0-eln.xml
||| checking [/home/icub/hardware/electronics/head-cfw2_can0-eln.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/electronics/head-cfw2_can0-eln.xml
||| finding file hardware/mechanicals/head-cfw2_can0-mec.xml
||| checking [/home/icub/hardware/mechanicals/head-cfw2_can0-mec.xml] (pwd)
[... cut ...]
[DEBUG]cfw2can [0] printing boards infos:
None
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR] from BOARD 5, src LOCAL, adr 0, time 591s 616m 479u: (code 0x02000009, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data + INFO = NO MORE
[DEBUG]cfw2can [0] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms]

julijenv, 2015-09-24:
Hi Sheffield users,
did the robot got faulted and fell forward which lead to get the AEA wire pinched? (wire can be find following the belly metal plate and running inside of the bottom of the robot?
seee picture
[image]

take a look at this specific wire.see if in another case one of the wire is disoldered but most probably also one wire got broekn at the level of the board in between the 2 hips, stack of boards.
let us know

urielmtz, 2015-09-30:
Hi Julien,

Thanks for the picture. I think that wire seems to be broken, below there is a photo of the robot:

[image]

Do you think this could be the problem?
I was thinking to measure continuity but I am not sure which screws we should remove. Is there a procedure we could follow to check these wires on the torso?

Cheers
Uriel

urielmtz, 2015-09-30:
I forgot to mention that the robot also fell forward when there was a failure.

Cheers
Uriel

julijenv, 2015-10-01:
Hi @urielmtz ,
From the picture u sent i guess the cable is pretty severe and the simplest way to repair it is to replace it as the pinch mark is a bit further from the edge of the cable. We will send you the replacement part asap. Could u please give us the add on which we have to send u the spare part and a name and phone number of the person in charge of retrieving the part?

urielmtz, 2015-10-01:
Hi @julijenv,

Thanks for the fast reply. The person that will receive the cable is James Law and the details for sending the parcel are below.

Dr James Law
Senior Experimental Officer
Sheffield Robotics

The University of Sheffield
Rm B04 Pam Liversidge Building
Sir Frederick Mappin Building
Mappin Street
Sheffield
S1 3JD
UK
Tel. +44 (0)114 222 5690

Please let me know if you need more information.

Cheers
Uriel

super-ste, 2015-10-06:
Shipping spare today.

Tracking here:

http://www.dhl.com/cgi-bin/tracking.pl?AWB=6288587734

urielmtz, 2015-10-08:
We received the cable today, we replaced it and the torso is working again!

Thanks for the support.

Cheers
Uriel

super-ste, 2015-10-08:
YEIIIIIII!!! 
Closing the Issue then!!

closed 2015-10-08 by super-ste