robotology/mesh-support #58: segm fault on robotInterface

reafrancesco, 2015-09-22:
@randaz81 here is the last sentence printed out by the robotInterface. @julijenv is helping fixing few problems on the hardware and he is aware there is HW_FAULT but we think this is not the problem.

[ERROR]Unable to set the control mode of a joint ( j:2) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:3) in HW_FAULT
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
[INFO]shutdown phase finished. 
Module finished
*** glibc detected *** robotInterface: corrupted double-linked list: 0x788432f8 ***
======= Backtrace: =========
/lib/i386-linux-gnu/i686/cmov/libc.so.6(+0x70c91)[0xb6f2ec91]
/lib/i386-linux-gnu/i686/cmov/libc.so.6(+0x7252c)[0xb6f3052c]
/lib/i386-linux-gnu/i686/cmov/libc.so.6(cfree+0x6d)[0xb6f3363d]
/usr/lib/i386-linux-gnu/libstdc++.so.6(_ZdlPv+0x1f)[0xb70b44bf]
======= Memory map: ========
08048000-083bc000 r-xp 00000000 00:16 3359422    /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
083bc000-083ca000 rw-p 00374000 00:16 3359422    /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
0a137000-10f35000 rw-p 00000000 00:00 0          [heap]
57100000-57168000 rw-p 00000000 00:00 0 
57168000-57200000 ---p 00000000 00:00 0 
782b2000-785f1000 r-xp 00000000 00:16 4571651    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
785f1000-78600000 rw-p 0033f000 00:16 4571651    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
78600000-78658000 rw-p 00000000 00:00 0 
78658000-78700000 ---p 00000000 00:00 0 
78700000-787ff000 rw-p 00000000 00:00 0 
787ff000-78800000 ---p 00000000 00:00 0 
78800000-78900000 rw-p 00000000 00:00 0 
789ea000-789eb000 ---p 00000000 00:00 0 
789eb000-791eb000 rw-p 00000000 00:00 0 
791eb000-791ec000 ---p 00000000 00:00 0 
791ec000-799ec000 rw-p 00000000 00:00 0 
799ec000-799ed000 ---p 00000000 00:00 0 
799ed000-7a1ed000 rw-p 00000000 00:00 0 
7a1ed000-7a3a4000 r-xp 00000000 00:16 4571653    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7a3a4000-7a3ac000 rw-p 001b7000 00:16 4571653    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7c3b0000-7c6f0000 r-xp 00000000 00:16 4571673    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusSkin.so
7c6f0000-7c6ff000 rw-p 00340000 00:16 4571673    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusSkin.so
7c6ff000-7c700000 ---p 00000000 00:00 0 
7c700000-7cf00000 rw-p 00000000 00:00 0 
7cf00000-7d000000 rw-p 00000000 00:00 0 
7d0a5000-7d0a6000 ---p 00000000 00:00 0 
7d0a6000-7d8a6000 rw-p 00000000 00:00 0 
7d8a6000-7d8a7000 ---p 00000000 00:00 0 
7d8a7000-7e0a7000 rw-p 00000000 00:00 0 
7e0a7000-7e0a8000 ---p 00000000 00:00 0 
7e0a8000-7e8a8000 rw-p 00000000 00:00 0 
7e8a8000-7e8a9000 ---p 00000000 00:00 0 
7e8a9000-7f0a9000 rw-p 00000000 00:00 0 
7f0a9000-7f0aa000 ---p 00000000 00:00 0 
7f0aa000-7f8aa000 rw-p 00000000 00:00 0 
7f8aa000-7f8ab000 ---p 00000000 00:00 0 
7f8ab000-800ab000 rw-p 00000000 00:00 0 
800ab000-803ee000 r-xp 00000000 00:16 4571667    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
803ee000-803fd000 rw-p 00342000 00:16 4571667    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
81c00000-81e00000 rw-p 00000000 00:00 0 
81e00000-81f00000 rw-p 00000000 00:00 0 
82800000-82900000 rw-p 00000000 00:00 0 
83200000-83400000 rw-p 00000000 00:00 0 
83400000-83500000 rw-p 00000000 00:00 0 
83e00000-83f00000 rw-p 00000000 00:00 0 
83fbe000-842f1000 r-xp 00000000 00:16 4571669    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusAnalogSensor.so
842f1000-842ff000 rw-p 00333000 00:16 4571669    /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusAnalogSensor.so
842ff000-84300000 rw-p 00000000 00:00 0 
84300000-84500000 rw-p 00000000 00:00 0 
84500000-84600000 rw-p 00000000 00:00 0 
85700000-857fa000 rw-p 00000000 00:00 0 
857fa000-85800000 ---p 00000000 00:00 0 
86900000-86a00000 rw-p 00000000 00:00 0 
88300000-88400000 rw-p 00000000 00:00 0 
88d00000-88e00000 rw-p 00000000 00:00 0 
[... 101 lines cut ...]
b750a000-b7667000 r-xp 00000000 08:01 33267      /usr/lib/libACE-6.0.3.so
b7667000-b7670000 r--p 0015d000 08:01 33267      /usr/lib/libACE-6.0.3.so
b7670000-b7672000 rw-p 00166000 08:01 33267      /usr/lib/libACE-6.0.3.so
b7672000-b7674000 rw-p 00000000 00:00 0 
b7674000-b7689000 r-xp 00000000 08:01 257        /lib/i386-linux-gnu/i686/cmov/libpthread-2.13.so
b7689000-b768a000 r--p 00014000 08:01 257        /lib/i386-linux-gnu/i686/cmov/libpthread-2.13.so
b768a000-b768b000 rw-p 00015000 08:01 257        /lib/i386-linux-gnu/i686/cmov/libpthread-2.13.so
b768b000-b768d000 rw-p 00000000 00:00 0 
b768d000-b76c3000 r-xp 00000000 08:01 32110      /usr/lib/i386-linux-gnu/libdc1394.so.22.1.7
b76c3000-b76c4000 rw-p 00036000 08:01 32110      /usr/lib/i386-linux-gnu/libdc1394.so.22.1.7
b76c4000-b7705000 rw-p 00000000 00:00 0 
b7705000-b7714000 r-xp 00000000 08:01 31943      /usr/lib/i386-linux-gnu/libraw1394.so.11.0.1
b7714000-b7715000 r--p 0000e000 08:01 31943      /usr/lib/i386-linux-gnu/libraw1394.so.11.0.1
b7715000-b7716000 rw-p 0000f000 08:01 31943      /usr/lib/i386-linux-gnu/libraw1394.so.11.0.1
b7716000-b774e000 r-xp 00000000 08:01 28696      /usr/lib/i386-linux-gnu/libjpeg.so.8.4.0
b774e000-b774f000 rw-p 00037000 08:01 28696      /usr/lib/i386-linux-gnu/libjpeg.so.8.4.0
b774f000-b775f000 r-xp 00000000 08:01 43041      /usr/lib/libcoinblas.so.1.2.0
b775f000-b7760000 rw-p 0000f000 08:01 43041      /usr/lib/libcoinblas.so.1.2.0
b7760000-b7763000 rw-p 00000000 00:00 0 
b7763000-b7764000 r-xp 00000000 00:00 0          [vdso]
b7764000-b7780000 r-xp 00000000 08:01 63006      /lib/i386-linux-gnu/ld-2.13.so
b7780000-b7781000 r--p 0001b000 08:01 63006      /lib/i386-linux-gnu/ld-2.13.so
b7781000-b7782000 rw-p 0001c000 08:01 63006      /lib/i386-linux-gnu/ld-2.13.so
bfa6a000-bfa8b000 rw-p 00000000 00:00 0          [stack]
Aborted

kt10aan, 2015-09-22:
I get a similar error each time I quit robotInterface. Everything shuts down fine,  but at the end, instead of quitting, it segfaults. System is Ubuntu 14.04 and the cub software is from June. I am not at work right now, I can give exact version and log tomorrow morning.

pattacini, 2015-10-11:
Any news on this?

For future posting on software bugs like `seg-faults`, let's try to rely on `yarp`, `icub-main` etc. tracking system rather than here. Actually, `icub-support` is meant pretty much for hardware\mechanics issues with the robots.

lornat75, 2015-10-20:
Moved to https://github.com/robotology/icub-main/issues/218

closed 2015-10-20 by lornat75