robotology/mesh-support #61: iCubLondon01 - Torso fault + smoke

Tobias-Fischer, 2015-09-24:
Hi all,

we have an issue with the torso of our iCub. One day we started up `robotInterface` as usually, when we heard a weird noise and smelled some burned electronics / plastics. We pressed the red button very quickly, and were able to avoid more damage. It happened just before the iCub is usually going to calibrate its parts (nothing moved yet).

We have the log file, were you can see that the main problem is with board 6 on can bus 0 (board label 0B3). We looked up the board in the motor and placement diagram, and visually inspected the board. We found some burning marks on one of the green connectors, as well as the one small black boards (see attached photos).

I guess we probably need a replacement board. Would it be hard to replace ourselves?

In the meantime, we can run the iCub with disabled torso and disabled cartesian solvers, which of course is no long term solution ;-).

Regarding issue https://github.com/robotology/icub-support/issues/48: It seems like since the board failed / has been deactivated, we have smoother movements of the head.

Thanks and best wishes,
Tobi, and @maxime-petit 
[image]
[image]
[image]
[image]

Log file:
https://gist.github.com/Tobias-Fischer/fad5b1f5766368d494f2

julijenv, 2015-09-24:
Hi @Tobias-Fischer and @maxime-petit ,
from the picture it looks quite clear that the BLP got burned, the reason, so far, remains unexplained and of course need to be understood before any further change or replace it.
in any case please from now check the flat cable from BLP (the burned one) and the board at the other extremity of the flat cable to see if there are some used part, or for exmaple the red white and black cables from motor going to the BLP to seen if they are worn.
I am going to talk to @super-ste about it and understand how we can give you and send you the spare parts you need (surely BLP = power board the one already burned and BLL = control board).

super-ste, 2015-09-25:
The replacement parts are on the way.
Please contact us if you need help in replacing them.

In particular the BLP (the board with the black heat exchanger) is to be attached to the aluminum support with the given tape.

Tobias-Fischer, 2015-09-25:
Great, thanks so much for shipping them over so quickly! We'll get in touch via Skype to replace the parts.

Best, Tobi and @maxime-petit

Tobias-Fischer, 2015-09-28:
Hi @super-ste and @julijenv,
we were able to replace the boards, and iCubLondon01 is running fine again! Thanks a lot :)!

Best,
Tobi and @maxime-petit

closed 2015-09-28 by Tobias-Fischer