robotology/mesh-support #62: eye lids on blue icub (03) at IIT get stuck

hagenlehmann, 2015-09-24:
When trying to run the blinker on the blue iCub, the eye lids are not controllable because they get stuck in different positions on the eye balls. It seems to be a structural problem. It would be important for us to use the eye lids on bluey because we need to test the blinker together with the face recognition / following which is, due to different cameras, only possible on the black and the blue iCub.

Many thanks for looking into the problem,
Hagen

pattacini, 2015-09-24:
This issue would need attention 'cause we have a guest coming to visit us next week, who is going to work with the eye lids too.

hagenlehmann, 2015-09-25:
The blinking appears to work great now! Many thanks! We will check it over a longer period of time next week and let you know.

Many thanks again,
Hagen

closed 2015-09-25 by hagenlehmann