robotology/mesh-support #63: iCubLyon01

AnneLaureM, 2015-09-25:
Hi icub-support !!

Since we received our iCub we were able to start several times successful (good calibration ...). 
Two days ago, after starting the robot, we had the joints 8 to 15 in "Hardware-fault" .
[image]

Until now, we have essentially used the vision (no motor action with arms). We tried to remove the cover to check that the connectors are well connected. At first blush, they seem connected properly ...

Here are the logs of robotInterface :
```
// Startup

[INFO]Left_Hand_Calibrator : starting calibration 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  0  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  1  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  2  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  3  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  4  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  5  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  6  before starting the calibration procedure! 
[ERROR]Left_Hand_Calibrator : detected an hardware fault on joint  7  before starting the calibration procedure! 
[DEBUG]Left_Hand_Calibrator : Joints calibration order: ( 0 1 3 5 ) ( 2 4 6 7 )  
...

[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  1  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:1) in HW_FAULT
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  3  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:3) in HW_FAULT
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  5  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:5) in HW_FAULT
...

[DEBUG]sharedcan [0] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.01[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None
[DEBUG]sharedcan [5] msg from board 7: MAIS error 
[DEBUG]sharedcan [5] msg from board 5: MAIS error 
[ERROR]Left_Hand_Calibrator :Timeout while calibrating  0 
 
[ERROR]Left_Hand_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit
 
[ERROR]Unable to set the control mode of a joint ( j:0) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:1) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:3) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:5) in HW_FAULT
[DEBUG]Left_Hand_Calibrator : skipping maxPwm=0 of joint  2 
[DEBUG]Left_Hand_Calibrator : skipping maxPwm=0 of joint  4 
[DEBUG]Left_Hand_Calibrator : skipping maxPwm=0 of joint  6 
[DEBUG]Left_Hand_Calibrator : skipping maxPwm=0 of joint  7 
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  2  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:2) in HW_FAULT
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  4  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:4) in HW_FAULT
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  6  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:6) in HW_FAULT
[DEBUG]In calibration  Left_Hand_Calibrator : enabling joint  7  to test hardware limit 
[ERROR]Unable to set the control mode of a joint ( j:7) in HW_FAULT

// When we close robotInterface ...

[ERROR]Left_Hand_Calibrator : joint  0  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  1  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  2  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  3  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  4  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  5  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  6  has an hardware fault, skipping park 
[ERROR]Left_Hand_Calibrator : joint  7  has an hardware fault, skipping park 
[ERROR]positionMoveRaw: skipping command because    joint  0 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  1 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  2 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  3 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  4 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  5 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  6 is not in VOCAB_CM_POSITION mode 
[ERROR]positionMoveRaw: skipping command because    joint  7 is not in VOCAB_CM_POSITION mode 
[DEBUG]Head_Calibrator : Moving to park position, joint: 0 
Registering new thread 2 out of 500
[DEBUG]Left_Arm_Calibrator : Moving to park position, joint: 0 
Registering new thread 2 out of 500
[DEBUG]Torso_Calibrator : Moving to park position, joint: 0 
Registering new thread 3 out of 500
[DEBUG]Right_Arm_Calibrator : Moving to park position, joint: 0 
Registering new thread 3 out of 500
[DEBUG]Right_Hand_Calibrator : Moving to park position, joint: 0 
Registering new thread 2 out of 500
[DEBUG]Left_Hand_Calibrator : Park  completed 
[ERROR]Unable to set the control mode of a joint ( j:0) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:1) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:2) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:3) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:4) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:5) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:6) in HW_FAULT
[ERROR]Unable to set the control mode of a joint ( j:7) in HW_FAULT
[DEBUG]left_hand_calibrator finished park device left_hand_mc 
```
Cheers,

Anne-Laure

julijenv, 2015-09-25:
After a little check on slkype, it appears that the first FC connector (little boards above the knuckles on the hand got a little misplaced) repuuting it on place (parallel to MAIS board look like it solve the issue.
This issue remains open until no further complaints about it.

lornat75, 2015-10-20:
@marijnfs can we close this issue?

AnneLaureM, 2015-10-27:
Hi all,
I think I have finally identified the cable which most of the time prevents fingers run correctly. This cable is located on the second FC connector and it is desoldered. Do you think it is possible to repeat the missing welding at INSERM? And if so, can you explain us where the cable must be connected?
I have attached two pictures showing the disconnected cable:

[image]
[image]

Cheers,
Anne-Laure

AnneLaureM, 2015-11-02:
Hi all,
Does anyone have an idea so that we can fix this issue ?
Thanks,
Anne

julijenv, 2015-11-02:
be on skype todqy and i'll show you where to solder it or in the qworst case scenario i'll put an arrow on where to solder back the wire.
C U Later

julijenv, 2015-11-04:
Hiu @AnneLaureM , Any news after the resoldering, does it solve the issue?

AnneLaureM, 2015-11-04:
Hi !!

Finally, we have resoldered two black cables. 
The first cable from the second card to the first card. And the second cable from the fourth card to the fifth card.
After starting the robot, we get some errors (see attached log files). We have started and stopped robotInterface three times:
[logRobotInterface_04_11_15-1631.txt](https://github.com/robotology/icub-support/files/27228/logRobotInterface_04_11_15-1631.txt)
[logRobotInterface_04_11_15-1635.txt](https://github.com/robotology/icub-support/files/27227/logRobotInterface_04_11_15-1635.txt)
[logRobotInterface_04_11_15-1640.txt](https://github.com/robotology/icub-support/files/27229/logRobotInterface_04_11_15-1640.txt)

For each of these three times, the fingers of the left hand move correctly when we start the robot but we had four joints in "idle" mode on yarpmotorgui:
[image].
Both LEDs of each color of FC connector was lighted.

So we have stopped robotInterface, turn off the motors, restarted the motors and robotInterface. And after this, a lot of errors appears (see log file):
[logRobotInterface_04_11_15-1644.txt](https://github.com/robotology/icub-support/files/27236/logRobotInterface_04_11_15-1644.txt)
At that time the orange LED of the FC connector was off.

Thanks !

Anne

AnneLaureM, 2015-11-05:
Hi !

We have resoldering several cables, the purple one on the second board and the black cable (on the board 4 and 5)...
After restarting the robot, I get these logs:
[robotInterface_logs_05_11_15_1911.txt](https://github.com/robotology/icub-support/files/28262/robotInterface_logs_05_11_15_1911.txt)
with this picture of yarpmotorgui:
[image]

So I have switched off the motors & robotInterface, then restart both. I saw that the orange/red LED on the fifth MAIS board was off. I brought up the card over and replace it. The LED is turned on again. I launched robotInterface. These are the new logs:
[robotInterface_logs_05_11_15_1914.txt](https://github.com/robotology/icub-support/files/28266/robotInterface_logs_05_11_15_1914.txt)
[image]
So each time we restart the motors, it seems that we lose the connection to the fifth MAIS board.

Anne

julijenv, 2016-02-08:
Hi @AnneLaureM , Is everything alright with iCub?

julijenv, 2016-10-04:
Hi @AnneLaureM , Is everything alright with iCub? can we close this issue???

closed 2018-01-16 by julijenv