robotology/mesh-support #64: Left shoulder tendon broken

bhigy, 2015-09-28:
One of the wires in the left shoulder broke, same as what happened a few days ago (#60).

I was using the actions rendering engine when this happened, performing the EXPECT command. Ugo suggested that the cartesian solver, which is supposed to prevent this, is not working properly with the purple iCub.

pattacini, 2015-09-28:
Well, what I actually suppose is that there might be some mechanical differences with standard shoulders the solver is unaware of.

bhigy, 2015-09-28:
Apparently, this already happened twice at iCub Summer School, while trying to make the red ball demo work on the purple Both time, it was the left arm.

pattacini, 2015-09-28:
Both `ARE` and `demoRedBall` make use of the **Cartesian Solver**, which in turn is instructed to find a suitable configuration of the joints that prevent shoulders cables from breaking/snapping out due to their length limits.

If this unwanted event keeps happening it means that the constraints in the solver are somehow not adequate for the left arm :arrow_right: there must be some differences in the mechanics.

spada1978, 2015-09-29:
Replaced the cable and the cable 7 1 because they are broken at the crimp.

spada1978, 2015-09-29:
Sorry the cable are 8 and 1

pattacini, 2015-09-29:
thanks @spada1978

pattacini, 2015-10-11:
@super-ste do we have any indication that the mechanics is actually different?

super-ste, 2015-10-12:
Surely the mechanics is completely different! Being V2 instead of V1.
We never used such a shoulder though so I have no previous experience of this repetitive fault.

On the other hand the two shoulders have been designed to share the same tendons and hardware limits and hand-crimped tendons can fail without a reason (variability in the production).

So I'd keep monitoring the issue a bit longer before raising up the red flag (this issue has been raised 14 days ago and no other failure has been reported since then).

pattacini, 2015-10-12:
@super-ste 
I wondered whether there is any difference between the two shoulders, 'cause only the left one is having problems. By the way, the solvers are aware that they're working with `v2` mechanics (I've checked the configuration files in the repository).

I'm afraid no other failures have been reported since then simply because I told @bhigy not to use the left arm before further investigations.

lornat75, 2015-10-20:
Moved to https://github.com/robotology/icub-main/issues/217

super-ste, 2015-10-22:
We agreed to keep using both arms (suspect fatality in high mortality ratio of tendon 8)

@bhigy, next time you use your swt don't make any difference and keep using the left arm as normal.

bhigy, 2015-10-29:
I don't think I will do the experiment with the purple again. Should I close this issue?

super-ste, 2015-11-03:
Yep. Let's close it. 

We will reference this issue if we will encounter this in the future.

super-ste, 2015-12-10:
(*) Verify PID gains for all left shoulder joint (make sure they are consistent with the right arm ones

(*) bring the robot in a controllef fashion close to the known breaking position and verify encoder and position reading form the sensors. Compare with the right arms

Report here the findings

DanielePucci, 2015-12-10:
:+1: for the second point.

randaz81, 2016-05-09:
Investigating the encoders measurements while keeping the left shoulder joints in idle, showed that  the yaw admissible range is (-37 80) but the hardware limit is encountered at -35. That's the reason why the tendon breaks. 
Please note that (-37 80) is the correct (or, at least the same) value employed on all robots equipped with a V2 arm. I'm thus supposing that the joint is currently exceeding the limit because it is miscalibrated. I currently reduced the joint limit to (-32 80) because the robot was about to leave for a demo but:
- the correct way to proceed is to recalibrate the arm, while keeping the original joint limits.
- we should check if other robots are affected by the same issue. It may be that these limits are too tight (a safe band of 5 degrees between the software and the hardware limit is recommended) 
- even if the raw calibration is ok, if the user sets an invalid (or very large) calibrationDelta, the operating range of the joint may be shifted in the same erratic way.
- the automatic test  https://github.com/robotology/icub-tests/tree/master/src/jointLimits is always able to detect this failure and should be always run on new robots (or if a joint a recalibrated)!

closed 2018-01-16 by julijenv