robotology/mesh-support #66: Right thumb doing weird unexpected movements

bhigy, 2015-09-30:
When asking the robot to give an object through ARE, the thumb is sometimes making weird movements. Could be due to the new mechanics in the hand.

@randaz81 I can show you when you have a moment.

lornat75, 2015-10-20:
Any update on this?

super-ste, 2015-11-17:
@randaz81 can you have a look at this?
Is this an issue with the coupling of the joint in the thumb or is it there any hardware problem?

kt10aan, 2015-11-17:
If it is any help, the  same issue happens to iCubOsaka01 when using ARE, on the thumb of the right wrist. And it's going on for quite a few months now. It is just that we were not using ARE much to open an issue about it.

super-ste, 2015-11-17:
@kt10aan @bhigy does it happen ONLY on the right hand? in both cases?

Sounds weird to me ... but this may be a lead ...

bhigy, 2015-11-17:
I didn't use the left hand a lot so I can't really say.

kt10aan, 2015-11-19:
@super-ste It happens only on the right hand. I will take a video and post it here.

kt10aan, 2017-05-02:
Regarding the Osaka01, this issue can be closed.

closed 2017-08-16 by julijenv