robotology/mesh-support #68: torso failure on blue iCub (03) in IIT

hagenlehmann, 2015-09-30:
Blue iCub collapses after calibration frequently because of hardware failure in one of the torso joints.

cheers,
Hagen

pattacini, 2015-10-07:
@hagenlehmann is this issue really fixed? I've seen people still struggling to figure it out.

hagenlehmann, 2015-10-07:
After Marco’s intervention we (Frank, Reza and I) have not experienced any further problems last week. 
Giulia just told me that she has experienced the error once in the last two days. She talked to Julien and they found a problem with one of the encoders. I was not aware of this, sorry - I will reopen the issue

pattacini, 2015-10-07:
:+1: thanks

julijenv, 2015-10-07:
it has been changed once more the sensor of the belly, let's hope that it will solve the issue once for all, or else I guess that you will need to find the root of the issue somewhere else.

closed 2015-10-12 by julijenv