robotology/mesh-support #69: [icubGenova01] Issues after IROS demo

alecive, 2015-10-02:
* [ ] FT sensor on left leg is spiking and needs to be replaced (the one in the SS toolbox is probably not calibrated)
 * [ ] Hall sensor, left hand, index, proximal is flaky (i.e. `licchelacche`)
 * [ ] Neck harmonic drive (right hand side) gets stuck (o da ingrassare o da cambiare)

pattacini, 2015-10-02:
Don't use the robot name in the issue title, we have the labels for that :smirk:

super-ste, 2015-10-02:
@pattacini label suck I prefer the name in the title. It's clearer and easier to look.

pattacini, 2015-10-02:
That's at least questionable.
With labels you have a much easier way to sort issues by robots.

alecive, 2015-10-02:
With titles you have a better way to sort issues by robots **without** resorting to a further step ( ie filtering them by hand).

pattacini, 2015-10-02:
Where is the option to sort by *issue title*? To do so, you have to fill in the search field with the robot name.

By contrast, with labels, you just click on the desired robot label and that's all folk.

spada1978, 2015-10-05:
Repaired proximal thumb right hand and glued fingertip left hand anulare

spada1978, 2015-10-06:
Replaced ft sensor s7n 209

spada1978, 2015-10-07:
Neck harmonic drive fatted (right hand side)

spada1978, 2015-10-07:
verified sensor proximal index and middle everything is working properly

pattacini, 2015-10-11:
Thanks @spada1978 
If you deem this issue complete, please close it.

spada1978, 2015-10-12:
It was not closed because it lacks the functionality test of the repaired finger tip, but if you feel like I close

pattacini, 2015-10-12:
Nope, go ahead, I thought all the points of the issues were addressed.

closed 2015-10-12 by spada1978