robotology/mesh-support #70: Issues when running all the motors on Vizzy

plinioMoreno, 2015-10-05:
@randaz81, please have a look to the warnings  at the file [warnings_while_running_ball_tracking.txt](https://github.com/robotology/icub-support/files/7625/warnings_while_running_ball_tracking.txt) and [errors_during_calibration.txt](https://github.com/robotology/icub-support/files/7621/errors_during_calibration.txt), to let me know how to tackle them.

For the rest of the issues, please forward this message to the people that work with the cfw-002 kernel module. Thanks in advance. The text as follows:

After recent upgrades of the yarp and icub-main on our robot Vizzy around july (the last commit added on the versions we have running on Vizzy) as follows:

YARP
commit e2c4160f1093280d5b84ba9ca04996432c83f39e
Author: Daniele E. Domenichelli <daniele.domenichelli@iit.it>
Date:   Thu Jul 30 16:22:22 2015 +0200

icub-firmware-shared
commit a3c8ae09323d9a39a3f6a2c576fb9cd32baedd53
Author: davege89 <davide.pollarolo@iit.it>
Date:   Mon Jun 29 17:56:15 2015 +0200

icub-main
commit f8e147693712753b9b3c6e337aee82f3691f1f0c
Author: marco.accame <marco.accame@iit.it>
Date:   Thu Jul 30 15:30:06 2015 +0200

We started to have problems again when running Vizzy and all the modules for the ball tracking demo. In the attachments you can see the errors and warnings, and I will explain them later. Before, I would like to ask the following questions:

1. We would like to upgrade our ubuntu version to 14.04, which I checked has the 3.19 kernel version. It looks like the newer ubuntu versions went back to kernels less than 3.2, which I guess is good for running the cfw002 kernel module. Do you know if the cfw002 driver works with the kernel 3.19? If that is the case, I will go ahead and install the 14.04 ubuntu.

2. Do you have an upgrade of the cfw-002 firmware file? If yes, can you point to me where it is?

Regarding the problems, we are getting some errors and warnings that I would like to know if we can do something to solve them:

1. When starting the calibration procedure, we got the errors in the attached file errors_during_calibration.txt
2. At random times, the robot stops working because of the errors in the attached file read_error_that_stops_everything.txt
3. The warnings that we get from robotInterface after the read error happens are in the attached file warnings_after_read_error.txt

Other type of warnings that we get are in the attached file warnings_while_running_ball_tracking.txt . I wonder if these warning can be removed and how to do it.

P.S. I attached also the picture of our cfw-002 for you to recall our setup.
[errors_during_calibration.txt](https://github.com/robotology/icub-support/files/7621/errors_during_calibration.txt)
[read_error_that_stops_everything.txt](https://github.com/robotology/icub-support/files/7623/read_error_that_stops_everything.txt)
[warnings_after_read_error.txt](https://github.com/robotology/icub-support/files/7624/warnings_after_read_error.txt)
[warnings_while_running_ball_tracking.txt](https://github.com/robotology/icub-support/files/7625/warnings_while_running_ball_tracking.txt)
[image]

lornat75, 2015-10-20:
For the CFW02 I think the driver should be compatible, otherwise we can fix the compatibility issues.

@randaz81 maybe you can have a look at the logs?

plinioMoreno, 2015-11-20:
We did a change in the hardware setup of the cfw002 and now this is not happening. I'll open it again if we see this error again.

closed 2015-11-20 by plinioMoreno